ifm/ifm3d #195: what's the principle of camera calibration of "O3D303"?

Carolinadeveloper, 2020-02-05:
Hi guys,

I understand that we can turn on the rectification flags for amplitude and the distance image through "ifm3d dump" to calibrate the camera. In addition, I have already looked through other issues but I am still confused about the principle of calibration, since the answer in other issues link to different paper and multiple resources. Therefore, could you please provide a link/paper that can explain the principle of calibration.

Looking for your reply,

carolina

theseankelly, 2020-02-13:
Hi @Carolinadeveloper -- could you help me understand a bit more about what you're trying to learn?

There are two types of calibrations:
* intrinsic: This releates to the physics of the camera lens and how light is distorted passing through the lens to the photo imager. This information can be used to do rectification (de-warping of the data). Our cameras are factory calibrated for intrinsics and as you've noted there are built-in features to do this work automatically. I think most of the other issues you're talking about are referring to this type of calibration

* extrinsic: This is the process of registering the camera's optical coordinate frame (the coordinate frame in which data values are returned from the sensor) against another global coordinate frame, say a vehicle frame, or with respect to other cameras.

Do either of these sound like what you're trying to achieve?

Carolinadeveloper, 2020-02-13:
> Hi @Carolinadeveloper -- could you help me understand a bit more about what you're trying to learn?
>
> There are two types of calibrations:
>
> * intrinsic: This releates to the physics of the camera [...]

Hi theseankelly @theseankelly ,

What I want to learn is how u guys factory calibrate for intrinsic? Since I want to recalibrate the camera and check the parameters you provided to improve the accuracy. I did Zhang's calibration with grayscale/rgb of chessboard images collected before, but for O3D303 I only found DistanceImage, AmplitudeImage, and ConfidenceImage, I don't know which images information can be used for recalibration? 

Carolina

graugans, 2020-02-13:
@Carolinadeveloper Carolinadeveloper for the camera distance calibration we have different ways to do this. Due to the fact we need distance information from defined situations to calibrate the 3D offset to fit the camera body those values are calculated during this process. This is different from the way a classical 2D camera is calibrated. This information is considered confidential an exceeds the detail level of an GitHub issue. Please get in touch with your ifm representative to discuss the details.
For using the checkerboard the amplitude image is the image you are looking for.

Carolinadeveloper, 2020-02-13:
> @Carolinadeveloper for the camera distance calibration we have different ways to do this. Due to the fact we need distance information from defined situations to calibrate the 3D offset to fit the [...]

Hi  @graugans,

Thank you for your reply, if I use checkerboard & amplitude image to recalibrate, how can I modify the calibrated parameters, is there provide any interface like "ifm3d dump" ?

theseankelly, 2020-02-18:
Hi @Carolinadeveloper -- there is no mechanism for saving the calibration parameters back to the device. Your custom calibration will produce a new set of unit vectors, which you will need to apply to the radial distance image and generate cartesian information yourself.

That said, we have a fairly advanced factory calibration process and I'd love to learn more about why you feel the need to recalibrate -- what kinds of measurement error are you seeing that's driving this effort?

theseankelly, 2020-03-16:
Hi @Carolinadeveloper I'm closing this issue because as far as I can tell the questions have been answered. Feel free to reactivate if not true.

closed 2020-03-16 by theseankelly