ifm/ifm3d #196: Manage camera power off

HEVA-sea, 2020-02-06:
Hi,

I use a OD303 camera and there's a problem that I haven't managed to resolve yet :
The power supply of the PC linked to the camera is independent of the power supply of this camera  (the PC is connected to an inverter). It is therefore possible that the power supply of the camera is cut, while the application managing it on the PC is still running. My application uses the FrameGrabber tool to get the images and in this case, the main FrameGrabber's thread is stopped due to the loss of communication with the camera. However, no error is thrown. The only information available are these traces in the consol :

```
W0127 17:08:03.399521  7700 frame_grabber_impl.hpp:561] Unknown error
I0127 17:08:03.399521  7700 frame_grabber_impl.hpp:569] FrameGrabber thread done.
```

When the power to the camera comes back and when an image is requested, there is nothing happening since the FrameGrabber is no longer operational, but there is still no exception thrown.

Is there a way to catch this kind of events to be able to properly manage it ?

Thanks!

theseankelly, 2020-02-13:
Hey @HEVA-sea thanks for the report. We'll look in to it and get back to you.

theseankelly, 2020-03-18:
Hi @HEVA-sea -- am I correct to assume you're using `fg->WaitForFrame()` without specifying a timeout value, and since no exception is thrown, this blocks forever?

HEVA-sea, 2020-03-18:
Yes, it's exactely that

theseankelly, 2020-03-18:
Which OS and which version of `ifm3d` are you working with?

HEVA-sea, 2020-03-18:
I'm working on Windows 10 and I have the version 0.12.0

HEVA-sea, 2020-06-09:
Hi, is there any improvement regarding my issue?

theseankelly, 2020-06-09:
I do not have any updates at this time. I shall add it to my list for this week.

theseankelly, 2020-06-18:
Hey @HEVA-sea -- by design, the `ifm3d` library is using blocking sockets and will not timeout at the TCP/IP layer in the case of a rebooted camera. I recommend calling `FrameGrabber::WaitForFrame` with a reasonable timeout value for your application. If the timeout condition hits (`FrameGrabber::WaitForFrame` returns `false`) then your error handling logic may recreate the `ifm3d::FrameGrabber` object (i.e. `fg.reset(new ifm3d::FrameGrabber(camera))` to re-establish the connection.

That said, your error message and the termination of the framegrabber thread is unexpected. While the advice above still applies and would catch this case, it might be worth to investigate what's really happening. Unfortunately I haven't been able to reproduce your issue with the following minimal example: 

```cpp
#include <iostream>

#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main()
{
  auto c = ifm3d::Camera::MakeShared();
  auto fg = std::make_shared<ifm3d::FrameGrabber>(c);
  auto im = std::make_shared<ifm3d::ImageBuffer>();

  std::uint32_t i = 0;
  while(fg->WaitForFrame(im.get(), 0))
  {
    std::cout << "Got frame " << i++ << '\n';
  }

  return 0;
}
```

While the loop runs, I remove power to my camera and observe the system logs. The library continues blocking on the socket forever (by design as discussed above). Can you confirm whether the issue occurs for you using this basic example?

Alternatively can you provide a minimal example which does reproduce your issue?

theseankelly, 2020-08-03:
@HEVA-sea -- have you been able to check the suggestions above? Anything else I can do to help?

commit 5cec06ac (2023-01-20): Merge branch '196-software-update-changes-for-the-1-0-x-o3r-release' into 'main'

Add O3R recovery based update

Closes #196 and #193

See merge request syntron/support/csr/ifm3d/ifm3d!202