ifm/ifm3d #197: Not able to install IFM in ubuntu 18.04 because the public key is not available.

ShettyHarapanahalli, 2020-02-06:
Not able to install IFM in ubuntu 18.04, 
I get below error when I try to install,   
$ sudo sh -c 'echo "deb [arch=amd64] https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic main" > /etc/apt/sources.list.d/ifm-robotics.list' 
$$ sudo apt-get update 
....
Err:14 https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic InRelease                               
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8AB59D3A2BD7B692
Reading package lists... Done                                                                                           
W: GPG error: https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8AB59D3A2BD7B692
E: The repository 'https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Packages (Packages) is configured multiple times in /etc/apt/sources.list:51 and /etc/apt/sources.list.d/cuda.list:1
W: Target Translations (en_IE) is configured multiple times in /etc/apt/sources.list:51 and /etc/apt/sources.list.d/cuda.list:1
W: Target Translations (en) is configured multiple times in /etc/apt/sources.list:51 and /etc/apt/sources.list.d/cuda.list:1
W: Target Packages (Packages) is configured multiple times in /etc/apt/sources.list:51 and /etc/apt/sources.list.d/cuda.list:1
W: Target Translations (en_IE) is configured multiple times in /etc/apt/sources.list:51 and /etc/apt/sources.list.d/cuda.list:1
W: Target Translations (en) is configured multiple times in /etc/apt/sources.list:51 and /etc/apt/sources.list.d/cuda.list:1

theseankelly, 2020-02-06:
Hi @ShettyHarapanahalli  -- in the command line you posted, you're missing the step which adds the key:

```
$ sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 8AB59D3A2BD7B692
```

ShettyHarapanahalli, 2020-02-07:
I had added the key, I had rebooted the pc today now it's working. 

Thanks for your help.

closed 2020-02-07 by ShettyHarapanahalli