ifm/ifm3d #200: Can't write point cloud data to a PCD file

MYSystemLogistics, 2020-02-21:
I'm using an ifm O3D303 ToF camera, I can ping the device and I can see the point-cloud with the ifm3d-pcl-viewer.
I'm trying to save grabbed point cloud to a pcd file following the given _ex-file_io.cpp_ example, but I get the following error when i run it:
```
terminate called after throwing an instance of 'pcl::IOException'
  what():  : [pcl::PCDWriter::writeASCII] Number of points different than width * height!
Annullato (core dump creato)
```
The code:
```
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <pcl/io/pcd_io.h>
#include <pcl/point_cloud.h>
#include "o3d3xx_camera.h"
#include "o3d3xx_framegrabber.h"
#include "o3d3xx_image.h"

int main(int argc, const char **argv)
{
  o3d3xx::Logging::Init();

  o3d3xx::Camera::Ptr cam = std::make_shared<o3d3xx::Camera>();
  o3d3xx::ImageBuffer::Ptr img = std::make_shared<o3d3xx::ImageBuffer>();
  o3d3xx::FrameGrabber::Ptr fg =
    std::make_shared<o3d3xx::FrameGrabber>(
      cam, o3d3xx::IMG_AMP|o3d3xx::IMG_RDIS|o3d3xx::IMG_CART);

  if (! fg->WaitForFrame(img.get(), 1000))
    {
      std::cerr << "Timeout waiting for camera!" << std::endl;
      return -1;
    }

  pcl::io::savePCDFileASCII("point_cloud.pcd", *(img->Cloud()));
  imwrite("amplitude.png", img->AmplitudeImage());
  imwrite("radial_distance.png", img->DepthImage());

  return 0;
}
```
How can I solve it?
Thanks.

theseankelly, 2020-02-24:
@MYSystemLogistics Thanks very much for the code example -- nothing stands out on a first glance. I'll have a look in the next day or two.

theseankelly, 2020-03-17:
Hi @MYSystemLogistics -- I've had time to investigate and I'm unable to reproduce the issue using the exact code you've pasted along with my o3d303.

First a note: the code shared above is from the [`libo3d3xx`](https://github.com/ifm/libo3d3xx) library which is the predecessor of `ifm3d` and has limited support and no new feature development at this point. Do you have strong need to use `libo3d3xx` instead of `ifm3d`? For reference, here is the equivalent sample to what you posted using `ifm3d` instead (you'll see the usage pattern is nearly identical and should be an easy port):

```cpp
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <pcl/point_cloud.h>
#include <pcl/io/pcd_io.h>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main(int argc, const char **argv)
{
  auto cam = ifm3d::Camera::MakeShared();
  auto fg = std::make_shared<ifm3d::FrameGrabber>(
      cam, ifm3d::IMG_AMP|ifm3d::IMG_RDIS|ifm3d::IMG_CART);
  auto img = std::make_shared<ifm3d::ImageBuffer>();

  if (! fg->WaitForFrame(img.get(), 1000))
    {
      std::cerr << "Timeout waiting for camera!" << std::endl;
      return -1;
    }

  pcl::io::savePCDFileASCII("point_cloud.pcd", *(img->Cloud()));
  imwrite("amplitude.png", img->AmplitudeImage());
  imwrite("radial_distance.png", img->DistanceImage());

  return 0;
}
```

That said, I did try to reproduce the issue using your example code based on libo3d3xx and was **unable** to reproduce the failure. I receive a .pcd file of size 709k which renders correctly in `pcl_viewer`. Best I can tell, your code looks correct to me.

Could you do a couple things for me:
1. Output your camera configuration by running `ifm3d dump` and pasting the output to this issue?
2. Dump some basic information about the `img->Cloud()` you're receiving, like so:
```cpp
std::cout << "Cloud size: ("
               << img->Cloud()->width << ", "
               << img->Cloud()->height << ")" << std::endl;
std::cout << "isOrganized: " << img->Cloud()->isOrganized() << std::endl;
```

MYSystemLogistics, 2020-03-18:
Hi @theseankelly , thank you for your reply.

1. This is what I see with ifm3d dump:
```
$ ifm3d dump 
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "Run 3D camera at 10Hz without overheating",
        "Id": "476707713",
        "Imager": {
          "AutoExposureMaxExposureTime": "600",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "true",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "true",
          "EnableFilterAmplitudeImage": "true",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "7700",
          "ExposureTimeList": "1100;7700",
          "ExposureTimeRatio": "7",
          "FlyingPixelEraser": "2",
          "FrameRate": "10",
          "MaxAllowedLEDFrameRate": "13.8",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_moderate",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Test_10Hz",
        "PcicEipResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"di [... cut ...]
# Create executable
add_executable(File_IO File_IO.cpp)
target_link_libraries(File_IO
		      ifm3d::camera
                      ifm3d::framegrabber
                      ifm3d::image
                      ${OpenCV_LIBRARIES}
		      ${PCL_LIBRARIES}
                      ${PCL_COMMON_LIBRARIES}
                      ${PCL_IO_LIBRARIES}
                      )
```

4. Can I somehow exploit `XYZImage()` or `Cloud()` to separately save into vectors/arrays the X, Y and Z coordinates of the point-cloud?

theseankelly, 2020-03-18:
1. I see no issue with that config. I've loaded it to my camera and the code is still successful.

2. Those values look correct to me.

3. Maybe. Here's my CMakeLists.txt:
```cmake
cmake_minimum_required(VERSION 3.10)

find_package(ifm3d CONFIG REQUIRED 
  COMPONENTS
    camera
    framegrabber
    image)
find_package(PCL 1.7.1 REQUIRED COMPONENTS common io)
find_package(OpenCV REQUIRED)

include_directories(${PCL_INCLUDE_DIRS})
link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

add_executable(example main.cpp)
target_link_libraries(
  example
  ifm3d::camera
  ifm3d::framegrabber
  ifm3d::image
  ${PCL_LIBRARIES}
  ${OpenCV_LIBRARIES}
  )
```
The most obvious difference is the version of PCL being used -- are you actually using PCL 1.2 (from 2011) or is there a more recent version installed?

4. Sure, that's a good idea. `XYZImage()` returns a 3-channel `cv::Mat` of type `16S`. You can use `cv::split` to separate the channels into individual arrays like this:
```cpp
  cv::Mat xyz[3];
  cv::split(img->XYZImage(), xyz);
```
And then I used the example [here](https://stackoverflow.com/questions/26681713/convert-mat-to-array-vector-in-opencv) to convert those individual cv::Mats into vectors (but you will loose the organization of rows/cols):
```cpp
  std::vector<std::int16_t> x;
  if (xyz[0].isContinuous())
  {
    x.assign(xyz[0].data, xyz[0].data + xyz[0].total());
  }
  else
  {
    for (int i = 0; i < xyz[0].rows; ++i)
    {
      x.insert(
          x.end(), xyz[0].ptr<std::int16_t>(i),
          xyz[0].ptr<std::int16_t>(i) + xyz[0].cols);
    }
  }
```
(Repeat for `y` and `z` using `xyz[1]` and `xyz[2]`.

MYSystemLogistics, 2020-03-19:
1. & 2. I tried with your code and with your CMakeLists.txt file, but again the same error. It seems there is a problem with `*(img->Cloud())`.
Is it normal that the cloud width is a number around 10^6 and the cloud height is around 32k? 
I think `savePCDFileASCII` is expecting a smaller cloud size, like 176 * 132 ≈ 23k points, which is my camera resolution.

3. Regarding the version of PCL being used, that 1.2 was just a copy&paste typo. The version installed on my Ubuntu 18.04 is the PCL 1.8.1.

4. I split the coordinates into vectors as you suggested. I noticed that the new vectors containing the xyz coordinates are full of values like 0 and 255. Is it normal? Which is the unit of measure of the returned XYZImage() channels? I expect them to be in millimeters.
I also tried to create a PCD file following the [documentation](http://www.pointclouds.org/documentation/tutorials/writing_pcd.php) on the pointcloud website, but when I try to visualize it the `pcl_viewer` shows me a messy point cloud cube, which is far away from what I see with the `ifm3d-pcl-viewer`.
Is this due to the lost of the rows/cols organization?

theseankelly, 2020-03-23:
@MYSystemLogistics sorry, I misread your comment and saw the valid dimensions for the `XYZImage` and missed the fact that the dimensions of the `img->Cloud()` object were invalid. In my not-repro, the dimensions are as expected (176x132).

Can you grab logs per the guidance here:
https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#ifm3d-logging

MYSystemLogistics, 2020-03-23:
@theseankelly no problem.
Since I have no `foo.json` file, I created a json file saving my `ifm3d dump` sensor state.
So I made the logging with the `IFM3D_VLOG=15 ifm3d config < TestSensorState.json` command.
The output of the `ifm3d.INFO` file is:
```
Log file created at: 2020/03/23 12:19:55
Running on machine: Arcanite
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0323 12:19:55.346953 19022 frame_grabber_impl.hpp:178] Camera connection info: ip=192.168.0.69, port=50010
I0323 12:19:56.341259 19023 frame_grabber_impl.hpp:596] FrameGrabber thread done.
```
There isn't any .ERROR or .WARNING file in the tmp folder.

theseankelly, 2020-03-23:
@MYSystemLogistics sorry, I've given you poor instructions. I didn't re-read them in detail before sending the link and I thought they were more clear. The whole "`config < foo.json`" part of the instruction is just an example.

The key is, can you set that environment variable `IFM3D_VLOG=15` to enable verbose logging, and then reproduce the scenario. For example, you can run `IFM3D_VLOG=15 ./example` where 'example' is to be replaced by the name of your binary.

MYSystemLogistics, 2020-03-23:
@theseankelly Ok, maybe I was taking the example way too literally.
This is the new output:
```
Log file created at: 2020/03/23 15:57:00
Running on machine: Arcanite
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0323 15:57:00.051069 22814 camera_impl.hpp:406] Increasing XML-RPC response size limit...
I0323 15:57:00.051697 22814 camera_impl.hpp:410] Initializing Camera: ip=192.168.0.69, xmlrpc_port=80, password=
I0323 15:57:00.051728 22814 camera_impl.hpp:414] XMLRPC URL Prefix=http://192.168.0.69:80
I0323 15:57:00.126024 22814 camera.cpp:207] Instantiating O3D...
I0323 15:57:00.126368 22814 camera_impl.hpp:406] Increasing XML-RPC response size limit...
I0323 15:57:00.126395 22814 camera_impl.hpp:410] Initializing Camera: ip=192.168.0.69, xmlrpc_port=80, password=
I0323 15:57:00.126413 22814 camera_impl.hpp:414] XMLRPC URL Prefix=http://192.168.0.69:80
I0323 15:57:00.126435 22814 camera_impl.hpp:419] Dtor...
I0323 15:57:00.126796 22814 frame_grabber_impl.hpp:474] c_command: 1000L000000644
1000c000000628
      {
        "layouter": "flexible",
        "format"  : {"dataencoding":"ascii"},
        "elements":
         [
           {"type":"string", "value":"star", "id":"start_string"},
           {"type":"blob", "id":"distance_image"},
           {"type":"blob", "id":"normalized_amplitude_image"},
           {"type":"blob", "id":"x_image"},
           {"type":"blob", "id":"y_image"},
           {"type":"blob", "id":"z_image"},
           {"type":"blob", "id":"confidence_image"},
           {"type":"blob", "id":"extrinsic_calibration"},
           {"type":"string", "value":"stop", "id":"end_string"}
         ]
      }
   
I0323 15:57:00.126845 22814 frame_grabber_impl.hpp:496] t_command: 1001L000000007
1001t
I0323 15:57:00.141258 22814 frame_grabber_impl.hpp:178] Camera connection info: ip=192.168.0.69, port=50010
I0323 15:57:00.141641 22815 frame_grabber_impl.hpp:514] Framegrabber thread running...
I0323 15:57:00.141701 22815 frame_grabber_impl.hpp:542] Connecting to PCIC...
I0323 15:57:00.157835 22815 frame_grabber_impl.hpp:655] Full ticket: '1000L000000007
1000*
'
I0323 15:57:00.191690 22815 frame_grabber_impl.hpp:655] Full ticket: '0000L000255926
'
I0323 15:57:00.214082 22814 byte_buffer.hpp:275] xyzidx=18446744073709551615, xidx=93032, yidx=139544, zidx=186056, aidx=46520, raw_aidx=18446744073709551615, cidx=232568, didx=8, uidx=18446744073709551615, extidx=255848, gidx=18446744073709551615, intridx=18446744073709551615, invintridx=18446744073709551615
I0323 15:57:00.214159 22814 byte_buffer.hpp:377] xfmt=3, yfmt=3, zfmt=3, afmt=2, raw_afmt=4294967295, cfmt=0, dfmt=2, ufmt=4294967295, extfmt=6, gfmt=4294967295, intrfmt= 4294967295, invintrfmt= 4294967295
I0323 15:57:00.214184 22814 byte_buffer.hpp:397] npts=23232, width x height=176 x 132
I0323 15:57:00.274408 22815 frame_grabber_impl.hpp:655] Full ticket: '0000L000255926
'
I0323 15:57:00.371397 22815 frame_grabber_impl.hpp:655] Full ticket: '0000L000255926
'
```
Again no .ERROR or .WARNING file in the tmp folder.

theseankelly, 2020-04-01:
Hey @MYSystemLogistics -- I'm struggling to get to the bottom of this one without a solid repro -- there's clearly something different between your setup and mine. I wonder if ABI compatibility is coming into play.

* For PCL: did you get the library from apt (`'libpcl-dev`, right?) or build from source?
* For ifm3d: Did you build from source or pull from the apt repository?
  * If apt, could you confirm whether you've used the ROS or non-ROS version of the repository listed on the README?
* Which version of `ifm3d` do you have? (`ifm3d --version`)

MYSystemLogistics, 2020-04-04:
Hi @theseankelly, sorry for my late reply but I think I made a mess trying to fix the problem by my self:

- I have a PCL library `1.8` installed from `libpcl-dev` (`usr/include` directory) and a PCL library `1.10` installed from source (`usr/local/include` directory).
- I installed the ifm3d library following the README guide, so first I configured the ifm apt server (ROS version), then I Installed the software from `apt-get install ifm3d-...`
- ifm3d --version gives me `ifm3d: version=0.17.0`

Recently I updated my CMake version from `3.10` to `3.17` and the problem now is that I can't even build the source code: when I launch the `cmake ..` command from the build folder I get these errors:
 ```
CMake Error at CMakeLists.txt:22 (add_executable):
  Target "Save2PCD" links to target "ifm3d::camera" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:22 (add_executable):
  Target "Save2PCD" links to target "ifm3d::framegrabber" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:22 (add_executable):
  Target "Save2PCD" links to target "ifm3d::image" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
```
- Do you have and idea why?

- Small remark: is it possible that the firmware version of my camera is different from yours?

theseankelly, 2020-04-06:
Can you run `ldd` on your executable to determine which of your two PCL installations is being linked?

Which version of ROS do you have installed (should be melodic for ROS1 or one of the ROS2 variants)?

Camera firmware is not relevant at this layer. That said, I did test with the same version you have installed (it's 1.30.4123, found in the output of the `ifm3d dump` command you posted above).

theseankelly, 2020-04-06:
Also, since you're using a highly customized setup I might suggest you build ifm3d from source instead. That way you can link against the proper versions of PCL and other libraries you're trying to use. Be sure to source your ROS environment before building.

Instructions are pretty straightforward and can be found here:
https://github.com/ifm/ifm3d/blob/master/doc/source_build.md

And prior to building, `sudo apt remove` the currently installed `ifm3d` packages, remove the repository from your apt sources.list, and `sudo apt-get update`.

theseankelly, 2020-04-07:
Hi @MYSystemLogistics -- I had one more thought yesterday. If you don't actually need a custom version of PCL, I recommend using the version supplied via ROS. Additionally, we recently released ifm3d (including headers/libs and the command line tool) into the ROS distribution for Kinetic/Melodic. You can get them via `sudo apt-get install ros-kinetic-ifm3d` or `sudo apt-get install ros-melodic-ifm3d`. Again be sure to remove what you currently have installed and clear the ifm3d PPA from your sources.list.

theseankelly, 2020-04-23:
Hey @MYSystemLogistics just wanted to follow up after a couple weeks. Have you had any luck with your PCL dependencies?

MYSystemLogistics, 2020-04-23:
Hi @theseankelly sorry if I haven't replied, but lately I have dedicated myself to something else. As soon as I get back to PCL I will let you know (I hope to give you some news by Monday).

theseankelly, 2020-04-23:
@MYSystemLogistics no rush on my behalf -- I'll be here when you're back on this topic.

theseankelly, 2020-07-30:
HI @MYSystemLogistics it's been a while -- just wanted to follow up on this and see if I can help in any way.

MYSystemLogistics, 2020-07-30:
Hi @theseankelly, I'm sorry for my extremely late reply.
I "solved" the problem running the code on a different environment. Basically I started from scratch. I tried it both on Nvidia Jetson Nano and on a PC with Ubuntu 20, and it worked.
I think that my Ubuntu 18 PC is totally corrupted, maybe to many wrong installation and libraries conflicts that I'm not able to solve.
I forgot to update you, so sorry again.
Anyway, thank you very much for your support.

theseankelly, 2020-08-03:
Interesting -- too bad I wasn't able to get to the root cause on your Ubuntu 18 system, but it's good to hear the issue was solved!

closed 2020-08-03 by theseankelly