ifm/ifm3d #201: I need the library on Windows

swx954068967, 2020-02-25:
I need the library on Windows,but  https://github.com/ifm/ifm3d  :
Windows
Coming soon...

When can you provide it?
Is this compilation scheme correct?It's a little difficult for me.I want to have compiled libraries, and interface documentation, like O3D SDK
https://github.com/ifm/ifm3d/blob/master/doc/windows.md

theseankelly, 2020-02-26:
Hi @swx954068967 -- the "coming soon" string refers to making pre-compiled binaries available. The library does build and run on Windows, and the instructions you linked are the correct instructions.

Regarding the task of auto-generating and shipping pre-compiled binaries, this is on our backlog and I don't have a target date at this time.

swx954068967, 2020-02-28:
Thanks for your reply.
Yes, I need the precompile driver on Windows, just like the 03D SDK.Compiling from source code is a bit difficult and time-consuming for me, and I've tried and failed, due to boost or VS versions.As a product for sale, I think it is necessary to provide precompiled libraries for common platforms. I look forward to it as soon as possible

Arag24, 2020-02-28:
@swx954068967 : if you're interested with precompiled binaries VS 2017 64bit,
Please can you send me your email in PM, please?

swx954068967, 2020-02-29:
> @swx954068967 : if you're interested with precompiled binaries VS 2017 64bit,
> Please can you send me your email in PM, please?

Thank you. My email is swx954068967@live.cn. Sorry for the late reply.

theseankelly, 2020-03-02:
Hey @swx954068967 I've recorded your email and sent to @Arag24 internally. I have removed it from the message above for your privacy.

Arag24, 2020-03-02:
@swx954068967 : Did you de receive, download and test package sir? Do you need something else ? Thanks for feedback.

theseankelly, 2020-03-18:
@swx954068967 my understanding is @Arag24 has provided you the necessary binaries privately -- do reach out again if you require further assistance.

closed 2020-03-18 by theseankelly