ifm/ifm3d #208: Can I modify IFM 03X100 exposure time setting between camera frames?

pjbesl44, 2020-04-28:
Can I modify IFM 03X100 exposure time setting between camera frames? 
How do I do that from the basic 
camera, framegrabber, image interface? 
Do you have any sample code?

in byte_buffer.h I see ExposureTimes( ). which I guessing is the Getxxx Method for the variables.
Is there a Setxxx method?  or do you just access the exposure_times_ std vector directly. 
Thank you for any pointers on this.

theseankelly, 2020-04-28:
Hey @pjbesl44 have a look at this sample: https://github.com/ifm/ifm3d-examples/blob/master/exposure_time/ex-exposure_times.cpp

Does that answer the question?

pjbesl44, 2020-04-28:
Thank you for the quick response!

It's interesting and seems like it would be close, but I compile and link it and it does not work in the
sense that it just returns 3  zero exposure times and it says the illu_temp is 2.93882e+29.

I know from running ifmAssistant that my current exposure time is 1050 us with an ExposureRatio of 10.

In my working code, I added  exposure_times = img->ExposureTimes();
but I also just get 0's for the exposure time.

auto global_ifmCamera       = ifm3d::Camera::MakeShared();
auto global_ifmFrameGrabber = std::make_shared<ifm3d::FrameGrabber>(global_ifmCamera);
auto global_ifmImageData    = std::make_shared<ifm3d::ImageBuffer>();
....
       while(1) {
   if (! global_ifmFrameGrabber->WaitForFrame(global_ifmImageData.get(), 1500))
        {
                std::cerr << "Timeout waiting for IFM O3X100 camera at 192.168.1.69 !" << std::endl;
                return( -1 );
        }
        amp  = global_ifmImageData->AmplitudeImage();
        conf = global_ifmImageData->ConfidenceImage();
        xyz  = global_ifmImageData->XYZImage();

        std::vector<cv::Mat> chans(3);
        cv::split(xyz,chans);
        zc = chans[0]; xc = chans[1]; yc = chans[2];

                       exposure_times = global_ifmImageData->ExposureTimes( );

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D61D71.72628EE0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain(tm)
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

pjbesl44, 2020-04-28:
I should mention I am running on ARM (Jetson Nano).

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D61D7A.174BFD80]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain(tm)
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

[cid:image001.jpg@01D61D7A.174BFD80]<http://www.symbotic.com/>

theseankelly, 2020-04-29:
Hi @pjbesl44 -- apologies; I overlooked the fact that dynamic parameter changing is handled differently on the O3X100 as compared with the O3D3XX cameras. The sample I shared is intended to work with the O3D3XX family of cameras which have a feature allowing a subset of parameters to change on the fly.

Unfortunately the O3X100 does not support this functionality (including the `ifm3d::EXP_TIME` and `ifm3d::ImageBuffer::ExposureTimes()` APIs). The exposure time can be retrieved and configured via JSON and the `ifm3d::Camera::FromJSONStr` API as follows:

```
#include <iostream>
#include <memory>

#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main()
{
  auto c = ifm3d::Camera::MakeShared();
  auto fg = std::make_shared<ifm3d::FrameGrabber>(
    c,
    ifm3d::IMG_CART | ifm3d::EXP_TIME | ifm3d::ILLU_TEMP);
  auto im = std::make_shared<ifm3d::ImageBuffer>();

  // Dump the configuration
  auto j = c->ToJSON();
  std::cout << "Exposure Setting: "
            << j["ifm3d"]["Apps"][0]["Imager"]["ExposureTime"]
            << std::endl;

  // Modify it and re-configure camera
  j["ifm3d"]["Apps"][0]["Imager"]["ExposureTime"] = "1234";
  c->FromJSON(j);

  // Alternatively set directly from a string literal
  std::string json_str = R"(
        {
          "ifm3d":
          {
            "Apps":
            [
              {
                "Imager":
                {
                    "ExposureTime": "1000"
                }
              }
            ]
          }
        }
      )";
  c->FromJSONStr(json_str);

  return 0;
}
```

pjbesl44, 2020-04-29:
Thank you!
Will give that a try.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D61E38.63413CF0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

pjbesl44, 2020-05-04:
Hi Sean,

That did the trick.  I can print to std::cout
and place compiled static strings into the exposure time.

I have a few C++ related questions.
Just curious, what is the type of
  j["ifm3d"]["Apps"][0]["Imager"]["ExposureTime"] in C++?

I don’t seem to be able to treat it as a string in my C code.
I got a long message about it from g++.

Perhaps a dumber question:
More to the point, how would I programmatically
set the value to an integer between say 200 and 2700?
The example only handles compiled static strings, but I’ve been
unable so far to write a string in C that I can place into the JSON object.
Thanks for any pointers.

- Paul

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D621F5.8248C6C0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

theseankelly, 2020-05-04:
Hey @pjbesl44  -- the type of the object is a `std::string`, as in:

```
  std::string exp_time = j["ifm3d"]["Apps"][0]["Imager"]["ExposureTime"];
  std::cout << "Read: " << exp_time << std::endl;
```

Without seeing your compiler error, I might guess you're trying to use c-style strings instead of `std::string`? If so, check this out: https://stackoverflow.com/questions/4764897/converting-a-c-style-string-to-a-c-stdstring

theseankelly, 2020-05-04:
whoops, hit 'close' by accident.

pjbesl44, 2020-05-05:
Thank you very much.  The JSON values are all std::string’s.  I am all set now with that confusion on my part.
I can do everything I want now.

I can open up a separate topic if you like on the following question (really just an observation).

I found that I can change the exposure time while my other primary frame grabbing process is running.
Is it really changing the hardware’s exposure time?  It seems to be.
Is this Ok to do? (ie. Is this the intended functionality?  I like the flexibility just concerned that I am not understanding
something about how the sensor operates.

For example if ifm assistant has the IFM device open, my process cannot get to it.
And if my process is running, then ifm assistant cannot read from the device.  It gets a socket error.
I was just a little surprised that I can use the JSON interface to change the exposure time while
my process is running.  Am I understanding that correctly? Thanks!

- Paul

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D622C0.D66FBA80]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

graugans, 2020-05-05:
@pjbesl44 the O3X1xx can only provide a single connection. The O3D3xx can provide up to 5 simultaneous connections. To consume the data while you change the settings is totally fine.

pjbesl44, 2020-05-05:
That’s great to be able to do that. Thank you guys so much for the quick turnaround on questions.  Very helpful.
I think I am good to go now.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D622C9.07FF21A0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

theseankelly, 2020-05-05:
Sounds good! I'll close this issue now. Feel free to create a new issue/question if something new comes up later!

closed 2020-05-05 by theseankelly