ifm/ifm3d #211: ifm3d-camera 0.17.0 unavailable via ifm-robotics_ubuntu_xenial_amd64 apt repo

jobirobi, 2020-05-13:
**Describe the bug**
While attempting to install ifm3d-tools from ifm-robotics_ubuntu_xenial_amd64 (the no-ROS option), I had an error collecting all of the dependencies. I was able to trace it back to the fact that ifm3d-camera 0.17.0 is missing from that repository. I was able to install ifm3d-camera 0.17.0 from ifm-robotics_ubuntu_xenial_amd64_ros and follow with ifm3d-tools, but I can't be sure that I haven't installed an incompatible package.

**To Reproduce**
(On a machine which has never installed ifm3d-*)
1. Add the apt repository to sources:
    `sudo sh -c 'echo "deb [arch=amd64] https://nexus.ifm.com/repository/ifm-robotics_ubuntu_xenial_amd64 xenial main" > /etc/apt/sources.list.d/ifm-robotics.list'`
2. Add the apt key:
    `sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 8AB59D3A2BD7B692`
3. Install ifm3d-tools:
    `sudo apt-get install ifm3d-tools`

**Expected behavior**
ifm3d-tools 0.17.0 is installed.

**Output/Screenshots**
```$ sudo apt-get install ifm3d-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ifm3d-tools : Depends: ifm3d-camera (= 0.17.0) but it is not going to be installed
               Depends: ifm3d-image (= 0.17.0) but it is not going to be installed
               Depends: ifm3d-swupdater (= 0.17.0) but it is not going to be installed
               Depends: ifm3d-framegrabber (= 0.17.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

$ apt list | grep ifm3d

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ifm3d-camera/xenial 0.15.1 amd64
ifm3d-framegrabber/xenial 0.17.0 amd64
ifm3d-image/xenial 0.17.0 amd64
ifm3d-opencv/xenial 0.17.0 amd64
ifm3d-pcicclient/xenial 0.17.0 amd64
ifm3d-pcl-viewer/xenial 0.1.0 amd64
ifm3d-python/xenial 0.17.0 amd64
ifm3d-python3/xenial 0.17.0 amd64
ifm3d-swupdater/xenial 0.17.0 amd64
ifm3d-tools/xenial 0.17.0 amd64```

Here's my workaround:
```$ wget https://nexus.ifm.com/repository/ifm-robotics_ubuntu_xenial_amd64_ros/pool/i/ifm3d-camera/ifm3d-camera_0.17.0_amd64.deb
$ sudo dpkg -i ifm3d-camera_0.17.0_amd64.deb
$ sudo apt-get install ifm3d-tools```

**Configuration/Environment (please complete the following information):**
 - OS: Ubuntu 16.04.6
 -  ifm3d version: 0.17.0
 - using ROS: no

theseankelly, 2020-05-14:
Yes it looks like I botched something. My apologies; I'll have this corrected today.

theseankelly, 2020-05-14:
Thanks for the detailed report and workaround -- I've re-posted the 0.17.0 binary to the apt repository. Confirmed that I could reproduce before but not after. Should be fixed now.

closed 2020-05-14 by theseankelly