ifm/ifm3d #214: Anybody running O3X100 code on Raspberry Pi 4 Raspian Buster?

pjbesl44, 2020-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

theseankelly, 2020-05-20:
Hey @pjbesl44 I have not tried to compile or run ifm3d on any Raspberry Pi platform. These platforms are not typically suited for easy use with computer vision workflows, but there are some great tutorials on how to get started. For example, here's one on what it takes to install OpenCV (just one dependency of ifm3d) on a Raspberry Pi 4: https://www.pyimagesearch.com/2018/09/26/install-opencv-4-on-your-raspberry-pi/

In theory it ought to work but the raspberry pi family isn't an officially supported platform for ifm3d at this time.

pjbesl44, 2020-05-21:
So I built  ifm3d 0.17.0 on Raspian Buster after installing OpenCV4.3 
some of the make check messages come back with OK but most fail.
The ifm3d executable runs but with the exception of 'ifm3d schema'
it cannot contact the sensor.  Says the usual 'can you ping the sensor?'
Yes I can ping the sensor. 

This might seem not mainstream, so I tried installing ifm3d on a newer Jetson Nano. 
The old Jetson Nano has OpenCV 3.3 installed.  everything has been working great on that
Dev Kit system.  ifm3d gives a message about that the OpenCV3.3 might have conflicts with OpenCV3.2 that ifm3d prefers (followed the ARM Ubuntu instructions).

However, the new Jetson Nano only has OpenCV 4.1. installed.
Went thru the exact same ifm3d install procedure from github that I did for the OpenCV 3.3 system.
Does not work. I also get the 'Can you ping the sensor? Message.
And yes I can ping the sensor. 

So now I've got the following system situations with ifm3d 0.17.0.
Old Nano Dev Kit with OpenCV 3.3 : mostly OK.  Have separate issue to file.
New Nano Dev Kit with OpenCV 4.1:  does not work.
RPi4B compiled ifm3d from source with OpenCV 4.3:  does not work.

Could you comment?  If I need to use OpenCV 3.2 and only OpenCV 3.2 I can do that. Just need to know.

theseankelly, 2020-05-21:
Hey @pjbesl44 

For the networking issue can you share the following:

1. The output of `ifconfig`from a terminal
2. The IP address you have assigned to your camera
3. Does `ping 8.8.8.8` work?
4. The output of `curl x.x.x.x` from a terminal, where x.x.x.x is the IP address assigned to your camera

-----

For the OpenCV issue -- There's no specific OpenCV requirement in ifm3d -- we support OpenCV 2.x through 4.x. 

I assume you are installing ifm3d via the apt repository I previously created? The system version of OpenCV when those binaries were built was version 3.2, and so the precompiled binaries link against (and expect) OpenCV 3.2. On Ubuntu, system-wide dependencies tend to lock in a certain version per release. It looks like Linux for Tegra, despite being Ubuntu based, takes certain liberties to push libraries (such as OpenCV) to newer versions than what's available in the core Ubuntu distribution. 

For example, OpenCV 3.2 is available from the Ubuntu distribution but Linux for Tegra overrides this with a newer version. This newer version is presumably optimized for the Tegra platform so you should use it.
```
uskellse@uskellse-tx2:~$ apt-cache policy libopencv-dev
libopencv-dev:
  Installed: 4.1.1-2-gd5a58aa75
  Candidate: 4.1.1-2-gd5a58aa75
  Version table:
 *** 4.1.1-2-gd5a58aa75 500
        500 https://repo.download.nvidia.com/jetson/common r32/main arm64 Packages
        100 /var/lib/dpkg/status
     3.2.0+dfsg-4ubuntu0.1 500
        500 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 Packages
        500 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 Packages
     3.2.0+dfsg-4build2 500
        500 http://ports.ubuntu.com/ubuntu-ports bionic/universe arm64 Packages
```

So it looks like in order to properly supply binaries for Tegra based platforms I need to create individual repositories that are tied to the JetPack version being used (which version of JetPack are you targeting for your application?) I will look into this but the infrastructure to set it up will take a few days. In the meantime, this topic of library compatibility between shared libraries is the reason by most C++ projects, including ifm3d, advise to be built from source when possible.

pjbesl44, 2020-06-02:
The output of ifconfigfrom a terminal below: 
New NVIDIA Jetson Nano at 192.168.1.189   that does NOT work with IFM
Old NVIDIA Jetson Nano at 192.168.1.238  that works GREAT with IFM O3X100
The IP address you have assigned to your camera: 192.168.1.69
Does ping 8.8.8.8 work? Yes
The output of curl x.x.x.x from a terminal, where x.x.x.x is the IP address assigned to your camera
curl 192.168.1.69
<html><body><h1>It works!</h1></body></html>

New Jetson that does not work:ep3avatar@ep3avatar-jnano:~$ ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:04:4b:ec:1d:66  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 151  base 0xf000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 3008  bytes 199789 (199.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3008  bytes 199789 (199.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

rndis0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 4e:98:89:ab:8b:75  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 4e:98:89:ab:8b:77  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.189  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::25b2:a28:2634:1f95  prefixlen 64  scopeid 0x20<link>
        ether 00:c1:41:06:0e:2b  txqueuelen 1000  (Ethernet)
        RX packets 12040  bytes 9043065 (9.0 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7448  bytes 1699133 (1.6 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Hope this helps. 
This information was in the other emal in the attachment.

pjbesl44, 2020-06-02:
btw, tried your new version 0.18.0 install procedure with both the Jetpack 4.3 and 4.4. Still same error. 
Summary:
**PC running IFM Vision assistant at 192.168.1.244**
netmask 255.255.255.0 
Default gateway: 192.168.1.1
Has No Problem contacting IFM at 192.168.1.69

**Older Jetson unit: 192.168.1.238 has no problem talking to IFM at 192.168.1.69**

two new Jetson units with latest jetpack. 
install of ifm3d software has no problem. 
Yet 
ifm3d version
0.18.0
ifm3d ls 
'can you ping the sensor?

pjbesl44, 2020-06-02:
In case I did not say it:  All 4 Systems: PC and 3 Nanos:  all can 'ping 192.168.1.69' successfully.
curl 192.168.1.69   says "It works!" on all 4 systems. 
Makes no sense to me. 
Do you have a Nano to test this on?

theseankelly, 2020-06-02:
Hey @pjbesl44 -- you're using a non-default IP address so when you run the `ifm3d` commands, it's trying to contact 192.168.0.69 instead of 192.168.1.69. Two options:

1. Specify the IP address in the command, i.e. `ifm3d --ip=192.168.1.69`
2. Set the `IFM3D_IP` environment variable, i.e. `export IFM3D_IP=192.168.1.69`, followed by regular commands i.e. `ifm3d ls`

Try 'ifm3d --ip=192.168.1.69 ls'

pjbesl44, 2020-06-02:
Thanks, Sean.  I seem to have added that to my .bashrc file back in March (when I was reading instructions for a change) and totally forgot all about it.  Won't forget that again. Moreover, everything is working on the RPi4 now too even based on my compiling everything from source.  Turns out my app runs a bit faster on RPi4 than the Nano.  Interesting development.  Need to go back and check my "compiled from source" ifm3d on another Nano to see if that one worked too. 
Very much appreciate your help and patience.  - Paul

pjbesl44, 2020-06-02:
Responded using web interface. This might duplicate response.

  1.  That was it. I had forgotten I put that in my .bashrc file.
  2.  That fixed my ifm3d on the RPi4 also.
Thanks for your help and your patience.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D638E6.D1E2BEE0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

theseankelly, 2020-06-02:
Sure thing, @pjbesl44 glad we were able to get to the bottom of it.

commit f8e4313d (2023-03-06): Merge branch '214-ods-application-framegrabber-start' into 'main'

Add ODS chunks to the schema generator

Closes #214

See merge request syntron/support/csr/ifm3d/ifm3d!258
closed 2020-06-02 by theseankelly