ifm/ifm3d #217: I want to access 2 USB3 Cameras on same system accessing IFM O3X100.

pjbesl44, 2020-05-21:
I want to access 2 USB3 Cameras on same system accessing IFM O3X100.
I am using Jetson Nano Dev Kit with OpenCV 3.3 installed. 
If I run a 2 camera capture loop in a separate executable while running my IFM executable, they can run for 3 days without mishap. 
If I integrate my OpenCV code in the same executable so that the code does an IFM grab followed by 2 USB camera grabs, this will run for hours but usually fails if running overnight. One time it ran for 36 hours before the hang.  The IFM O3X seems like it's OK but I get 2 "select timeout" errors, one for each camera.  Why would the same code not be robust in the IFM executable when it runs fine in standalone OpenCV executable?  

I also have repeatable circumstances (very rare) where the IFM grab returns "junk" meaning the XYZ data is 36 meters away and the amplitude image was zeros.  The documentable cases where this has occurred take longer than 12 hours to show up.  Moreover, you can run the same executable for 36 hours and it won't show up. It's spotty. 

I would like to like to try using Ifm3d with the OpenCV connection to reduce dependence on other packages, but the first framegrab source I found used OpenCV frame grab loop to get the data. 
Do you have a template simple framesource without using OpenCV?  Thank you.

- Paul

theseankelly, 2020-05-22:
I can't speak to the stability of your system when merging implementations. I'm not sure what the "select timeout" is, sounds like it's for the two USB cameras or is the error from ifm3d? Please share the `ifm3d` log output files at /tmp/ifm3d.INFO.

Thanks for the report about the 'junk' data. I'll keep an eye on it. Does it happen for your standalone executable or only when running the problematic code?

Our OpenCV dependency is just on the cv::Mat datatype, nothing more. Shouldn't be much to go on there. Nonethless we have a [simple image container](https://github.com/ifm/ifm3d-examples/tree/master/simpleimage) example which shows how to implement a custom ImageBuffer using stl data structures instead of OpenCV. Does this help?

theseankelly, 2020-06-02:
Hey @pjbesl44 are you still hitting this issue?

pjbesl44, 2020-06-03:
So you can ignore the 'Select timeout' issue above. 
I am attempting converting over to the simpleimage_buffer example.
I am getting amplitude and confidence fine this way.  It matches the OpenCV implementation. 

When I try using XYZImage( ) in the simpleimage_buffer, I am NOT getting the same x,y,z data. 
With OpenCV I could access the coordinates inside … xyz = img->XYZImage();  using a simple
cast    float *xyzPtr = (float*)xyz.data ;
This does not work with simpleimage_buffer.h  
Tried PointCloud option and that did not work either.

The example code does not actually access the xyz data even though it specifies ifm3d::IMG_CART.
If I use simpleimage_buffer framework, how do I get the x,y,z coordinates as floats in meters (or millimeters)?  Thx as always for your help.

pjbesl44, 2020-06-03:
Trying to follow the simple image buffer example on #217.
Still hitting a snag.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D6391A.BA3F93B0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

theseankelly, 2020-06-17:
Hey @pjbesl44 can I close out this issue now that you're unblocked on the simple image buffer/

pjbesl44, 2020-06-18:
I don't know the strikethrough happened.
pbesl@pbesl-desktop:~/ifm3d-tmp-errors$ more ifm3d_summary_for_tsk.txt
::::::::::::::
ifm3d.ERROR
::::::::::::::
Log file created at: 2020/05/22 02:54:51
Running on machine: pbesl-desktop
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
E0522 02:54:51.560717 28158 camera_impl.hpp:259] http://192.168.1.69:80/api/rpc/v1/com.ifm.efector/->getParameter
:Unable to transport XML to server and get X
ML response back.  HTTP response code is 500, not 200
E0522 02:54:51.563212 28158 camera_impl.hpp:259] http://192.168.1.69:80/api/rpc/v1/com.ifm.efector/->getParameter
:Unable to transport XML to server and get X
ML response back.  HTTP response code is 500, not 200
::::::::::::::
ifm3d.INFO
::::::::::::::
Log file created at: 2020/05/22 07:33:07
Running on machine: pbesl-desktop
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
W0522 07:33:07.769934 13733 camera.cpp:776] Skipping Imager, null value -- should be string!
W0522 07:33:07.771755 13733 camera.cpp:776] Skipping SpatialFilter, null value -- should be string!
W0522 07:33:07.771899 13733 camera.cpp:776] Skipping TemporalFilter, null value -- should be string!
I0522 07:33:07.811168 13733 frame_grabber_impl.hpp:178] Camera connection info: ip=192.168.1.69, port=50010
W0522 07:33:07.813452 13761 frame_grabber_impl.hpp:588] Connection refused
I0522 07:33:07.813899 13761 frame_grabber_impl.hpp:596] FrameGrabber thread done.
::::::::::::::
ifm3d.WARNING
::::::::::::::
Log file created at: 2020/05/22 07:33:07
Running on machine: pbesl-desktop
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
W0522 07:33:07.769934 13733 camera.cpp:776] Skipping Imager, null value -- should be string!
W0522 07:33:07.771755 13733 camera.cpp:776] Skipping SpatialFilter, null value -- should be string!
W0522 07:33:07.771899 13733 camera.cpp:776] Skipping TemporalFilter, null value -- should be string!
W0522 07:33:07.813452 13761 frame_grabber_impl.hpp:588] Connection refused
pbesl@pbesl-desktop:~/ifm3d-tmp-errors$

theseankelly, 2020-06-18:
Hey Paul -- the error output looks fine to me, except that it's indicating connection to the camera is failing. I'm guessing this is an old log and we're passed the connection issues?

pjbesl44, 2020-06-19:
No, here’s my errors, info, warning messages from today.
I get these messages consistently.

Never seem to have had any effect but may be a performance issue that I never noticed?

The discovery python script works great.
So there are socket comms you can do to get the info we need it looks like.
Is there is C version of this by chance?
Looks like I could write one from python source actions.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D645B5.8D489670]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

theseankelly, 2020-06-23:
Hey @pjbesl44 was something supposed to be attached/included to that email? I didn't get anything. I'm not sure attachements work when responding to the github issue via email.

pjbesl44, 2020-07-01:
FYI: Those messages are triggered by these lines of code with O3X100. 

  auto cam = ifm3d::Camera::MakeShared( );
  auto fg = std::make_shared<ifm3d::FrameGrabber>(
    	    cam, ifm3d::IMG_CART|ifm3d::EXP_TIME|ifm3d::IMG_AMP|ifm3d::ILLU_TEMP);

I delete all the ifm3d.* files in /tmp.

Then I run a c++ program with these 2 lines. 

Then ifm3d.INFO and ifm3d.WARNING show up in /tmp.

So the MakeShared method with no arguments and then using it 
with the framegrabber   generates an ifm3d.INFO file and an ifm3d.WARNING file. 

Happens everything single time even if I do nothing else with the sensor..

pjbesl44, 2020-07-01:
Log file created at: 2020/06/30 20:37:20
Running on machine: pbesl-desktop
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
W0630 20:37:20.431402 25746 camera.cpp:776] Skipping Imager, null value -- should be string!
W0630 20:37:20.434231 25746 camera.cpp:776] Skipping SpatialFilter, null value -- should be string!
W0630 20:37:20.434486 25746 camera.cpp:776] Skipping TemporalFilter, null value -- should be string!
I0630 20:37:20.474550 25746 frame_grabber_impl.hpp:178] Camera connection info: ip=192.168.1.245, port=50010
I0630 20:37:21.975682 25747 frame_grabber_impl.hpp:596] FrameGrabber thread done.

pjbesl44, 2020-07-01:
Log file created at: 2020/06/30 20:37:20
Running on machine: pbesl-desktop
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
W0630 20:37:20.431402 25746 camera.cpp:776] Skipping Imager, null value -- should be string!
W0630 20:37:20.434231 25746 camera.cpp:776] Skipping SpatialFilter, null value -- should be string!
W0630 20:37:20.434486 25746 camera.cpp:776] Skipping TemporalFilter, null value -- should be string!

theseankelly, 2020-07-02:
Hey Paul -- thanks for the info. I can reproduce and understand why the warnings are showing -- they're harmless and don't impact operation.

pjbesl44, 2020-07-02:
Thanks for checking.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D65090.4BDB1510]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

closed 2020-07-30 by theseankelly