ifm/ifm3d #223: How do you get float x,y,z coordinates in meters or mm using the SimpleImageBuffer example.

pjbesl44, 2020-06-04:
Hopefully this is a simple C++ question.  I was directed to the SimpleImageBuffer example that shows a way to write PPM file data from the IFM sensor without using OpenCV or PCL.  I've tried to access X,Y,Z 3D floating point Cartesian coordinates (in millimeters or meters) using the SimpleImageBuffer object.  When the image is a cv::Mat object, I can get float coordinates easily using cv::Mat xyz = img->XYZImage() and 
float *xyzPtr = (float *)xyz.data. . When the image is SimpleImageBuffer object, I can only get integers out of the example object code, which do not appear to me to be 3d coordinates.  So I've attached 2 example programs which are very very similar, but I am missing something about how to get floating point XYZ data out of the SimpleImageBuffer object. 
I tried to use the Cloud() method with no luck. I tried to use the PointCloud object with no luck. 

[whatswrong2.tar.gz](https://github.com/ifm/ifm3d/files/4732036/whatswrong2.tar.gz)

Hoping this is as easy to resolve  as forgetting to set the IFM3D_IP env var. 
As always, your help is much appreciated.

theseankelly, 2020-06-09:
Hey @pjbesl44, when you write `float* xyzPtr = (float *)xyz.data` you're attempting to cast a `std::vector<uint8_t>` into a `float *` -- this won't work and is likely the cause of your problems. I recommend reading through the implementation of the `SimpleImageBuffer` to understand the data structures and types being used to represent the data. (Also sidenote -- using c-style casts is highly discouraged -- the compiler can't flag any of these problems for you)

FWIW the `Cloud()` method will return a structure called `PointCloud` which contains a `std::vector<Point> pots` member, where `Point` is a struct of three floats x y and z. I think this is what you want to use, but you have to adapt your client code accordingly as this is not a `cv::Mat` datastructure any longer.

Final note -- the samples are intended to be basic ideas and examples of how you might use `ifm3d` in your application. They are not necessarily tested and maintained to production-level quality. You should expect to modify and customize them according to your needs. If you're not comfortable with designing and maintaining a custom data structure for your application, I encourage you to stick with the tried-and-true `cv::Mat` from OpenCV.

pjbesl44, 2020-06-09:
I hope you are recovered from those storms passing thru.

I am going to express a perhaps archaic and an possibly uninformed position that you may or may not find in your other customers.
You guys have an xyz[172][224][3] float array in your sensor.
I am just asking to be able to query/grab  that with a time stamp using the C programming language.

OpenCV is a great achievement of open source and C++ and I understand everyone uses it.
But it is a huge cumbersome and constantly changing open source software which develops incompatibilities with other camera digitizers.
I need to be able to support this IFM sensor in production for 7 years with 2 video cameras.
You guys may support things but there must be some way to help a customer avoid potential problems in the future.

It takes me around 6 hours to compile OpenCV from source on a new Nano system.
I am in a phase right now between full production and development.
Simplicity would be helpful.

Has no one ever asked for a C interface to your sensor before?

Can you at least explain what I do get when I ask for the XYZImage()?
Can you document for me what bytes I get back?

Can you provide a sample code that uses  Cloud with SimpleImageBuffer and compiles?

The reason I have not used Cloud yet is because I tried 5 different ways to get to it and none of them compiled.
How are those bytes converted to XYZ coordinates?
I don’t see any mathematics in your SimpleImageBuffer example that gives me anything to go on.

There’s a difference between production worthy and works at all.

I’ve sent very simple examples. It seems to me either the answer to my question is simple or you don’t have an answer you can give me.
i.e. you are not going to support that.

When I follow the instructions that you guys support on ARM (and I thank you for that)  I get OpenCV3.2 code.
I’ve installed 4.1 and 4.3?  I get an incompatibility with the video frame grabber code that interferes with me digitizing video camera signals.
In addition the video grabbing in OpenCV does not seem to be very prompt at delivering up to date video frames when used in conjunction with
the IFM frame grabbing.

I am still confused at your response.
I do have something that works now.
Could you please answer the middle question in yellow above?
Thank you.
- Paul

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D63E43.D82913F0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

pjbesl44, 2020-06-09:
I am comfortable developing and maintaining my application if you can get me more information on the byte data that is returned by XYZImage().

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D63E44.3C4B14A0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

theseankelly, 2020-06-09:
I think I didn't communicate this topic well to you -- my goal in pointing you at the SimpleImage example was to show how you might design and implement your own image container to avoid OpenCV. The actual SimpleImage example itself isn't subject to unit testing or anything like that by ifm -- customers are free to take, use and otherwise manipulate these samples to adapt to their workflows. But, ifm won't be able to provide detailed support on the image container if you go that route.

Here are my answers to your various questions:

---------------------------------------------------

For the XYZ image attempt you have this line:
```
//float *xyzPtr = (float *)xyz.data; // this does not work w/simpleImage
```
Indeed this won't work since `xyz` is of type `ifm3d::SimpleImageBuffer` and doesn't have a `data` member. You'd need to traverse the `std::vector` and individually convert each pixel from  `std::uint8_t` (mm) to `float` (m). This is where it might be interesting to develop your own image container that encodes Images directly as floating point values if you wish.

---------------------------------------------------

For the Cloud image, you have this line:
```
//ifm3d::SimpleImageBuffer::PointCloud zxy = cld->Cloud(); // no good
```
That's because `cld` is type `ifm3d::SimpleImageBuffer::PointCloud` -- the line should read:
```
ifm3d::SimpleImageBuffer::PointCloud zxy = img->Cloud();
```
However when I changed this I did discover our sample is missing the implementation of `ifm3d::SimpleImageBuffer::PointCloud::Cloud()` and I just [pushed a commit](https://github.com/ifm/ifm3d-examples/commit/01a8d2a79279480025fab494c5af9d9954fa6bed) which adds it. Perhaps this is one of the permutations you tried already which failed. Hopefully that helps.

---------------------------------------------------

We do not have a C library for the camera interface at this time.

---------------------------------------------------

Did you see the latest instructions with the 0.18.0 release from last week? I've created specific apt repositories for L4T/JetPack systems like the Nano. NVIDIA ships a GPU optimized version of OpenCV as part of JetPack and thus there's no need to compile from source. The binaries on the L4T apt repositories are built targeting that version of OpenCV.

pjbesl44, 2020-06-09:
Excellent! Thank you!

You hit the problem I hit:

ifm3d::SimpleImageBuffer::PointCloud zxy = img->Cloud();

However when I changed this I did discover our sample is missing the implementation of ifm3d::SimpleImageBuffer::PointCloud::Cloud() and I just pushed a commit<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_ifm_ifm3d-2Dexamples_commit_01a8d2a79279480025fab494c5af9d9954fa6bed&d=DwMCaQ&c=Q1CA-4L9RDYyn3rai5IPmQ&r=mB-jD-lHhNgjxG7QbVgUFuOiBbRqYEgiRvcdpijwJiw&m=dr2deGV63OqJMll-tQu2Yhv86gQoPa90ZrLaHOYA-Hg&s=zQ8kKDg8TAEvoPGh-KcrZH3makIvYUJSE1sZoo7dnhM&e=> which adds it. Perhaps this is one of the permutations you tried already which failed. Hopefully that helps.  >>>> YES!

Will let you know how it goes, but the fact that you saw that problem is very good.  That’s what I was looking for.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D63E53.7FA5AC60]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

pjbesl44, 2020-06-09:
I had a feeling it was something simple missing.  Those 6 lines of boilerplate C++  fixed all the problems I had.  Thank you.  I will try to do a better job with my communication as well. It's difficult to communicate all 10 things I tried and the compiler errors or bad data errors I got. 

There's no timing or performance difference in my initial tests grabbing 1000 frames. 
Only thing I noticed is the row, column parameterization seems to be opposite but 
that's not important to me at this point.  You can close out this item.

theseankelly, 2020-06-09:
No problem, and it's equally difficult for me to communicate solutions sometimes. Either way glad we got to the bottom of it -- let me know if anything else crops up.

commit 48b753df (2023-04-05): Merge branch '223-python-doc-is-missing-ifm3dpy-device-error' into 'main'

Resolve "Python doc is missing ifm3dpy.device.Error"

Closes #223

See merge request syntron/support/csr/ifm3d/ifm3d!282
closed 2020-06-09 by theseankelly