ifm/ifm3d #228: Is there any example of pallet detection?

GoodTekken, 2020-06-15:
Hello, 
    Is there any example of pallet detection? Thank you.

[https://www.ifm.com/us/en/us/robotics/software-solutions/pds/integrating-pds/integrating-pds](https://www.ifm.com/us/en/us/robotics/software-solutions/pds/integrating-pds/integrating-pds)
[https://www.ifm.com/us/en/us/robotics/software-solutions/pds/pds](https://www.ifm.com/us/en/us/robotics/software-solutions/pds/pds)

theseankelly, 2020-06-15:
Hey @GoodTekken -- our Pallet Detection System (PDS) solution itself proprietary. We have a few client interfaces (ROS, Python, C++) which we provide source to, but they're not on GitHub at the moment. Could you send me an email at sean.kelly@ifm.com?

closed 2020-06-17 by theseankelly

DUANlaoda, 2021-08-06:
Could you send me the Python or C++ source of Pallet Detection System (PDS) , duanfxld@gmail.com , thanks

graugans, 2021-08-06:
@DUANlaoda please get in touch with @GarrettPlace

GarrettPlace, 2021-08-09:
@DUANlaoda - you can reach me at garrett.place@ifm.com

robologics1, 2024-06-25:
I would like to get source of Pallet Detection System, tlg.kayin@gmail.com , thanks

graugans, 2024-06-25:
> I would like to get source of Pallet Detection System, [tlg.kayin@gmail.com](mailto:tlg.kayin@gmail.com) , thanks

As stated by Sean the Solution itself is closed source, the interface is available please reach out to Garrett his contact is available in 
 the comment https://github.com/ifm/ifm3d/issues/228#issuecomment-895175296