ifm/ifm3d #229: Several Network questions: Default hostname of IFM3D O3X100 ?

pjbesl44, 2020-06-18:
I was quite happy with the Static IP method of using the IFM3D sensor on a Class C IPv4 network.
However, I am now asked to have the IFM as a DHCP node on a Class B IPv4 network. 
I've been asked 
"Do the IFM sensors have a default hostname?"
"Can one set the hostname of an O3X100 sensor? and If so how? 

How do I detect what DHCP address was given to a particular IFM sensor on Class C IPv4 network?

theseankelly, 2020-06-18:
The default hostname on my O3X100 is 'O3D3XX-41-1F-56` where the three octets correspond to the MAC address of my device. I suspect this cannot be changed (@graugans knows more and can chime in if I'm wrong) and also means this probably won't really work for writing generic software.

Regarding finding the IP address, we recently released some python code to do network discovery of IFM cameras: https://github.com/ifm/python-ipv4-discovery and we're currently working on integrating this into the core `ifm3d` library over the next several weeks (see issue #194).

Can you use a static IP that's on the same class B subnet, but outside the assignable address range of your DHCP server?

pjbesl44, 2020-06-19:
My network guy said that static IP’s in the Class B range outside the DHCP range is a possibility.

graugans, 2020-06-22:
Hey @theseankelly, you are right the hostname is O3X1xx plus the last three octets of the MAC address. This can not be changed at the moment. What you can try is using avahi to discover your devices: https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#dhcp-enabled-device-on-dhcp-network

pjbesl44, 2020-06-22:
Clarification:  on O3X100, the hostname shows as  O3D3XX-41-BD-69

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D64877.1E041360]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

closed 2020-07-30 by theseankelly