ifm/ifm3d #230: FYI: Using the Jetpack 4.4 instructions leads to some error messages. NVIDIA's problem?  Does not stop camera from working.

pjbesl44, 2020-06-20:
Using 32.4.2  Jetpack 4.4 from sd-blob-b01.img
wondering if you get that too.

botuser@vis00003:~$ sudo apt-get update
Get:1 file:/var/cuda-repo-10-2-local-10.2.89  InRelease
Ign:1 file:/var/cuda-repo-10-2-local-10.2.89  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease         
Get:5 file:/var/cuda-repo-10-2-local-10.2.89  Release [574 B]
Get:6 file:/var/visionworks-repo  Release [2,001 B]                                                  
Get:7 file:/var/visionworks-sfm-repo  Release [2,005 B]                                                                              
Get:5 file:/var/cuda-repo-10-2-local-10.2.89  Release [574 B]                                             
Get:8 file:/var/visionworks-tracking-repo  Release [2,010 B]                                                                       
Get:6 file:/var/visionworks-repo  Release [2,001 B]                                                                                    
Get:7 file:/var/visionworks-sfm-repo  Release [2,005 B]                                                                                                          
Get:8 file:/var/visionworks-tracking-repo  Release [2,010 B]                                                                                                                       
Hit:9 https://repo.download.nvidia.com/jetson/common r32.4 InRelease                                                                                                               
Get:10 https://repo.download.nvidia.com/jetson/t210 r32.4 InRelease [2,559 B]                                                                                    
Hit:11 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                                                                       
Hit:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease                           
Hit:14 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease                          
Hit:15 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease                           
Get:17 https://nexus.ifm.com/repository/ifm-robotics_l4t_jetpack_4_4_arm64 melodic InRelease [1,558 B]
Get:20 https://repo.download.nvidia.com/jetson/t210 r32.4/main arm64 Packages [8,232 B]
Err:20 https://repo.download.nvidia.com/jetson/t210 r32.4/main arm64 Packages
  File has unexpected size (8479 != 8232). Mirror sync in progress? [IP: 23.36.39.139 443]
  Hashes of expected file:
   - Filesize:8232 [weak]
   - SHA256:a6f3d01f9fd0b6ac4318633fe1b049f3524c3e9bc36cdda2bc0ff693570470b5
   - SHA1:e7e34ada6a048427d083ae673a58c2326193ab40 [weak]
   - MD5Sum:037995b9e01d2e8a4621173482b95ac4 [weak]
  Release file created at: Tue, 28 Apr 2020 06:22:43 +0000
Get:21 https://nexus.ifm.com/repository/ifm-robotics_l4t_jetpack_4_4_arm64 melodic/main arm64 Packages [1,867 B]
Fetched 5,984 B in 3s (2,102 B/s)
Reading package lists... Done
E: Failed to fetch https://repo.download.nvidia.com/jetson/t210/dists/r32.4/main/binary-arm64/Packages.gz  File has unexpected size (8479 != 8232). Mirror sync in progress? [IP: 23.36.39.139 443]
   Hashes of expected file:
    - Filesize:8232 [weak]
    - SHA256:a6f3d01f9fd0b6ac4318633fe1b049f3524c3e9bc36cdda2bc0ff693570470b5
    - SHA1:e7e34ada6a048427d083ae673a58c2326193ab40 [weak]
    - MD5Sum:037995b9e01d2e8a4621173482b95ac4 [weak]
   Release file created at: Tue, 28 Apr 2020 06:22:43 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
botuser@vis00003:~$

graugans, 2020-06-22:
Hey @pjbesl44,

please try the following:

```
apt-get clean
apt-get autoclean
```
This fixed a similar issue at my side.

pjbesl44, 2020-06-22:
Thanks, Christian.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D64874.C0081650]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

commit fb7d2f00 (2023-03-31): Merge branch '230-framegrabber-crashes-when-p-command-is-called-multiple-times' into 'main'

Fixed a crash in FrameGrabber when the p command is called multiple times

Closes #230

See merge request syntron/support/csr/ifm3d/ifm3d!279
closed 2020-06-23 by theseankelly