ifm/ifm3d #231: Add ' $ sudo apt-get install libgoogle-glog-dev ' to the L4T instructions for Jetpack 4.4

pjbesl44, 2020-06-20:
You can't build or run app without glog which I think you probably already know. 
Trying to complete my "download option" instructions.

theseankelly, 2020-06-22:
Thanks @pjbesl44 -- I'll check it out. If the library has an external dependency on `libgoogle-glog-dev` (as opposed to the regular `libgoogle-glog`) but that package isn't being listed in the debian dependencies, then there's likely a bug in the CMake target export (probably markeed some

That said, the workaround you've found works great.

pjbesl44, 2020-06-22:
> CAUTION: This email originated from outside of the Symbotic organization. Do not click links or open attachments unless you recognize the sender and know the content. Please report any suspicious emails to Symbotic IT Helpdesk.

I may not have stated the conditions correctly.

If I prepare a Nano L4T system with the exact specifications that are on github.com/ifm/ifm3d,
I cannot link my app.

At this point, I can leave out   ifm3d-opencv  ifm3d-image ifm3d-pciclient
And if I have glog I can link.  If I don’t have it I cannot link (Jetpack 4.4  L4T  32.4.2)

Because I am linking my app on the system, I used your source instructions to use libgoogle-glog-dev.
I can try libgoogle-glog   without the dev to see what happens.

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D64874.7ED73E90]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

pjbesl44, 2020-06-23:
btw, if one does   sudo  apt-get install libgoogle-glog 
you'll get a 
Unable to locate package libgoogle-glog 
 message.

theseankelly, 2020-07-30:
I'm closing this as 'wontfix' because we have plans to move from glog to sdplog in the near future, which has no library dependencies. See #203

closed 2020-07-30 by theseankelly