ifm/ifm3d #232: Depth/Pointcloud on high reflective items

ShettyHarapanahalli, 2020-06-23:
How effective is the depth estimation for the "high reflective" devices like mirrors, stainless steel components, components sprayed with water, or other fluids?  
Please point me to relevant studies, experiments.

GreNait, 2020-06-23:
Hi  @ShettyHarapanahalli,

this is hard to answer with a general statement. Still, I will try my best ;)

Mirrors:
You have no chance to detect them reliable. Why? Because a mirror is designed to reflect light as good as possible. This leads to the problem, that the incoming light is reflected with the same angle outwards. In this case, the camera might see the ceiling, or even objects behind the camera. 

Stainless steel:
This depends highly on how good the steel is reflecting light. Actually, if you want to see a stainless steel object, that might work out just fine. But if you have stainless steel as "walls" etc. this can be a problem. Why? Because the walls reflect the light and will cause so called MPI (multi path issue). In this case, light directly from the camera and light bouncing of the wall will be mixed and causing wrong measurements. Hard to say how big this influence is. Normally, this is a problem if you have objects nearer (<2m) to the camera.

Water, other fluids:
Depending on the material which water is sprayed on, this can be a problem. Concrete is getting like a mirror with some water on it. This can be a challenge. Bright objects might be over exposed. 

We see all 3 effects happening with the camera. MPI is the biggest issue, because you cannot easily compensate for it. The other two effects, might be less of a problem. Some good filter magic and your good to go. 

I hope this is making sense :)

theseankelly, 2020-07-30:
Hi @ShettyHarapanahalli -- have your questions been answered sufficiently?

closed 2022-11-30 by github-actions[bot]