ifm/ifm3d #233: 03D303 connection issues

Wanru-He, 2020-06-24:
Hi there. I cannot connect my camera to my ubuntu 20.04 system when typing "ifm3d ls" into command prompt. I have tried to ping the camera but it shows nothing beyond "xxx bytes of data received". I also tried finding the camera on angry ip scan, but the host seems to be dead. Is there a way I can resolve this? Thanks.

Wanru-He, 2020-06-24:
I could connect to the ifm vision assistant on windows, but i cannot connect to the sdk.

graugans, 2020-06-24:
Hey @Wanru-He have you already checked this troubleshooting guide: https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md

Wanru-He, 2020-06-24:
> Hey @Wanru-He have you already checked this troubleshooting guide: https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md

Yeah, I have tried the methods on the guide, but there is still no information shown after I ping the camera besides the above message.

graugans, 2020-06-24:
You are sure that both your Linux box and the camera are on the same IP network? You may experience issues with your host set-up to DHCP and you have set-up a point to point connection with the camera.

Wanru-He, 2020-06-25:
Hi, I have manually reset the ip for the camera on the windows visual application. I can connect to my linux system now with my new ip address. Thanks for the help.

closed 2020-06-25 by theseankelly

zener90818, 2020-07-22:
@Wanru-He , I meet the same error, can you help me? https://github.com/ifm/ifm3d/issues/241