ifm/ifm3d #235: Limiting number of simultaneous connections

VitoTringolo, 2020-07-01:
I noticed that actually up to 5 clients can be connected to the camera, I am wondering if there is a way to limiting this to 1?

graugans, 2020-07-01:
Hey @VitoTringolo, can you please provide more details what exactly you want to achieve? I expect that this is some sort of a security question? Maybe we can find another way to solve this.

VitoTringolo, 2020-07-01:
I just want to make possible to use the camera only with a client at time. I use the camera to drive robots and I want to be sure that exactly one client can own the camera.

I don't know if you need some more details.

theseankelly, 2020-07-02:
There can be up to five clients connected for the purposes of receiving data -- this is how we can have applications consuming data concurrently with viewer tools for example.

Of those connections, modifications to the camera settings are wrapped in what we call an 'edit session' which provides mutex-like access to the device and imager settings such that no two clients can 'clobber' each other. Furthermore these edit sessions can be password protected so that only authorized parties can manipulate device settings.

So, we won't be able to restrict the number of connections but there are a couple mechanisms in place to restrict concurrent access to device settings. Does that help what you're trying to achieve?

graugans, 2020-07-02:
@VitoTringolo the easiest way to ensure there is only one client per camera is to establish one data connection and four [pcicclient](https://github.com/ifm/ifm3d-examples/blob/master/pcicclient_set_io/ex-pcicclient_set_io.cpp#L65) connections. This will ensure that no other instance is getting data from the camera. Currently there is no XML-RPC setting for reducing the amount of connections available. Like @theseankelly already mentioned internally we do use more than one connection for our tooling like the Vision Assistant.

VitoTringolo, 2020-07-02:
@theseankelly Yeah I know that for sure there are cases were this kind of configuration can be useful. 
Let me try to find an example to explain why I want to limiting this behaviour.

We have two robots R_A and R_B each one of this robot is working on a different "scene" S_A and S_B and we have one camera C. All this setup is running with the robot R_A that is using the camera to recognize objects and then my software give an offset to drive the robot to the picking point. If the second robot R_B at some point is connected to the same camera that the robot R_A is using to pick objects in scene S_A the robot R_B will receive offset that are completely wrong (we are receiving offset that are correct for the scene S_A) for the scene S_B.
I know that this kind of scenario can be solved configuring the network in the correct way, for example configuring each PC used to drive the robot in a different subnet. But this kind of setup is not under my control, so I was trying to use the camera in an exclusive way.

@graugans It's like a super hack ;) but could be work.

theseankelly, 2020-07-02:
Hey @VitoTringolo I don't think I quite follow. Are you saying you have a single, shared camera (maybe mounted to some fixed location?) that an entire fleet of vehicles is connecting to in order to obtain depth information? In this case I would assume the individual robots are aware of their position with respect to the camera and transform the point cloud accordingly. R_A and R_B could consume data from the same camera concurrently without even knowing about each other. Perhaps I'm missing something fundamental?

VitoTringolo, 2020-07-02:
@theseankelly thanks for the reply.
Probably I'm not explaining myself correctly, but is exactly the opposite that you are describing. The camera is mounted on the robot and every robot has its own camera. What I want is to avoid that one robot can use a camera already in use by another robot.

theseankelly, 2020-07-02:
Ah, got it. So your camera and robots are each on the same subnet via DHCP?

Really the only realistic way to fully solve this is through static IP configuration and vehicle-local subnets, similar to how other IP devices do it (LiDARS, etc). Even if we could restrict access to one client at a time, I'd start to worry about boot-up race conditions and deadlocks across the fleet.

VitoTringolo, 2020-07-02:
Thanks for you time :)

closed 2020-07-02 by VitoTringolo

graugans, 2020-07-03:
Hey @VitoTringolo  this is quite easy to achieve. We do have a description field in the JSON data retrieved from the camera: https://github.com/ifm/ifm3d/blob/master/doc/configuring.md#dump ``ifm3d.Device.Description`` this for example can be used to store the robot ID in the camera and check weather the camera is for this bot or it is for another one. As an alternative you could use the MAC address to identify the camera, which in my opinion is not that nice cause you have to change the configuration or code of the bot when a camera is swapped. By using the description you can assign the camera to the bot offline.

Or use a fixed IP setting like @theseankelly already described. If using the the default IP range of the O3Ds I do recommend not using the ``.69`` as a device address. That's because if you bring a camera out of the box into the network a address conflict is there. Starting by ``.70`` will avoid this.