ifm/ifm3d #237: O3D303: software trigger, dumping config and grabbing image

andybayer, 2020-07-08:
Hello,

the following bug (or missing documentation?) might be not very relevant for most people. But I came across it and wanted to   make others aware of this counterintuitive behavior.

**Describe the Bug**
When grabbing images from a camera triggered by an external event (hardware or software) you need to directly grab the image after triggering the camera (or after image acquisition finished?). If you dump the camera's config between triggering and calling `FrameGrabber::WaitForFrame` you will be rewarded with a timeout when waiting for the image.

**Expected Behavior**
Instead of a timeout without further information I would at least expect an exception or some kind of warning in the documentation about this tricky behavior.

**Minimal Working Example**
```c++
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main(int argc, const char **argv)
{
    auto cam = ifm3d::Camera::MakeShared("10.12.228.51");

    ifm3d::ImageBuffer::Ptr img = std::make_shared<ifm3d::ImageBuffer>();
    ifm3d::FrameGrabber::Ptr fg = std::make_shared<ifm3d::FrameGrabber>(cam,
        fm3d::IMG_AMP|ifm3d::IMG_CART|ifm3d::IMG_RDIS);

    fg->SWTrigger();

    // Dump config between triggering and grabbing the image
    cam->ToJSON();

    if (!fg->WaitForFrame(img.get(), 1000))
    {
         std::cerr << "Timeout waiting for camera!" << std::endl;
         return -1;
    }

    imwrite("amplitude.png", img->AmplitudeImage());
    imwrite("radial_distance.png", img->DistanceImage());

    return 0;
}
```

**Environment**
 - OS: Ubuntu 20.04
 - ifm3d version: 0.18.0
 - using ROS: no

**Camera Configuration**
```json
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "476707713",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "263",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "351",
          "ClippingTop": "0",
[... 152 lines cut ...]
      "Date": "Wed Jul  8 12:55:07 2020",
      "HWInfo": {
        "Connector": "#!02_A300_C40_04057631_008023176",
        "Diagnose": "#!02_D322_C34_03950031_008026824",
        "Frontend": "#!02_F342_C34_19_00110_008023607",
        "Illumination": "#!02_I300_001_03389895_008001175",
        "MACAddress": "00:02:01:41:24:1D",
        "Mainboard": "#!02_M381_C41_04152517_008023690",
        "MiraSerial": "0830-81f1-0e66-07bc"
      },
      "SWVersion": {
        "Algorithm_Version": "2.2.14",
        "Calibration_Device": "00:02:01:41:24:1d",
        "Calibration_Version": "1.0.2",
        "Diagnostic_Controller": "v1.0.44-2dcaabbc73-dirty",
        "IFM_Recovery": "unversioned",
        "IFM_Software": "1.30.5309",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Mar 27 18:25:10 CET 2019",
        "Main_Application": "1.0.44"
      },
      "ifm3d_version": 1800
    }
  }
}
```

theseankelly, 2020-07-08:
Hey @andybayer -- thanks for taking the time to fill out a detailed bug report. The behavior you've discovered is by design. Frame memory (via the `ifm3d::Image` object). If no frame buffer has been pended on the `FrameGrabber` (via `ifm3d::FrameGrabber::WaitForFrame`) at the moment a frame is received from the camera, the frame is discarded. We do not cache the previous frame internally in the FrameGrabber.

And so yes -- the pattern of `SWTrigger` followed by an immediate call to `WaitForFrame` is technically a race condition. However given the time for camera exposure and data readout, it's a race that will always be won. Inserting additional computation (like dumping config) between these calls will make it more likely to lose the race (the frame arrives on the wire but the buffer hasn't been pended yet so it is discarded. Then `WaitForFrame` is called, but the frame has already been missed so the camera sits idle waiting for a trigger input)

For some extra timing guarantees, you could consider spinning up a thread to handle the blocking `WaitForFrame` before issuing the SWTrigger command. This would guarantee the buffer is pended and allow your primary loop to continue executing other tasks while waiting. Keep in mind that `ifm3d::ImageBuffer` is not thread safe, so protect access accordingly.

andybayer, 2020-07-08:
Thank you for your quick response.

This is interesting. We planned to wait for PCIC notifications when image acquisition finished (I disabled EnableAcquisitionFinishedPCIC in the test above because I wasn't sure if it has any influence on the observed behavior). After we receive the notification we call `FrameGrabber::WaitForFrame`. We tried this setup in some simple demo application and it worked (so far…). It seems that we should update this design according to your suggestion. 

I guess that `FrameGrabber` is also not thread safe. If someone protected `FrameGrabber` with a mutex, then call a blocking `WaitForFrame` how would he then call `SWTrigger`? `FrameGrabber` is then already blocked by one thread.

theseankelly, 2020-07-08:
How are you receiving the PCIC notification -- the `ifm3d::pcicclient` library? I'm not too familiar with the mechanics of the PCIC notification to know how it works (for example does it fire just before frames are read?) but it does sound like you're risking some race conditions in that design.

It's safe to use `FrameGrabber` in multiple threads in this context because there's no internal state being stored that would need synchronization primitives -- when I say "`ImageBuffer` is not thread safe" technically what I mean is that access to the buffers contained in the `ImageBuffer` object are not protected by synchronization primitives inside the class implementation -- so you shouldn't be reading/writing the images (`im->XYZImage()` for example) while the buffer is pending via a `WaitForFrame` call.

andybayer, 2020-07-14:
Sorry for the late response.

Yes, I am receiving the notification over `ifm3d::pcicclient`. I guess that I will change my design to call  `FrameGrabber::WaitForFrame` before triggering like you suggested. I won't use the notification for triggering anymore.

Thank you also for clarifying that using `FrameGrabber` in multiple threads in this context is safe. I am usually afraid to use non const entities concurrently in multiple threads if not stated explicitly.

closed 2020-07-21 by andybayer

graugans, 2020-07-28:
Hey @andybayer , it is not recommended to use the ``ifm3d::pcicclient`` notifications to start the ``WaitForFrame``. The  signal ``AcquisitionFinished`` is not guaranteed to meet the requirement that you have enough time to start a corresponding ``WaitForFrame`` call this is because your OS can decide to re schedule another task between those calls.. The safest way is to start ``WaitForFrame`` and then trigger the camera by ``SWTrigger``.

The intend of the PCIC  ``AcquisitionFinished`` notification was to have a Software way of chaining devices.

andybayer, 2020-07-28:
Hello @graugans,

thanks for your clarification.

What do you mean with your last sentence? N devices in a chain connected with hardware triggers and the first is triggered by software? What advantage would bring the ` AcquisitionFinished` notification to the table then?

graugans, 2020-07-28:
If you have the ability to chain devices with the ``AcquisitionFinished`` hardware output and the trigger input of the next camera this is the recommended way. If there is no such possibility you can simulate this in software with the help of the ``PCICAcquisitionFinished`` signal.

For details please check this document: https://github.com/inbangsa/ifm3d-examples/tree/feature/ex-multiple-camera-parallel-grabber/multi_camera_parallel_grabber

Please be also aware that in case of specific Events it is impossible to trigger the cameras. Those cases are:
- Over voltage
- Under voltage
- Over temperature
- Wrong trigger mode
- ....

This will break your chain. Currently @theseankelly and myself are in discussion how to improve the ``SWTrigger`` call to get a feedback of the current camera state. The same applies to your chained cameras. If the get stuck by one of the above issues your chain is broken. You have to implement an strategy to get it working again.

andybayer, 2020-07-28:
Thanks, that is very enlightening.