ifm/ifm3d #238: Camera output issue

Leo16112, 2020-07-14:
Hello,
we have some issue with camera 303. The problem is, that sometimes it starts to provide strange images.
Sometimes during usage, sometimes first 15 minutes after start. The problem is, that tomorrow we have project handover, and it its a bit problematic.

Sometimes it helps just switch it off for 5 minutes ( in case it appears during usage)
In case it happens after start- after 15 minutes it is away on its own.

This are the pictures of how it should be
[image]

[image]

And an image, that shows the problem
[image]

Version of firmware is 0.17

Kind Regards,
Anton.

graugans, 2020-07-14:
Hey @Leo16112, are you using a single camera or multiple cameras? Are there any changes in the scenery like shine objects? I also loop in @GreNait who has more experience with TOF artifacts than I do. You can also reach out to "Syntron, Support" <support.sy@ifm.com> which may make communication more direct. From the screenshots I do see you are based in Germany, like most of the support team is.

GreNait, 2020-07-14:
@Leo16112, the images are rather hard for me to understand. I guess, the black "blob" is the invalid part? Perhaps you can also share some pointcloud data. 

I have some ideas what you could check:
- It might be a problem between the first and second integration time (wrong setting) - You could try to change that and see what is happening
- Because you mentioned that the error is also disappearing, I would highly recommend to check the power supply. I saw issues with the image, when power drops during the image capture time. Try to get a different power supply (for testing) with at least 2.5 A
- Firmwareversion 0.17 is rather ... confusing? The official firmware is V1.30.5309. Please check, if that might be an issue too. 

If nothing of the above is helping, write me a message directly. So we can figure out what happens.

Leo16112, 2020-07-14:
@GreNait, @graugans 

I've just contacted my colleges there, we have 24V 10A (DRP-240-24) power supply for Camera(2.4A), IFM Barcode-Scaner(0.4A), and passive cooling IPC from Beckhoff with i7-8700. So it should be ok.

Firmwareversion... yep. I mixed it up with assistent software.  

I will contact the local support. Thank you for advise.

Kind Regards,
Anton.

Leo16112, 2020-07-16:
So we just put the problem to the instructions book and handover went just perfect, so I would close issue.

closed 2020-07-16 by Leo16112