ifm/ifm3d #240: set FrameRate faild

comwise, 2020-07-17:
I set the frequency(FrameRate) first ok, 

but i set MaxAllowedFrameRate is failed, can you tell me why?

set command is following:

`jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq .ifm3d.Apps[0].Imager.FrameRate
"20"
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq .ifm3d.Apps[0].Imager.MaxAllowedFrameRate
"4.16666666666667"
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq '.ifm3d.Apps[0].Imager.MaxAllowedFrameRate="20"' | ifm3d --ip=192.168.1.68 config
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq .ifm3d.Apps[0].Imager.MaxAllowedFrameRate
"4.16666666666667"
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq '.ifm3d.Apps[0].Imager.MaxAllowedFrameRate="20"' | ifm3d --ip=192.168.1.68 config
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq .ifm3d.Apps[0].Imager.MaxAllowedFrameRate
"4.16666666666667"
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq '.ifm3d.Apps[0].Imager.MaxAllowedFrameRate="15"' | ifm3d --ip=192.168.1.68 config
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq .ifm3d.Apps[0].Imager.MaxAllowedFrameRate
"4.16666666666667"
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq '.ifm3d.Apps[0].Imager.MaxAllowedFrameRate="2"' | ifm3d --ip=192.168.1.68 config
jz@jzrobot-x:~$ ifm3d --ip=192.168.1.68 dump | jq .ifm3d.Apps[0].Imager.MaxAllowedFrameRate
"4.16666666666667"`

graugans, 2020-07-20:
Hey @comwise the value of *MaxAllowedFrameRate* is a read only parameter. The O3X comes with a [VCSEL](https://en.wikipedia.org/wiki/Vertical-cavity_surface-emitting_laser) as an active IR illumination, due to eye safety constraints there is a maximum of emitted light budget. To avoid any eye damage the maximum frame rate is limited by the eye safety factor. This value is calculated for each mode and setting. The safety mechanism itself is implemented in hardware the software part is to avoid to trigger the hardware shutdown of the VCSEL.

comwise, 2020-07-21:
@graugans hi thanks for your reply, i want to know how to setting camera to get a larger frequency， such as  > 10HZ

graugans, 2020-07-21:
Hey @GreNait , can you please provide some tuning tipps

GreNait, 2020-07-21:
@comwise I have some bad news for you. 20Hz are only possible with some certain set parameters.

- One integration time
- Low exposure time setting
- Single frequency

Then you might reach 20Hz. 
If you use two integration times, you cut the frame rate by half(10Hz). Using two frequency's, another cut (5Hz.)

You have to choose now between several trade offs if you want to reach >10Hz. Double integration but single frequency (ambigous range issues could occur) = ca. 10Hz. Single integration and double frequency (problems with black/white objects) = ca. 10Hz.

I hope this is helping

theseankelly, 2020-07-30:
Hey @comwise have you found @GreNait's answers to be sufficient?

comwise, 2020-08-11:
@theseankelly @GreNait @graugans  yes thanks for your reply

closed 2020-08-11 by theseankelly