ifm/ifm3d #241: o3d303: Lib: XMLRPC Timeout - can you `ping' the sensor?

zener90818, 2020-07-22:
I am trying to connect to the O3D303 Sensor using debian 9 and ROS Melodic.When I run catkin_make  run_tests, l get these errors:
[image]
Here is the failed  trying:
[image]
[image]
[image]

zener90818, 2020-07-22:
mway@mway:~$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
=;enp1s0;IPv4;O3D3XX-41-8D-BA-7\032\09100\05802\05801\05841\0588d\058ba\093;Workstation;local;O3D3XX-41-8D-BA-7.local;169.254.102.130;9;

inbangsa, 2020-07-22:
@zener90818 its seems your  host system has ip 192.168.0.69 which i think is conflicting  with the default ip of the device and hence  device is taking a local-link address.  Quick first solution would be to change the ip of the your system to 192.168.0.xx where xx takes other values than 69. checkout our [troubleshoot guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md) for connection issues.

zener90818, 2020-07-22:
@inbangsa thanks for reply. i change the ip to 192.168.0.80, it doesn't seem to work.  Did i change the ip of the device in the right way?
[image]

theseankelly, 2020-07-22:
Hey @zener90818 it seems like you've changed the IP address of your camera -- @inbangsa suggested you change the IP address assigned to your host (the enp1s0 interface). Nevertheless, this should work but you have to set an environment variable.

Add the following to your `.bashrc` and logout/login:
```
export IFM3D_IP=192.168.0.80
```

Alternatively if you want to avoid the environment variable you can pass the IP directly to the ROS node and the ifm3d tool:
```
$ ifm3d --ip=192.168.0.80 ls
$ roslaunch ifm3d camera.launch ip:=192.168.0.80
```

Finally just an FYI -- the `ifm3d` library as well as the ROS package [are available through the ROS distro](https://index.ros.org/p/ifm3d/github-ifm-ifm3d-ros/#melodic):
```
$ sudo apt install ros-melodic-ifm3d
```

inbangsa, 2020-07-22:
@zener90818 how you changed the ip of the device ?  or you changed the ip of the host system ?  

I assume your camera  is at ip address of ``192.168.0.69``` and your host system is at ``` 192.168.0.80``` then as per troubleshoot guide please follow this [steps](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#if-you-are-behind-the-proxy-network)  and then try

```
$ ifm3d --ip=192.168.0.69 ls 
```

zener90818, 2020-07-22:
@inbangsa @theseankelly Thanks a lot. the problem is solved.

closed 2020-07-22 by zener90818

himanshuak, 2023-04-04:
Hi @zener90818 , I am trying to connect to O3R device and I am facing similar issue with respect to connection. I have followed the steps described in this discussion and I have set the IP address of my pc. PC IP settings 
[image]
I tried connecting to the sensor using the default IP address, but i was not successful. And also I am not able to retrieve the ip address of the camera using the troubleshooting guidelines provided [here](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md). 
Any suggestions would be very helpful. Thanks in advance.