ifm/ifm3d #249: Intrinsic parameter is not supported by Firmware

ShettyHarapanahalli, 2020-08-04:
Hi, 

I am using the 30313 camera when I try to compute the Cartesian coordinates I get the error as "Intrinsic parameter is not supported by Firmware". I have ifm3d: version=0.18.0 firmware version and I am executing the example from the link -  https://github.com/ifm/ifm3d/blob/master/doc/compute_cartesian/O3D_Cartesian_Computation.ipynb. 

Can you please share a workaround for this problem.

theseankelly, 2020-08-04:
Hi @ShettyHarapanahalli the firmware in question is the camera firmware itself, not the ifm3d driver. What version of camera firmware are you running? You can check via `ifm3d dump` and look at the `IFM_Software` field under `SWVersion`

My records show the intrinsic information is supported as of firmware 1.23.x

ShettyHarapanahalli, 2020-08-05:
Hi, @theseankelly I have "1.6.2114" version on my camera. I see the latest version 1.30.xx in the web portal, can you please share the link where I can get compatible firmware.

graugans, 2020-08-05:
Hey @ShettyHarapanahalli the download package can be found here: https://www.ifm.com/us/en/search?query=O3D301#!/infomaterialanddownloads?_type=infomaterialanddownloads

ShettyHarapanahalli, 2020-08-06:
This link is for the latest version 1.30.5309, will this support intrinsic values. 

If there is a python API/interface to get the cartesian coordinates can you please share the same.

theseankelly, 2020-08-06:
There are python bindings to `ifm3d` and you can read about those [here](doc/python.md)

We also have two samples which illustrate how to work with the intrinsics (which are written with the python bindings). However, you can get cartesian data straight from the device if you don't want to compute it.

- [Image Rectification](https://github.com/ifm/ifm3d-examples/blob/master/image_rectification/python/Image_Rectification.ipynb)
- [Intrinsics to Cartesian](https://github.com/ifm/ifm3d-examples/blob/master/intrinsics_to_cartesian/python/intrinsics_to_cartesian.ipynb)

closed 2020-08-10 by theseankelly

ShettyHarapanahalli, 2020-08-11:
Have updated the firmware to the latest one shared, now I get the connection error - 
>>> fg = ifm3dpy.FrameGrabber(cam, ifm3dpy.IMG_AMP | ifm3dpy.INTR_CAL | ifm3dpy.IMG_RDIS | ifm3dpy.IMG_CART)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Lib: XMLRPC Timeout - can you `ping' the sensor? 

I am able to ping to the sensor, below is the response - 
ping 192.168.0.90
PING 192.168.0.90 (192.168.0.90) 56(84) bytes of data.
64 bytes from 192.168.0.90: icmp_seq=1 ttl=64 time=0.481 ms
64 bytes from 192.168.0.90: icmp_seq=2 ttl=64 time=0.445 ms
64 bytes from 192.168.0.90: icmp_seq=3 ttl=64 time=0.414 ms
64 bytes from 192.168.0.90: icmp_seq=4 ttl=64 time=0.411 ms
64 bytes from 192.168.0.90: icmp_seq=5 ttl=64 time=0.427 ms
64 bytes from 192.168.0.90: icmp_seq=6 ttl=64 time=0.430 ms
^C
--- 192.168.0.90 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5094ms
rtt min/avg/max/mdev = 0.411/0.434/0.481/0.033 ms

theseankelly, 2020-08-11:
Can you share the line where you create the camera object as well?

ShettyHarapanahalli, 2020-08-12:
Sharing the cmd line dump - 
suman@mlaidl:~$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
=;enp59s0;IPv4;O3D313-40-2F-75\032\09100\05802\05801\05840\0582f\05875\093;Workstation;local;O3D313-40-2F-75.local;192.168.0.90;9;
^Z
[1]+  Stopped                 avahi-browse -arp 2> /dev/null | grep --color=auto ^=.*O3D.*Workstation
suman@mlaidl:~$ ifm3d --ip=192.168.0.90
ifm3d: version=0.18.0
suman@mlaidl:~$ ifm3d ls
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
suman@mlaidl:~$ ping 192.168.0.90
PING 192.168.0.90 (192.168.0.90) 56(84) bytes of data.
64 bytes from 192.168.0.90: icmp_seq=1 ttl=64 time=0.811 ms
64 bytes from 192.168.0.90: icmp_seq=2 ttl=64 time=0.403 ms
64 bytes from 192.168.0.90: icmp_seq=3 ttl=64 time=0.479 ms
64 bytes from 192.168.0.90: icmp_seq=4 ttl=64 time=0.462 ms
^C
--- 192.168.0.90 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3053ms
rtt min/avg/max/mdev = 0.403/0.538/0.811/0.162 ms
suman@mlaidl:~$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ifm3dpy
>>> cam = ifm3dpy.Camera()
>>> fg = ifm3dpy.FrameGrabber(cam, ifm3dpy.IMG_AMP | ifm3dpy.INTR_CAL | ifm3dpy.IMG_RDIS | ifm3dpy.IMG_CART)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Lib: XMLRPC Timeout - can you `ping' the sensor?
>>> exit()
suman@mlaidl:~$ ifm3d dump
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
suman@mlaidl:~$

theseankelly, 2020-08-12:
Thanks!

Since you're using a non-default IP you have to specify it when instantiating the camera object:
```python
cam = ifm3dpy.Camera('192.168.0.90')
```

Hopefully that does the trick; let me know.

ShettyHarapanahalli, 2020-08-12:
It is not working, I have shared the dump below - 

>>> import ifm3dpy
>>> cam = ifm3dpy.Camera('192.168.0.90')
>>> fg = ifm3dpy.FrameGrabber(cam, ifm3dpy.IMG_AMP | ifm3dpy.INTR_CAL | ifm3dpy.IMG_RDIS | ifm3dpy.IMG_CART)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Lib: XMLRPC Timeout - can you `ping' the sensor?

ShettyHarapanahalli, 2020-08-12:
ok, got it working.

theseankelly, 2020-08-12:
Good news! Let us know if anything else comes up.

commit 7dedfe7a (2023-08-03): Merge branch 'feature/release_notifications' into 'main'

Add release notifications

Closes #249

See merge request syntron/support/csr/ifm3d/ifm3d!315