ifm/ifm3d #256: 'cat'ing a .swu file to `ifm3d swupdate` has no effect

theseankelly, 2020-08-09:
I believe the recent cxxopts change has broken piping swu files into `ifm3d swupdate`.

The command:
```
$ cat firmware.swu | ifm3d swupdate
```

Appears to have no effect. However the underlying swupdate function does work and performs the full firmware update.
```
$ ifm3d swupdate --file=firmware.swu
```

Perhaps an issue in how cxxopts is handling stdin for this module.

theseankelly, 2020-08-09:
Hey @inbangsa can you take a look? I believe this is related to the cxxopts change.

graugans, 2020-08-10:
Hey @theseankelly, last Friday we had an issue when we installed the ifm3d from the snap store and tried  

```
$ ifm3d swupdate --file=firmware.swu
```

Even with the full path it could not open the file. When we used

```
$ ifm3d swupdate < firmware.swu
```
it worked.

theseankelly, 2020-08-10:
Hey @graugans thanks for the report -- I believe this is due to snap confinement. The snap is disallowed from accessing any filesystem resources so the firmware must be redirected via stdin as you have done. I've created issue #257 to track adding the appropriate interfaces to the snap in a future release.

I do think my issue is separate and is isolated to the develop branch, whether it's cxxopts or something else.

commit 17ab45c5 (2023-09-28): Merge branch '256-deserializer-module-not-available-in-docker-container' into 'main'

Add deserializer module to docker container.

Closes #256

See merge request syntron/support/csr/ifm3d/ifm3d!338
closed 2022-11-26 by github-actions[bot]