ifm/ifm3d #259: NTP activation error

robinbansal08, 2020-08-19:
I am trying to activate NTP synchronisation in O3D303 sensor, but I am getting error as described below.  

Here is the dump of the sensor: 
```
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "476707713",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "263",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "351",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "false",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "10000",
          "ExposureTimeList": "2500;10000",
          "ExposureTimeRatio": "4",
          "FlyingPixelEraser": "0",
          "FrameRate": "5",
          "MaxAllowedLEDFrameRate": "4.9",
          "MinimumAmplitude": "2",
          "Resolution": "1",
          "SpatialFilter": {
            "MaskSize": "1"
          },
          "SpatialFilterType": "1",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_moderate",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Box Detection Application",
        "PcicEipResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"diagnostic_data\" }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "RtspOverlayStyle": "{\"ROI\": {\"default\": {\"visible\": true, \"filled\": false, \"use_symbol\": false, \"label_alignment\": \"top\", \"label_content\": \"\", \"label_background\": \"black\", \"label_fontsize\": 8, \"label_failonly\": false}, \"model_defaults\": {}, \"specific\": {} } }",
        "TemplateInfo": "",
        "TriggerMode": "2",
        "Type": "Camera"
      }
    ],
    "Device": {
      "ActiveApplication": "1",
      "ArticleNumber": "O3D303",
      "ArticleStatus": "AD",
      "Description": "",
      "DeviceType": "1:2",
      "EIPConsumingSize": "8",
[... 73 lines cut ...]

```

**NTP activation error**
```
$ ifm3d --ip=192.168.0.11 dump | jq '.ifm3d.Time.SynchronizationActivated="true"' | ifm3d --ip=192.168.0.11 config
ifm3d error: [json.exception.type_error.302] type must be string, but is number
```

```
$ ifm3d --ip=192.168.0.11 dump | jq '.ifm3d.Time.NTPServers="192.168.0.2"' | ifm3d --ip=192.168.0.11 config
ifm3d error: [json.exception.type_error.302] type must be string, but is number
```

```
$ echo {"ifm3d":{"Time":{"NTPServers": "192.168.0.2"}}} | ifm3d --ip=192.168.0.11 config
ifm3d error: -100002
Lib: Error processing JSON
```

```
$ echo {"ifm3d":{"Time":{"SynchronizationActivated": "true"}}} | ifm3d --ip=192.168.0.11 config
ifm3d error: -100002
Lib: Error processing JSON
```

robinbansal08, 2020-08-25:
@inbangsa  thanks a lot for the information.

inbangsa, 2020-08-25:
@robinbansal08  new firmware is available at ifm website.

[firmware link]( https://www.ifm.com/gb/en/downloadarea/DualisContent )

let us know if you face any more issues.

robinbansal08, 2020-08-26:
@inbangsa thanks for the update. I have updated the camera with newest firmware and the issue is now resolved.

Thanks a lot for your help.

robinbansal08, 2020-09-02:
I have updated the firmware and I am facing some new issues, I cannot get response from camera for ifm3d _dump_ and _time_ commands. 

I am able to ping the camera successfully:
```
$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=7.68 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=6.60 ms
64 bytes from 192.168.0.69: icmp_seq=3 ttl=64 time=4.28 ms
64 bytes from 192.168.0.69: icmp_seq=4 ttl=64 time=4.40 ms
^C
--- 192.168.0.69 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3005ms
rtt min/avg/max/mdev = 4.287/5.747/7.685/1.451 ms

```

But ifm3d commands dump and time are unresponsive:
```
pactris-nuc@pactrisnuc-NUC7i5BNH:~$ ifm3d --ip=192.168.0.69 ls
[
  {
    "Active": true,
    "Description": "",
    "Id": 476707713,
    "Index": 1,
    "Name": "Sample Application"
  }
]
$ ifm3d --ip=192.168.0.69 dump
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
$ ifm3d --ip=192.168.0.69 time
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
$ ifm3d --ip=192.168.0.69 time
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
$ ifm3d --ip=192.168.0.69 dump
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
$ ifm3d --ip=192.168.0.69 dump
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?

```
**Observation:** When I am sending the _dump_ and _time_ commands, OUT1 (Yellow) LED goes off for a brief moment as soon I send the command. The running application in the camera is configured for process triggered.

inbangsa, 2020-09-02:
Hey @robinbansal08, I have tried to reproduce the dump and time command issue as described by you, but all working fine on the new firmware.  I think there might be IP address conflict but not sure please check the following  
  
* multiple cameras with default IP address in network. 
* is there any other device with same IP address on the network. 

for me it seems be a network issue and not a firmware issue.

robinbansal08, 2020-09-02:
@inbangsa There is only camera connected in the network and no other device is connected in the network with same IP. I did reboot the camera but the issue didn't resolve but _rebooting the PC_ resolved this issue. This issue has reappeared for the second time, and I am not sure why this issue is appearing in the idle state (camera is connected in the network for several days). The o3d303 commands seems to work fine except few of the ifm3d commands.

inbangsa, 2020-09-03:
@robinbansal08 can you try this commands from other system on the same network when your current system shows this behavior. This might help to zoom into the issue, can this be related to a network hardware of your system? ( as rebooting the PC resolves this)

robinbansal08, 2020-09-08:
@inbangsa The problem hasn't reappeared yet, I will try the commands with another system in the same network once this issue reappears again.

inbangsa, 2020-09-21:
@robinbansal08 any update on the issue ? 

the discussion is diverted from the issue topic and hence if the issue persists  for "ifm3d dump and time" command i request you to create a  new issue.

please let me know if we can close this issue or you still facing issue with NTP ?

robinbansal08, 2020-09-21:
@inbangsa The issue regarding "ifm3d dump and time" command, didn't reappear again yet, and also NTP is resolved.

closed 2020-09-21 by robinbansal08