ifm/ifm3d #260: Programmatically delay the start of data acquisition after a trigger

flajolet, 2020-09-08:
Is there a way to programmatically delay the start of data acquisition after a hardware trigger has been registered for the O3D family of cameras?

My usecase is the following: I have multiple O3D cameras running with a single electrical trigger line to simplify the hardware setup. However to minimize interferences, I would like the cameras to be triggered sequentially. One way to achieve that for instance would be to delay in time the start of data acquisition by say 10 milliseconds for the second camera and 20 milliseconds for the third camera and so on.

I have looked at your examples and this https://github.com/inbangsa/ifm3d-examples/tree/feature/ex-multiple-camera-parallel-grabber/multi_camera_parallel_grabber is very promising but it requires a more complex hardware setup. I am also using different channels for the cameras but it seems that I am still getting a bit of cross-interferences compared to when using a single camera.

Thanks for your help!

inbangsa, 2020-09-09:
@flajolet  sorry there is no way to put the delay in image capture after the trigger is received, you can add this delay in trigger system. one way is to solve  the issue of interference is sequential triggering the device and to explain that we provide [multi_camera_parallel_grabber Example]( https://github.com/inbangsa/ifm3d-examples/tree/feature/ex-multiple-camera-parallel-grabber/multi_camera_parallel_grabber). This is under review stage and we would like to know your concerns about this example. we tried to explain the hardware trigger setup required in this [README](https://github.com/inbangsa/ifm3d-examples/blob/feature/ex-multiple-camera-parallel-grabber/multi_camera_parallel_grabber/README.md).  Let us know  if you have any issue in this configuration. As per my understanding the following  changes are needed in your setup.

* All camera must be configured in hardware trigger mode 
* First camera will get the trigger from your trigger system ( your system trigger  to "Trigger pin" of the first camera )
* Configure all devices except first with Logical graph connecting  soft signal "Image capture finished" to a digital out pin ( this can be done by configuring camera through code, this is also done in example)
* Now use the digital output to trigger next camera ( this is by connecting digital out pin to the trigger pin of next camera)

camera pins are explained at this  [link](https://www.ifm.com/in/en/product/O3D303?tab=accessories) in wiring section

flajolet, 2020-09-11:
Ok thanks, I'll consider using this well-documented example

closed 2022-11-25 by github-actions[bot]