ifm/ifm3d #261: Certificate error when fetching the ifm3d debian repository updates

wurza, 2020-09-11:
**Describe the bug**
Since some time I get a certificate error from the ifm3d repository, when fetching updates via `apt update` in the command line:

```
Err:11 https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic Release                                                                              
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 212.62.205.12 443]
```

My first thought was, that the problem was on my side, but today i tried to help a colleague to install the ifm3d library and he faced the same problem.

**To Reproduce**

1. Open Terminal on Linux
2. Use the instruction from https://github.com/ifm/ifm3d/blob/master/README.md#ubuntu-linux-via-apt-amd64arm64 to add the debian package repository
3. Type `apt update`

**Expected behavior**
Updated finished successfully and I can install the newest version of the library via `apt install ...`.

**Output/Screenshots**
See terminal snippet in description.

**Configuration/Environment (please complete the following information):**
 - OS: KDE Neon (with Ubuntu 18.04 base) and vanilla Ubuntu 18.04 on my colleagues machine.
 -  ifm3d version: -
 - using ROS: no

graugans, 2020-09-14:
Hey @wurza can you please try to execute the following commands to verify if the issue is still present?

```
apt-get clean
apt-get autoclean
```

wurza, 2020-09-14:
Good morning. I tried your commands to clean the APT cache, but the problem is still present.

```
...
Err:5 https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic Release                                                
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 212.62.205.12 443]
...
Reading package lists... Done                       
E: The repository 'https://nexus.ifm.com/repository/ifm-robotics_ubuntu_bionic_amd64 bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

graugans, 2020-09-14:
Hey @wurza ,
thanks for the feedback. I'll get in touch with our IT department, so they can check the issue.

graugans, 2020-09-16:
Hey @wurza, there was issue with the SSL certificate chain. Our IT department has fixed this please try again.

wurza, 2020-09-16:
Hello @graugans, I just tried to update my package list. it is working again. Thank you very much for your help!

closed 2020-09-16 by wurza