ifm/ifm3d #272: Offset between distance image recorded via python implementation and ifm 3d vision Assistant

Taaaaaantrum, 2020-11-02:
We are using a O3D303 camera and want to extract the amplitude and the distance image via python. Implementation is working, but there is an offset in the distance image between the recorded data and the data shown in the ifm vision Assistant. What could be the issue there? I assume there needs something to be added in the code. Attached the sample usage of the camera. 

Many thanks in advance!
Felix Hess

[sample_code.txt](https://github.com/ifm/ifm3d/files/5474017/sample_code.txt)

graugans, 2020-11-02:
In your example code you are using ``0xFF`` as an mask:
```python
fg = ifm3dpy.FrameGrabber(cam, 0xFF) #without 0xFF i cannot access the distance image
```
This will enable all images to be transferred. The recommended way is to select the images you are interested in. This should work:

```python
fg = ifm3dpy.FrameGrabber(cam, ifm3dpy.IMG_AMP|ifm3dpy.IMG_CART|ifm3dpy.IMG_RDIS)
```

The difference you see in your data and the data shown in ifm Vision assistant is most probably caused by the fact you are requesting radial distance image and the Vision Assistant is using the Cartesian data  ``IMG_CART`` to display the data.

Taaaaaantrum, 2020-11-06:
Ok, thank you!

closed 2020-11-06 by Taaaaaantrum