ifm/ifm3d #273: Basic installation question

jbies, 2020-11-12:
When it comes to C++ I am green at best so forgive me for such a basic question.

On Ubuntu 20.04 I have installed all software using apt. I follow the installation instructions word for word from the "README.md". The command line tool works fine, the pcl-viewer works fine. But, when I try to run the code example from the "basic_usage.md"  I get build errors that I am missing files for opencv2 (<opencv2/core/core.hpp>). I also appear to be missing glog which is included in one of the header files. 

Question: What is the exact list of required libraries/packages/files that need installed prior to building and running a ifm3d client program? I am only needing to install opencv2 and glog? All I have done is follow the installation instructions, then I copy and paste the example, try to build and execute, and no luck.

Thanks in advance.

graugans, 2020-11-12:
Hey @jbies,

no problem we all started as beginners. In this case it is the documentation which is unfortunately not very detailed. Please take a look at our examples repository: https://github.com/ifm/ifm3d-examples this includes the source, code plus the relevant CMake build scripts. Which you'll need to set-up both the include path and the liobrary Path of the dependencies.

In case of any question please feel free to reach out.

Best,
Christian

jbies, 2020-11-12:
Christian,

Thank you for the prompt response, I really appreciate it. I will clone the examples repo and try to get that working. 

Jake

jbies, 2020-11-12:
Ok, so here are the steps I took:

clone the examples repository. (git clone https://github.com/ifm/ifm3d-examples.git)

cd into the repo directory which includes the "CMakeLists.txt" file 

Then followed the build the examples steps:
$ mkdir build
$ cd build
$ cmake ..
$ make

After entering "cmake .." I get a CMake error. Attached is a screenshot of the error. If someone could please tell me what step I am missing/skipping I would really appreciate it. 

[image]

Before running into installation errors with the example repo I also tried pointing directly to the lib/include directories by changing the path settings in codelite.

[image]

jbies, 2020-11-20:
This may be silly of me to even be updating but just in case there is someone else out there like me I figured I would update this post. After playing with o3d3xx-python I decided to move on to ifm3dpy. After just trying to run the setup.py file from the root directory of the ifm3d repo I ran into installation problems yet again. After starring at the installation instructions for (I don't know how long, but too long) I realized when I had installed everything I had used the command (sudo **apt** install <ifm3d-package>) instead of (sudo **apt-get** install <ifm3d-package>). Me not knowing any better thought these commands were identical but I thought I would wipe the slate clean and try again. So I did this:

1) uninstall all ifm3d packages (sudo apt purge <ifm3d-package) 
    - I did this one by one because sudo apt purge ifm3d* didn't work
2) I installed all dependencies listed from the Linux source build instructions
   - Probably unnecessary but I had nothing to lose at that point
3) re-install ifm3d packages (sudo **apt-get** install <ifm3d-package>
4) re-cloned the ifm3d repo and ran setup.py

After that everything seemed to work fine with ifm3dpy. I still am having trouble with the opencv not found error when trying to run any cpp example. But for now I am not going to worry about cpp, with the amount I need to learn it is best I start with python anyway. This may have been an obvious issue to a lot of IFM users, but for how much I struggled with it I thought it wouldn't hurt to post about it.

graugans, 2020-11-23:
Hey @jbies,
if you are planning to use the Python binding you can use the pre compiled binaries. They are installed with apt/apt-get wen you have this on your commandline ``ifm3d-python3`` the dependency on OpenCV is resolved when you are following the instructions on [how to build from source](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md). The package in question is this one: ``libopencv-dev`` if you are gonna to plan using ROS then the OpenCV package should be taken from the ROS repository.
Best,
Christian

jbies, 2020-11-23:
@graugans,

Thank you for following up, I appreciate it. I do not plan on using ROS at this time but that is good to know. 

Jake

jbies, 2021-01-25:
After playing with ifm3dpy for a while now I am circling back to the cpp examples. I am still having an issue with opencv not found when trying to build any example program. I believe my problem is that when I install ifm3d dependencies it installs opencv to (usr/include/opencv4) and when I build a program the compiler is looking for opencv in (opencv2/core/core.hpp). That file does exist but it is found in (/usr/include/opencv4/opencv2) Can someone please tell me what I am doing wrong or how to work around this issue? 

I did try copying opencv2 out of (/usr/include/opencv4) and pasting it to (/usr/include) which solves the opencv2 problem but then I run into the same kind of problem with pcl. PCL installs to (/usr/include/pcl-1.10) when installing ifm3d instead of (/usr/include/pcl).

I believe my issue is very similar to issue #283, if not the same.

If there is a better, detailed, step-by-step instruction on the proper installation of ifm3d it would be very helpful to beginners like myself. Thanks in advance.

graugans, 2021-01-26:
Hey @jbies,

I just checked with my Ubuntu 20.04 Docker container, which is able to resolve the OpenCV and PCL headers as expected. Due to the fact I have tested on a slightly newer version of the ifm3d source code I am just rerunning the build with the version **v0.18.0** one question, the version you are currently testing with is the one from the apt repository or the one you have build from source?

One thing which confuses me is that you have selected __CROSS GCC__ in one of the screenshots above.

graugans, 2021-01-26:
I just finished my Docker build and it worked as expected on the Ubuntu 20.04 this is how I tested.

My build script
```bash
#!/bin/bash

set -e # terminate if a command fails

# we do use some BuildKit feature like ssh mounts
export DOCKER_BUILDKIT=1

# Start the SSH Agent and add the SSH key to it
eval $(ssh-agent)
# This may need adjustments to fit your key name and location
ssh-add ~/.ssh/id_rsa

#UBUNTU_VERSIONS_TO_BUILD=(18.04 20.04)
UBUNTU_VERSIONS_TO_BUILD=(20.04)

for VERSION in "${UBUNTU_VERSIONS_TO_BUILD[@]}"
do
    # Build the Docker container and forward the SSH Agent to specific commands
    docker build \
                --build-arg UBUNTU_VERSION=${VERSION} \
                --build-arg VERSION_TAG=v0.18.0 \
                --rm \
                --pull \
                --ssh default=$SSH_AUTH_SOCK \
                -t ifm3d-debs-${VERSION} \
                .
done
```

This is the ``Dockerfile``
```Docker
#syntax=docker/dockerfile:1.2

ARG UBUNTU_VERSION=18.04

FROM ubuntu:${UBUNTU_VERSION} AS buildstage

ENV PYBIND_VERSION 2.3.0

#####################################################
#####################################################

ENV DEBIAN_FRONTEND noninteractive

RUN apt-get update && apt-get install -y --no-install-recommends \
                       libboost-all-dev \
                       git \
                       openssh-client \
                       libcurl4-openssl-dev \
                       libgtest-dev \
                       libgoogle-glog-dev \
                       libxmlrpc-c++8-dev \
                       libopencv-dev \
                       libpcl-dev \
                       libproj-dev \
                       python3-dev \
                       python3-pip \
                       build-essential \
                       coreutils \
                       findutils \
                       cmake \
                       ninja-build \
                       dpkg-dev \
                       locales && \
                       rm -rf /var/lib/apt/lists/*

# Update PIP this maybe required for some packages
RUN python3 -m pip install -U pip

# Install missing Python packages
RUN yes | python3 -m pip install pytest numpy setuptools wheel

RUN mkdir -p -m 0600 ~/.ssh && ssh-keyscan github.com >> ~/.ssh/known_hosts
RUN mkdir -p -m 0600 ~/.ssh && ssh-keyscan gitlab.dev.ifm >> ~/.ssh/known_hosts

RUN echo "en_US.UTF-8 UTF-8" >> /etc/locale.gen && \
    locale-gen en_US.UTF-8 && \
    /usr/sbin/update-locale LANG=en_US.UTF-8
ENV LANG en_US.UTF-8
ENV LANGUAGE en_US:en
ENV LC_ALL en_US.UTF-8

RUN mkdir -p /opt/src
WORKDIR /opt/src

RUN git clone --branch v${PYBIND_VERSION} https://github.com/pybind/pybind11 && \
    cd pybind11 && \
    mkdir -p build && \
    cd build && \
    cmake -GNinja -DPYBIND11_TEST=OFF .. && \
    ninja && \
    ninja install

# ifm3d git release tag to build
ARG VERSION_TAG=master
ARG IFM3D_REPO_CLONE_URL=https://github.com/ifm/ifm3d.git

# clone the repo
RUN --mount=type=ssh \
    git clone --branch ${VERSION_TAG} ${IFM3D_REPO_CLONE_URL} && \
    cd ifm3d && \
    mkdir -p build && \
    cd build && \
    cmake -GNinja -DCMAKE_INSTALL_PREFIX=/usr \
            -DBUILD_TESTS=OFF \
            -DBUILD_MODULE_OPENCV=ON \
            -DBUILD_MODULE_PCICCLIENT=ON \
            -DBUILD_MODULE_PYBIND11=ON -DPYTHON_EXECUTABLE=/usr/bin/python3 \
            .. && \
    ninja && \
    ninja package && \
    ninja repackage && \
    mkdir -p /opt/apt-repo && \
    cp /opt/src/ifm3d/build/*.deb /opt/apt-repo && \
    cd /opt/apt-repo && \
    (dpkg-scanpackages . /dev/null | gzip -9c > Packages.gz)

# The run stage
FROM ubuntu:${UBUNTU_VERSION} AS deployment

ENV DEBIAN_FRONTEND noninteractive

COPY --from=buildstage /opt/apt-repo/* /data/apt-repo/
RUN echo 'deb [trusted=yes] file:/data/apt-repo/ ./' > /etc/apt/sources.list.d/ifm3.list && \
    apt-get update && apt-get install -y --no-install-recommends \
        python3 \
        jq \
        ifm3d-camera \
        ifm3d-framegrabber \
        ifm3d-image \
        ifm3d-opencv \
        ifm3d-pcicclient \
        ifm3d-tools \
        ifm3d-python3 && \
        rm -rf /var/lib/apt/lists/*

# The testing stage
FROM deployment AS testingstage

RUN mkdir -p /opt/src
WORKDIR /opt/src

COPY --from=buildstage /opt/apt-repo/* /data/apt-repo/
RUN echo 'deb [trusted=yes] file:/data/apt-repo/ ./' > /etc/apt/sources.list.d/ifm3.list && \
    apt-get update && apt-get install -y --no-install-recommends \
        git \
        libopencv-dev \
        libpcl-dev \
        libgoogle-glog-dev \
        ca-certificates \
        build-essential \
        coreutils \
        cmake \
        ninja-build

# clone the repo
RUN --mount=type=ssh \
    git clone https://github.com/ifm/ifm3d-examples.git && \
    cd ifm3d-examples && \
    mkdir -p build && \
    cd build && \
    cmake -GNinja .. && \
    ninja
```
Please do not worry when you not have used Docker before. The above Dockerfile is using BuildKit and a multistage Dockerfile which is also not as straight forward as most of the Dockerfiles you might used to. The idea behind this test set-up is to build ifm3d from source and create the appropriate Debian packages. Those are isnstalled in a deployment stage. Which has no build Pacakges installed and later in the testing stage I do compile the ifm3d-examples.

So even if you have not used Docker before the ``apt-get`` lines shall give you an insight what packages you have to install to get stuff working.

If you do wonder about the SSH agent stuff this is just an easy way to share secrets without pasting them in the Dockerfile an allows me building of internal repositories.

jbies, 2021-01-26:
@graugans, Thank you so much for the detailed reply. I apologize for the confusion, I am trying to explain the problem the best I can but being a beginner does not make that easy. In the screenshot above I was not even aware that Cross GCC was selected. When opening the build settings in Codelite, that is the default selection. Also, when initially setting up Codelite GCC is the default compiler, so I believe the include path I had setup for Cross GCC wasn't doing anything for me. I had decided to restart with a fresh install of Ubuntu a couple of weeks ago, here is what I have done to install ifm3d:

**Followed instruction on installing binaries from main readme**
1) added repo to sources.list
2) added the public key
3) installed all software using apt
sudo apt-get install ifm3d-camera \
                       ifm3d-framegrabber \
                       ifm3d-swupdater \
                       ifm3d-image \
                       ifm3d-opencv \
                       ifm3d-pcicclient \
                       ifm3d-tools \
                       ifm3d-python3 \
                       ifm3d-pcl-viewer \

**Followed instruction on installing all dependencies from build from source readme**
1) install dependencies
sudo apt-get update && sudo apt-get install -y libboost-all-dev git jq libcurl4-openssl-dev \
                                            libgtest-dev libgoogle-glog-dev  \
                                            libxmlrpc-c++8-dev libopencv-dev \
                                            libpcl-dev libproj-dev \
                                            build-essential coreutils cmake

**Followed instruction on building Pybind11**
1) clone and build pybind11
$ git clone https://github.com/pybind/pybind11
$ cd pybind11
$ git checkout tags/v2.3.0
$ mkdir build
$ cd build
$ cmake -DPYBIND11_TEST=OFF ..
$ make
$ sudo make install

2) install python dev packages
sudo apt-get install python3-dev

After I did all of those things I was able to get the ifm3dpy library working fine using virtualenv but I have never been able to run a basic cpp (like the one from the basic library usage page). I can include the camera header file (so I know the compiler is able to find it) but when I try to include the image header file I get errors for opencv because it is looking for the opencv header in /usr/include/opencv2/ but the header lives in /usr/include/opencv4/opencv2.

I have the same issue with pcl, it is looking for the pcl headers in /usr/include/pcl but it installed to /usr/include/pcl-1.10/pcl

I am going to attach some screenshots in hopes that it helps explain a little further. I will also need to familiarize myself with docker. I have never used it but I have seen it talked about a lot with the ifm3d library. Thank you again for your patience and most importantly your willingness to help. 

[image]
[image]
[image]
[image]
[image]

closed 2021-04-05 by jbies