ifm/ifm3d #279: very basic question

Reno989, 2020-12-11:
Hello all...

I am still at the very beginning of Python programming.

For a student project, I want to use the ifm3dpy library to read data from an O3D304.

Unfortunately I can't manage to install the library.
I am using a Raspberry Pi 3 Model B.

I have tried to follow the installation instructions, but without much success.
When I try this step:

Install the software:

$ sudo apt-get update
$ sudo apt-get install ifm3d-camera \
                       ifm3d-framegrabber \
                       ifm3d-swupdater \
                       ifm3d-image \
                       ifm3d-opencv \
                       ifm3d-pcicclient \
                       ifm3d-tools \
                       ifm3d-python3 \
                       ifm3d-pcl-viewer \

the terminal gives me the information that the objects are not found.

Unfortunately, I'm not a native speaker, so I don't know if I fully understand the instructions.

Could someone explain to me in simple terms how I can get the library to work?

I would be very happy to receive an answer and your help.

Best regards
Reno989

Reno989, 2020-12-11:
Addition:
When I use 'import ifm3dpy' in my python code I get the error message that there is no module named ifm3dpy.

graugans, 2020-12-11:
Hey @Reno989 ,
what is your language of choice? Due to the fact we are a international team we do speak multiple languages. My native language for example is German.

Best,
Christian

Reno989, 2020-12-14:
Hey @graugans ,
that's wonderful. Mine too.

Vielen Dank für deine schnelle nachricht. 
Über eine Antwort auf Deutsch, würde ich mich natürlich sehr freuen.

Beste Grüße
René

GreNait, 2020-12-15:
Hi @Reno989,

---------------------------------German-------------------------------------

solange die Installation der ifm3D über apt noch nicht funktioniert hat, wird auch ein Import der ifm3dpy nicht funktionieren. Wir sollten also zuerst schauen, wie wir die ifm3D richtig installiert bekommen.

Wenn ich mir deinen vorherigen Beitrag anschaue, dann sieht man das dein Kommando zur Installation nicht richtig war:
"$ sudo apt-get update
$ sudo apt-get install ifm3d-camera
ifm3d-framegrabber
ifm3d-swupdater
ifm3d-image
ifm3d-opencv
ifm3d-pcicclient
ifm3d-tools
ifm3d-python3
ifm3d-pcl-viewer \"

Hier fehlen die ganzen "\" :

sudo apt-get update
sudo apt-get install ifm3d-camera \
                       ifm3d-framegrabber \
                       ifm3d-swupdater \
                       ifm3d-image \
                       ifm3d-opencv \
                       ifm3d-pcicclient \
                       ifm3d-tools \
                       ifm3d-python3 \
                       ifm3d-pcl-viewer \

Wenn die Installation geglück ist, kannst du durch ein "ifm3d ls" oder ein "ifm3D dump" mit angeschlossener Kamera die Funktion überprüfen.

Ich hoffe dies hilft dir weiter!

-------------------------------------English----------------------------------------
As long as your isntallation of the ifm3D via apt didn't work, you will not be able to import ifm3dpy either. We should first check that your installation is correct.

If we look at your first thread, we can see that the commando you used wasn't right:

"$ sudo apt-get update
$ sudo apt-get install ifm3d-camera
ifm3d-framegrabber
ifm3d-swupdater
ifm3d-image
ifm3d-opencv
ifm3d-pcicclient
ifm3d-tools
ifm3d-python3
ifm3d-pcl-viewer \"

We are missing all the "\" :

sudo apt-get update
sudo apt-get install ifm3d-camera \
                       ifm3d-framegrabber \
                       ifm3d-swupdater \
                       ifm3d-image \
                       ifm3d-opencv \
                       ifm3d-pcicclient \
                       ifm3d-tools \
                       ifm3d-python3 \
                       ifm3d-pcl-viewer \

If the installation was successfull, you will be able to verify that by the command "ifm3d ls" or "ifm3d dump" with a connected camera.

I hope this helps!

Reno989, 2020-12-15:
Hallo @GreNait ,

vielen Dank für deine Antwort.
Leider kann ich damit nicht so viel anfangen. Ich bin der Meinung, dass ich den Aufruf an sich richtig gemacht habe.
Zum Installieren habe ich folgendes ins Terminal eingegeben:

pi@raspberrypi:~ $ sudo apt-get install ifm3d-camera \
> ifm3d-framegrabber \
>                        ifm3d-swupdater \
>                        ifm3d-image \
>                        ifm3d-opencv \
>                        ifm3d-pcicclient [...]

Die Reaktion des RaspberryPi's sieht wie folgt aus:

Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket ifm3d-camera kann nicht gefunden werden.
E: Paket ifm3d-framegrabber kann nicht gefunden werden.
E: Paket ifm3d-swupdater kann nicht gefunden werden.
E: Paket ifm3d-image kann nicht gefunden werden.
E: Paket ifm3d-opencv kann nicht gefunden werden.
E: Paket ifm3d-pcicclient kann nicht gefunden werden.
E: Paket ifm3d-tools kann nicht gefunden werden.
E: Paket ifm3d-python3 kann nicht gefunden werden.
E: Paket ifm3d-pcl-viewer kann nicht gefunden werden.

Bei weiteren Recherchen habe ich gelesen, dass die ifm3d Bibliothek für Ubuntu kompiliert ist. Auf meinem RaspberryPi läuft Raspbian. Kann dies zu meinen Problemen führen?
Des Weitern habe ich Folgendes gefunden: https://snapcraft.io/install/ifm3d/raspbian
Diese Seite verspricht, die Nutzung unter Raspbian zu ermöglich. Die Installation hat auch ohne Weiteres geklappt. Versuche ich aber, die Bibliothek in einem Pythonprogramm zu verwenden, erscheint eine Fehlermeldung.

Ich hoffe, ich konnte mein Problem und meine bisherigen Lösungsversuche bestmöglich beschreiben und freue mich über eure Antworten.

Beste Grüße
Reno989

----------------------------------English-----------------------------------------

Hello @GreNait ,

thank you very much for your answer.
Unfortunately, I can not do so much with it. I am of the opinion that I did the call itself correctly.
To install I entered the following into the terminal:

pi@raspberrypi:~ $ sudo apt-get install ifm3d-camera \
> ifm3d-framegrabber \
>                        ifm3d-swupdater \
>                        ifm3d-image \
>                        ifm3d-opencv \
>                        ifm3d-pcicclient [...]

The reaction of the RaspberryPi looks like this:

Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket ifm3d-camera kann nicht gefunden werden.
E: Paket ifm3d-framegrabber kann nicht gefunden werden.
E: Paket ifm3d-swupdater kann nicht gefunden werden.
E: Paket ifm3d-image kann nicht gefunden werden.
E: Paket ifm3d-opencv kann nicht gefunden werden.
E: Paket ifm3d-pcicclient kann nicht gefunden werden.
E: Paket ifm3d-tools kann nicht gefunden werden.
E: Paket ifm3d-python3 kann nicht gefunden werden.
E: Paket ifm3d-pcl-viewer kann nicht gefunden werden.

Upon further research, I read that the ifm3d library is compiled for Ubuntu. My RaspberryPi is running Raspbian. Can this cause my problems?
Furthermore, I found the following: https://snapcraft.io/install/ifm3d/raspbian
This site promises to allow the use under Raspbian. The installation worked without any problems. But when I try to use the library in a Python program, an error message appears.

I hope I was able to describe my problem and my previous attempts to solve it in the best possible way and I am looking forward to your answers.

Best regards
Reno989

graugans, 2020-12-16:
Hallo @Reno989,

- 🇺🇸 🇬🇧English below 🇬🇧🇺🇸 -
Hast Du versucht mit dem zu Verfügung gestellten [APT Repository]((https://github.com/ifm/ifm3d#ubuntu-linux-via-apt-amd64arm64)) zu arbeiten? Ich kann mir gut vorstellen, dass dies aus mehreren Gründen nicht funktioniert das kann zum einen daran liegen das wie Du schon festgestellt hast wir für Ubuntu bauen und Du ein Raspian installiert hast. Mein letzter Wissensstand ist noch das Raspian 32Bit und nicht 64Bit ist. Die Snap Pakete helfen Dir auch nicht weiter, da hier nur die Tools funktionieren um die Kamera zu parametrieren. Der sicherste Weg ist es die ifm3d [aus den Sourcen zu bauen](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md#installing-ifm3d-from-source) für die Verwendung von Python solltest Du diese auch [aktivieren](https://github.com/ifm/ifm3d/blob/master/doc/python.md).

- 🇺🇸 🇬🇧 English 🇬🇧🇺🇸 -
Have you tried to add the APT repository? As described in this section of the [README](https://github.com/ifm/ifm3d#ubuntu-linux-via-apt-amd64arm64)? You are absolutly right the issue might be that we do build our packages for the Ubuntu OS and you are using Raspbian. Another issue could be the fact that Raspbian is still 32Bit and not yet switched to 64. The Snap package is only for the tooling not the library itself. I guess the safest way would be building ifm3d from source. This might take a good portion of time but in the end you do have a system that will 100% fit your OS.

How to build from source is found [here](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md#installing-ifm3d-from-source). For Python support please also follow the instructions on this [Readme](https://github.com/ifm/ifm3d/blob/master/doc/python.md)

Reno989, 2020-12-23:
Hallo zusammen,

nochmals vielen Dank für eure Unterstützung.
In den letzten Tagen habe ich einen Freund zurate gezogen, der sich mit der Thematik weitaus besser auskennt als ich. Zusammen haben wir zunächst Ubuntu Server 20.10 in der 64bit Version auf dem Raspberry Pi installiert. Im Anschluss konnten wir auch die ifm3d installieren. 
Dies hat bis auch die ifm3d-pcl-viewer Komponente ohne Probleme funktioniert. Bei der genannten Komponente hat sich folgende Fehlermeldung ergeben:

> The following packages have unmet dependencies:
> ifm3d-pcl-viewer : Depends: libpcl-visualization1.10 (>= 1.10.0+dfsg-5ubuntu1) but it is not installable
>                               Depends: libpcl-common1.10 (>= 1.10.0+dfsg-5ubuntu1) but it is not installable

Zugreifen tue ich nun über SSH auf den Pi und programmiere Python in Visual Studio Code.
Grundlegende Programme wie "Hello World" funktionieren ohne Probleme und auch der " import ifm3dpy " führt zu keiner Fehlermeldung. Meine Freude war riesig.

Im nächsten Schritt wollte ich das Beispielprogramm aus https://github.com/ifm/ifm3d/blob/master/doc/python.md ausprobieren. Dieses sieht wie folgt aus:

>#!/usr/bin/env python
>import sys
>import ifm3dpy
>
>def main():
>   cam = ifm3dpy.Camera()
>    fg = ifm3dpy.FrameGrabber(cam)
>   im = ifm3dpy.ImageBuffer()
>
>    while True:
>        if not fg.wait_for_frame(im, 1000):
>            print("Timeout waiting for camera!")
>            return -1
>
>        amp = im.amplitude_image()
>        xyz = im.xyz_image()
>
>        # now do something with `amp` and `xyz`,
>        # which are of type numpy.ndarray
>
>if __name__=='__main__':
>    sys.exit(main())

Leider führt dies zur folgenden Fehlermeldung:

>ubuntu@ubuntu:~$ python3 master/ifm-test1.pyTraceback (most recent call last):  File "master/ifm-test1.py", line 22, in <module>
>    sys.exit(main())
> File "master/ifm-test1.py", line 7, in main
>    fg = ifm3dpy.FrameGrabber(cam)
>RuntimeError: Lib: XMLRPC Timeout - can you `ping' the sensor?

Kann mir damit einer weiterhelfen???
Mein ziel ist es, das x-Bild, das y-Bild und das z-Bild einzulesen.

Was muss ich tun, damit das geht?

Nochmals vielen Dank und frohe Weihnachten
Reno989

------------------english version--------------------------
Hello all,

Thanks again for your support.
In the last few days I have consulted a friend who is much more familiar with the subject than I am. Together we first installed Ubuntu Server 20.10 in the 64bit version on the Raspberry Pi. Afterwards we were able to install ifm3d as well. 
This worked up to also the ifm3d-pcl-viewer component without problems. With the mentioned component we got the following error message:

> The following packages have unmet dependencies:
> ifm3d-pcl-viewer : Depends: libpcl-visualization1.10 (>= 1.10.0+dfsg-5ubuntu1) but it is not installable
> Depends: libpcl-common1.10 (>= 1.10.0+dfsg-5ubuntu1) but it is not installable

I now access the Pi via SSH and program Python in Visual Studio Code.
Basic programs like "Hello World" work without problems and also the " import ifm3dpy " does not lead to any error message. My joy was huge.

In the next step I wanted to try the example program from https://github.com/ifm/ifm3d/blob/master/doc/python.md. This looks like this:

>#!/usr/bin/env python
>import sys
>import ifm3dpy
>
>def main():
>   cam = ifm3dpy.Camera()
>    fg = ifm3dpy.FrameGrabber(cam)
>   im = ifm3dpy.ImageBuffer()
>
>    while True:
>        if not fg.wait_for_frame(im, 1000):
>            print("Timeout waiting for camera!")
>            return -1
>
>        amp = im.amplitude_image()
>        xyz = im.xyz_image()
>
>        # now do something with `amp` and `xyz`,
>        # which are of type numpy.ndarray
>
>if __name__=='__main__':
>    sys.exit(main())

Unfortunately this leads to the following error message:

>ubuntu@ubuntu:~$ python3 master/ifm-test1.pyTraceback (most recent call last): File "master/ifm-test1.py", line 22, in <module>
> sys.exit(main())
> File "master/ifm-test1.py", line 7, in main
> fg = ifm3dpy.FrameGrabber(cam)
>RuntimeError: Lib: XMLRPC Timeout - can you `ping' the sensor?

Can anyone help me with this???
My goal is to read in the x-image, the y-image and the z-image.

What do I have to do to make this work?

Thanks again and Merry Christmas
Reno989

GreNait, 2020-12-23:
Hi @Reno989,

die Frage erscheint jetzt dumm, aber hast du einen Ping probiert? Hat die Kamera geantwortet? Mir ist das Gleiche öfters passiert wenn eine falsche IP-Adresse auf der Kamera eingestellt war, oder ich vergessen hatte meinen Rechner von DHCP auf eien feste IP umzustellen. Vielleicht ist dies der Fall?

Wenn du die Kamera über Ping erreichst (übrigens, default IP Adresse der Kamera ist: 192.168.0.69), wäre zu überprüfen ob die Anwendung die läuft richtig eingestellt ist. Es kann nämlich sein, dass die Kamera nicht auf "Freerun" steht und dann nur Bilder schießt, wenn dies von einem Nutzer ausgelöst wird.

Eine weitere Methode zum Überprüfen der Kamera wäre ein "ifm3d ls". Kannst du von deiner Console aus dies ausführen? Wenn ja, was ergibt der Dump für einen JSON-String? 

Falls der Ping funktioniert, kannst du auch gerne mal diesen Code ausprobieren: https://gist.github.com/GreNait/cf69dc63476100b0220a32d8fe301a3b
Dieses Beispiel sollte die Anwendung 1 auf "Freerun/Continous trigger" stellen. Danach sollte dein Code-Beispiel auch funktionieren.

-Grenait
-----------------------------------------------------English------------------------------------------------

The following questiosn might sound dumb, but have you tried to ping the camera? Did the camera answer? I had the same issues several times, if the wrong ip address was set on the camera or if I forgot to switch from DHCP to an static ip address on my desktop PC. Perhaps this is the case?

If you can reach the camera by pinging it (btw. default ip address of the camera is: 192.168.0.69), you should also check the running application. It might be that the application is not set to "freerun" and it is only capturing images if triggered by a user.

A further method would be to check the camera by using "ifm3d ls". Could you send this command via a terminal? If so, what is the JSON fump looking like?

If the ping works, you could also try this code: https://gist.github.com/GreNait/cf69dc63476100b0220a32d8fe301a3b
This sample should activate the "freerun/continous trigger" on the applcation 1. AFter that, your code-sample should also work.

-Grenait

inbangsa, 2020-12-23:
Hey @Reno989,

This seems a connection issue,  please check our [troubleshoot guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md).  Please try setting no_proxy variable to camera ip-address, steps explain in troubleshoot guide.

Reno989, 2021-01-06:
Hallo zusammen,

zunächst möchte ich allen ein frohes neues Jahr wünschen und mich herzlich bei @GreNait und @inbangsa für Ihre Antwort bedanken.

Zunächst habe ich noch mal die Einstellungen der Kamera überprüft.
Im ifmVisionAssistant ist unter Gerätekonfiguration>Netzwerk DHCP ausgeschaltet und die IP-Adresse 192.168.0.69 hinterlegt.

Wenn ich die Kamera versuche zu Pingen (ping 192.168.0.69) erhalte ich die Meldung "Destination Net Unreachable" heißt für mich, die Kamera ist vom RaspberryPi aus nicht zu erreichen.
Auch das Kommando "ifm3d ls" liefert die Antwort:

ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor? 

Als Nächstes habe ich die Schritte aus dem ifm3d - Troubleshooting Guide versucht. Leider ohne Erfolg.

Die Schritte:
$ ifm3d --ip=192.168.0.69 ls
$ IFM3D_IP=192.168.0.69 ifm3d ls
und
$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation

Was kann ich noch tuen, um die Kamera ans laufen zu bekommen?

Über eine Antwort würde ich mich sehr freuen.
Reno989

--------------------------English----------------------------------------------------
Hello everyone,

first of all I would like to wish everyone a happy new year and thank you @GreNait and @inbangsa for your reply.

First of all, I checked the settings of the camera again.
In the ifmVisionAssistant under Device Configuration>Network DHCP is switched off and the IP address 192.168.0.69 is stored.

When I try to ping the camera (ping 192.168.0.69) I get the message "Destination Net Unreachable" which means the camera is not reachable from the RaspberryPi.
Also the command "ifm3d ls" returns the answer:

ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor? 

Next I tried the steps from the ifm3d - Troubleshooting Guide. Unfortunately without success.

The steps:
$ ifm3d --ip=192.168.0.69 ls
$ IFM3D_IP=192.168.0.69 ifm3d ls
and
$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation

What else can I do to get the camera working?

I would be very happy about an answer.
Reno989

inbangsa, 2021-01-06:
@Reno989  wish you a very Happy New Year. have you tried setting no_proxy variable from the troubleshoot guide 
https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#if-you-are-behind-the-proxy-network 
 
can you try 
```
$export no_proxy=<camera IP>
$ifm3d ls 
```

lola-masson, 2021-01-06:
@Reno989 Happy New Year!
In addition to @inbangsa comments, did you set up your raspberry pi's IP to be in the same IP net as the camera? It should be something like 192.168.0.10. You can check the raspberry's IP using `$ifconfig`

GreNait, 2021-01-07:
Hi @Reno989,

This is simple. You provide the IP address as an parameter when you create the camaera object:

cam = ifm3dpy.Camera([IP-address])

`cam = ifm3dpy.Camera("192.168.0.75")`

Here is a little gist to show the whole set up to receive an amplitude image:
https://gist.github.com/GreNait/7d6e50b6c24f63431076c69fde4e439a

Have fun!

Reno989, 2021-01-07:
@GreNait @lola-masson @inbangsa and @graugans 

It works!

Many thanks to all of you for your active support. Alone I would not have come so far.

The actual work of my master thesis can now begin.
Again many many thanks.  If we ever meet, the drinks are on me......at least the first round :-D

Thanks
René

lola-masson, 2021-01-07:
Glad everything works now. Good luck for your thesis!

inbangsa, 2021-01-07:
@Reno989 Happy to help you. As the issue is resolved can we close this Topic. Please create new topic for any new issue.

Reno989, 2021-01-08:
@inbangsa Yes. I think we can close this topic.
Thanks again.

closed 2021-01-08 by Reno989

graugans, 2021-01-11:
Hey @Reno989 , I am glad it is working. Some information and hopefully some hints to debug any further issues. The IP address Range ``169.254.0.0 – 169.254.255.255`` is called [link local address](https://en.wikipedia.org/wiki/Link-local_address) due to the nature of this IP range it is possible that devices change their IP settings. The recommended way is to use a private IP range like the 192.168.x.x range or if you are in the lucky position to own a private subnet to use this. If you are not planing to change anything on your configuration you can of course continue to use the set-up. In case of any future communication issues this might help.