ifm/ifm3d #281: Question:  How to successfully start NTP synchronization?

pjbesl44, 2020-12-14:
Can you give/point to a code sample for starting NTP synchronization?
I see that specifying NTPServers the json string is not enough.

graugans, 2020-12-15:
Hey @pjbesl44,

have you checked our [documentation on NTP](https://github.com/ifm/ifm3d/blob/master/doc/ntp-setup.md)

GreNait, 2020-12-15:
HI @pjbesl44,

I created a little sample with the ifm3d python wrapper (ifm3dpy). I used my windows machine as a NTP server (don't forget also to enable the ports on the firewall), using the documentation provided by @graugans. Worked perfectly.

You can find the code snippit here:
[https://gist.github.com/GreNait/a1d00a05593eba03547cf74543655729](url)

I hope this helps!
-GreNait

closed 2022-11-10 by github-actions[bot]