ifm/ifm3d #283: Imported target "ifm3d::image" includes non-existent path "/usr/include/opencv"

sliangqk, 2021-01-06:
Hello there,

I am new to ifm cameras. When I try to build the examples at [https://github.com/ifm/ifm3d-examples](url), the terminal output "CMake Generate step failed.  Build files cannot be regenerated correctly." at the cmake .. step. All detailed errors are the same one(except for in different CMakeLists.txt files): fast_app_switch/CMakeLists.txt: **Imported target "ifm3d::image" includes non-existent path "/usr/include/opencv"**.  

I have checked in the folder /usr/include/, there is no opencv folder there. But there is a ifm3d folder in this repository, which is created when I install the ifm3d package. There is a opencv folder in ifm3d folder, i.e. /usr/include/ifm3d/opencv/ exists. 

I just want to know, how should I solve this problem? Thanks

inbangsa, 2021-01-06:
@sliangqk please specify OS, its version and how you installed ifm3d.  have you installed opencv on your system ?

sliangqk, 2021-01-07:
@inbangsa thanks for responding. 

My Ubuntu version is 16.04. Yes, I have installed opencv4.5.1 on the computer before trying to install the ifm3d-examples package. But those folders of opencv4 in the repository /usr/local/ are like this /usr/local/share/opencv4/,  /usr/local/include/opencv4/, /usr/local/lib/cmake/opencv4/, /usr/local/share/licenses/opencv4/. Those repositiories mentioned are the result when I search with the keyword "opencv" under the /usr/ folder. The only folder with the name opencv in the /user/ repository is the one provided by imf, i.e. **_/usr/include/ifm3d/opencv/._**, but not the required "**_/usr/include/opencv_**". 

According to my current knowledge to CMake, I should modify the repository which consumes the ifm3d::camera library in those CMakeLists.txt files（ifm3d::camera is one argument of the command _target_link_libraries()_ ). The only problem is, I do not know exactly how to modify it, since I am also a new learner of CMake.

Thanks again for your help!

sliangqk, 2021-01-07:
sorry, an error in the last message. It is ifm3d::image but not ifm3d::camera. The error is reported for ifm3d::image, like this:
_CMake Error in timestamp/CMakeLists.txt:
  Imported target "ifm3d::image" includes non-existent path

    "/usr/include/opencv"

  in its INTERFACE_INCLUDE_DIRECTORIES._

The way of installing ifm3d is just following the tutorial on the [page](https://github.com/ifm/ifm3d) with following commands:
$ sudo apt-get update
$ sudo apt-get install ifm3d-camera \
                       ifm3d-framegrabber \
                       ifm3d-swupdater \
                       ifm3d-image \
                       ifm3d-opencv \
                       ifm3d-pcicclient \
                       ifm3d-tools \
                       ifm3d-python3 \
                       ifm3d-pcl-viewer \

The way of installing ifm3d-example is following the [page] (https://github.com/ifm/ifm3d-examples) with commands:
$ mkdir build
$ cd build
$ cmake ..
$ make
And the error occurs at the $cmake .. step.

lola-masson, 2021-01-07:
@sliangqk your first message makes me want to ensure the cmake is called properly: did you use `$cmake .. ` in `your_path/ifm3d-examples/build`, and not in the separate modules?

inbangsa, 2021-01-07:
@sliangqk  ifm3d image module depends on headers of opencv and PCL.  which it assumes to be present at /usr/include , for your system opencv is installed on /usr/local//include/opencv and hence you are facing this issue. This is a known issue to us and we are in process of releasing the fix  in upcoming releases.

At the moment following can solve your issue:
1) install opencv headers to /usr/include
OR
2) compile ifm3d from  source on system by proving opencv path to cmake. This can be done by giving 
-DCMAKE_PREFIX_PATH=/usr/local;/usr  to cmake while building ( provide both as some of the libraries can be installed at /usr)
[instruction for compilation ](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md#building-from-source)

After this compilation and Installation  cmake export for image module will have correct paths.

sliangqk, 2021-01-08:
@lola-masson Thanks for your responding! 

Yes, I have called cmake .. in my_path/ifm3d - example/build, but not in separate modules. Should I compile those modules one by one?

sliangqk, 2021-01-08:
@inbangsa  Thanks!

I do not want to re-install opencv, since it has taken me a while to install the opencv-contrib.

So I have followed your second advice. However, it does not work. 

I have run the command **_cmake -DCMAKE_PREFIX_PATH=/usr/local;/usr .._**. The output is:  CMake Error: The source directory "/home/sliangqk/Software/ifm_camera/ifm3d-master/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
bash: /usr: Is a directory

Then I tried the command with only one path: **_cmake -DCMAKE_PREFIX_PATH=/usr/local .._**. The output is as following:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_glog
    linked by target "ifm3d_camera" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/camera/src/libifm3d_camera
    linked by target "ifm3d_framegrabber" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/framegrabber/src/libifm3d_framegrabber
    linked by target "ifm3d-fg-tests" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/framegrabber/test
    linked by target "ifm3d_swupdater" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/swupdater/src/libifm3d_swupdater
    linked by target "ifm3d-swupdater-tests" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/swupdater/test
    linked by target "ifm3d_image" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/image/src/libifm3d_image
    linked by target "ifm3d-image-tests" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/image/test
    linked by target "ifm3d_tools_shared" in directory /home/sliangqk/Software/ifm_camera/ifm3d-master/modules/tools/src/libifm3d_tools

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

Have I misunderstood your instruction or not? Thanks again~

inbangsa, 2021-01-11:
@sliangqk  Lets solve your issue by changing the OpenCV path in the cmake config file. As I said earlier ```ifm3d-image-targets.cmake``` assumes that opencv library installed at /usr. But we can change this file itself to solve this issue.

1) open file ```ifm3d-image-targets.cmake``` from location ```/usr/lib/cmake/ifm3d-0.18.0/```

2) replace ```/usr/include/opencv``` at line 67 to ```/usr/local/include/opencv``` .  

Above steps will need root permissions !!

Please let us know if this solves your issue.

----------------------------
on ifm3d compilation, I forgot to mention multiple paths to CMAKE should be provided as string so please enclosed them with ```""```
so from your build folder try following 

``` $ cmake -DCMAKE_PREFIX_PATH="/usr/local/;/usr"  -DCMAKE_INSTALL_PREFIX=/usr .. ```

if you are able to  compile and installation successfully you will find above changes in the ```ifm3d-image-targets.cmake```

last point :
from log it seems libglog is missing, please install ```libgoogle-glog-dev``` with apt-get.

commit f8bc685b (2024-04-05): Merge branch 'pcic_p_command_ack' into 'main'

PCIC p command ACK based frame reception.

Closes #283

See merge request syntron/support/csr/ifm3d/ifm3d!374
closed 2022-11-09 by github-actions[bot]