ifm/ifm3d #284: ifm3d Compiling error at swupdater app with VS2019

Arag24, 2021-01-06:
**Bug description**
When compiling ifm3d from source for Windonws installation, error is flagged during build of swupdater.cpp
```
C:\ifm3d\ifm3d\modules\swupdater\src\libifm3d_swupdater\swupdater_impl.hpp(557,31): error C2593: 'operator =' is ambigu
ous [C:\ifm3d\ifm3d\build\modules\swupdater\src\libifm3d_swupdater\ifm3d_swupdater.vcxproj]
```

**To Reproduce**
Building of library on WINDOWS was perfectly fine until [Building ifm3d](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#building-ifm3d)  section.

```
#run tests
cmake --build . --config %CONFIG% --target check   // SUCCESSFUL
``` 
```
#install
cmake --build . --config %CONFIG% --target INSTALL    // FAILED
```

**Output/Screenshots**

[image]

**Configuration/Environment (please complete the following information):**
 - OS: Windows 10
 -  ifm3d version: 0.18.0
 - VS 2019. V16.8.2
 - MSVC 19.28.29334.0

**Going further...**
Jumping to swupdater_impl.hpp (557,31), this error is triggered by an implicit cast : 
  
```
  auto json = nlohmann::json::parse(status_string.c_str());
  status_id = std::stoi(json["Status"].get<std::string>());
  status_error = std::stoi(json["Error"].get<std::string>());
  status_message = json["Msg"];   //<--Here
```
It looks like JSON library at the revision of 3.6.1 is suffering issue with VS2019 > V16.5 [json.hpp (15629)].
Triggered here : 
* [https://github.com/nlohmann/json/issues/2006](https://github.com/nlohmann/json/issues/2006)
* [https://github.com/sony/nmos-cpp/issues/136](https://github.com/sony/nmos-cpp/issues/136)

May we consider to update to a sonner version fo Modern json C++ library OR use instead : 
`status_message = json["Msg"].get<std::string>();`
This line was running OK on my Target computer.

inbangsa, 2021-01-11:
@Arag24 Thanks for issue, and solutions.  we will try to update ifm3d with one of your solution in coming release.

graugans, 2021-01-12:
My recommendation here is to start with a quick fix
```cpp
status_message = json["Msg"].get<std::string>();
```
With the quick fix applied we can add a task to update the JSON library to the latest release.

closed 2022-11-09 by github-actions[bot]