ifm/ifm3d #285: Set json without save settings

aispecialvideo, 2021-01-07:
Hello,
I noticed the method `ifm3d::Camera::FromJSON` force the save of the configuration on camera memory.

In some cases I want to take the modifications in camera volatile memory only, can you add a boolean in that method for disable it?

Thanks you,
Andrea Iannì

inbangsa, 2021-01-07:
@aispecialvideo which parameters are you looking for. OR can you share your use case. At ifm3d level you can restore and change parameters with ```ifm3d::camera::ToJson ``` and ```ifm3d::Camera::FromJson``` respectively.

aispecialvideo, 2021-01-08:
I'm not looking for a particular parameter, I want to edit Imager parameters or Application parameters in volatile memory only. When I call ifm3d::Camera::FromJson it edit the parameters using "setParameter" xml-rpc call and then it call xml-rpc "save" storing my modifications in persistent memory.

graugans, 2021-01-12:
Hey @aispecialvideo, unfortunately what you are planning to do is not fully supported. The idea of the underlying session handling is to always have an consistent camera state. For some parameters it is not trivial to have them both in temporary and persistant state.

Nevertheless there are a couple of parameters which can be changed without calling the save method:

[https://github.com/ifm/ifm3d/blob/133455c243248775289dc53850df36fc6febe304/modules/camera/include/ifm3d/camera/camera.h#L251](virtual void SetTemporaryApplicationParameters(
      const std::unordered_map<std::string, std::string>& params);)

Please let me know if there are more parameters you want to change on the fly. I am also interested in the use case as @inbangsa already pointed out.

closed 2022-11-09 by github-actions[bot]