ifm/ifm3d #288: provide windows binary releases

cwiede, 2021-01-12:
**Is your feature request related to a problem? Please describe.**
There are binary releases for linux, but not for windows. Requesting our customers to compile the library themselfs on windows is a big hurdle.

**Describe the solution you'd like**
Please provide binary releases also for windows as assets attached to releases.

tomatac, 2021-09-28:
It would be great to have binaries for windows. At least for major versions.
It takes some time to setup the environment to compile from source and it can be difficult for somebody with no C++ programming experience.

graugans, 2021-09-30:
Hey @tomatac, there are Windows builds available here: https://github.com/ifm/ifm3d/actions/workflows/windows.yml

In case of any questions or comments please reach out to @inbangsa

closed 2022-11-08 by github-actions[bot]