ifm/ifm3d #289: How does a single host computer access 2 or more different O3x100's with only one environment variable $IFM3D_IP?

pjbesl44, 2021-01-12:


graugans, 2021-01-12:
Hey @pjbesl44 what are you referring to the command line tools or to C++ or Python code?

pjbesl44, 2021-01-12:
> CAUTION: This email originated from outside of the Symbotic organization. Do not click links or open attachments unless you recognize the sender and know the content. Please report any suspicious emails to Symbotic IT Helpdesk.

Hi Christian,

Very much appreciate your help.  Figured it out assuming you were saying “it should be obvious.”

In my software interface to the IFM, I call WaitForFrame( ) with a camera and a framegrabber.

auto global_ifmCamera = ifm3d::Camera::MakeShared();
auto global_ifmFrameGrabber =
   std::make_shared<ifm3d::FrameGrabber>(global_ifmCamera,
   ifm3d::DEFAULT_SCHEMA_MASK|ifm3d::EXP_TIME|ifm3d::IMG_CART|ifm3d::ILLU_TEMP);

… // later on in a while loop

if (! global_ifmFrameGrabber->WaitForFrame(global_ifmImageData.get(), 1500))

I’d never seen any examples on your github with different IP addresses being using by the same C++ main.
I searched on MakeShared on github.  Never saw a single argument much less 3 but am sure it was just my search.
I now see MakeShared can take optional arguments.
In /usr/include/ifm3d/camera/camera.h I see

    static Ptr
    MakeShared(const std::string& ip = ifm3d::DEFAULT_IP,
               const std::uint16_t xmlrpc_port = ifm3d::DEFAULT_XMLRPC_PORT,
               const std::string& password = ifm3d::DEFAULT_PASSWORD);

I get it.  You can mark the question closed.

- Paul

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D6E8D3.E58BB490]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

graugans, 2021-01-13:
Hey @pjbesl44 , there is an example about multiple cameras available here: https://github.com/ifm/ifm3d-examples/blob/master/multi_camera_grabber/ex-multi_camera_grabber.cpp

closed 2021-01-13 by graugans

pjbesl44, 2021-01-13:
Danke

PAUL BESL | senior principal engineer
e: pbesl@symbotic.com<mailto:pbesl@symbotic.com>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
[cid:image001.jpg@01D6E986.DDE721D0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.