ifm/ifm3d #291: On an ARM architecture CPU with Ubuntu 18.04 installed, im.get() takes about 4-5s?Is it normal to be so slow? The network port is 100 Mbits/s..

yuefanhao, 2021-01-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior. For example:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Output/Screenshots**
If applicable, add console output and/or screenshots to help explain your problem.

**Minimal Working Example**
If applicable/possible, please provide a [minimal working example](https://stackoverflow.com/help/minimal-reproducible-example) which demonstrates the problem and can be run standalone.

```
#include <iostream>
int main()
{
  std::cout << "Hello World!" << std::endl;
  return 0;
}
```

**Configuration/Environment (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04, Windows 10]
 -  ifm3d version: [e.g. 0.14.0]
 - using ROS: [yes/no]

**Camera Configuration**
Please include the output of `ifm3d dump`

**Additional context**
Add any other context about the problem here.

yuefanhao, 2021-01-25:
@theseankelly @graugans @lpc921

lpc921, 2021-01-25:
im.get() is a method of the shared pointer. Do you mean WaitForFrame()?
Just to clarify, I have not used the ARM build of the library.

yuefanhao, 2021-01-25:
> im.get() is a method of the shared pointer. Do you mean WaitForFrame()?
> Just to clarify, I have not used the ARM build of the library.
yes, I mean WaitForFrame()

yuefanhao, 2021-01-25:
Yes, I mean WaitForFrame(). The library supports ARM platforms, doesn't it? 
@lpc921

yuefanhao, 2021-01-25:
[image]
[image]
As the picture shows, I can ping the camera succeed,  which prove that the connection between the host and the camera is working properly. But when I read the image, it usually takes about 4-5 seconds. The problem only occurred on ARM64 CPUs.

graugans, 2021-01-25:
Hey @yuefanhao,

could you please provide more details about the ARM64 system you are using (clock & frequency and load CPU load) and if you have build the library from source code or if you have used the pre-compiled libraries. It would also help to understand the issue when You provide the configuration the camera is running at.

yuefanhao, 2021-01-25:
I have used the pre-compiled libraries. The host CPU info and host system info as below:

[image]
[image]
The configuration of the O3D303 is:
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "636708761",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "true",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "false",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "7556",
          "ExposureTimeList": "1511.2;7556",
          "ExposureTimeRatio": "5",
          "FlyingPixelEraser": "0",
          "FrameRate": "5",
          "MaxAllowedLEDFrameRate": "6.7",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_moderate",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Sample Application",
        "PcicEipResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \ [... cut ...]
        "Calibration_Device": "00:02:01:41:e0:4f",
        "Calibration_Version": "1.0.2",
        "Diagnostic_Controller": "v1.0.44-2dcaabbc73-dirty",
        "IFM_Recovery": "unversioned",
        "IFM_Software": "1.30.5309",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Mar 27 18:25:10 CET 2019",
        "Main_Application": "1.0.44"
      },
      "ifm3d_version": 1800
    }
  }
}
@graugans

yuefanhao, 2021-01-25:
I install the library just as follows:
[image]
@graugans If you need any more information, just tell me, Thank you~

desengph, 2021-01-25:
Hi @yuefanhao 
we're in the process of trying to reproduce your problem on an arm64 based system (rasberry pi 4).

graugans, 2021-01-25:
Hey @yuefanhao from the provided output I do assume you are running the code on a Nvidia Jetson based platform. If this is the case please check: https://github.com/ifm/ifm3d#linux-for-tegra 

@desengph we do have Nvidia TX2 Jetson boards as-well in case you want to test with them please get in touch.

yuefanhao, 2021-01-25:
That would be great!!!Thank you very much!!! I have consulted your after-sale technical support, and he asked me to ask questions in the GitHub community. May I ask when I can get more detailed information about this problem, because this problem is quite urgent for me. @desengph

yuefanhao, 2021-01-25:
And this is the Ethernet interface info:
[image]
@desengph

yuefanhao, 2021-01-25:
This is more detailed information about the CPU, it doesn't seem to be an Nvidia Jetson platform based. @graugans @desengph 
[image]

desengph, 2021-01-25:
@yuefanhao We are still trying to reproduce your setup configuration.
Hopefully today we can give you more information.

yuefanhao, 2021-01-25:
ok, thank you very much~
@desengph

desengph, 2021-01-25:
Hi @yuefanhao,
So far, we have not been able to reproduce the big time delay of 4 seconds. Our system (Jetson Nano running Ubuntu 18.04 LTS) with precompiled ifm3d lib for ARM64 produces a time delay of 0.2 seconds. This is consistent with a frame rate of 5 Hz.

We used your camera config, only difference: changed the trigger type to continuous.

desengph, 2021-01-25:
Perhaps you could try continuous trigger mode for testing purposes. We used this cpp program.
```cpp
#include <iostream>
#include <memory>
#include <opencv2/core/core.hpp>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main(int argc, const char **argv)
{
  auto cam = ifm3d::Camera::MakeShared();
  auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
  auto im = std::make_shared<ifm3d::ImageBuffer>();

  while (true)
  {
    auto start_time = std::chrono::system_clock::now();
    if (! fg->WaitForFrame(im.get(), 1000))
    {
      std::cerr << "Timeout waiting for camera!" << std::endl;
      return false;
    }

    auto finish = std::chrono::system_clock::now();
    auto duration_time = std::chrono::duration_cast<std::chrono::microseconds>(finish - start_time);
    auto read_frame_time = duration_time.count() / 1000;
    std::cout << "read frame time: " << read_frame_time << " ms" << std::endl;
  }

  return 0;
}
```

yuefanhao, 2021-01-25:
@desengph Hi,I also have used the continuous type to test the camera, but the time delay is still big. I'll continue to check if there are any other problem tomorrow. Anyway, thank you very much. If there is any other problem, I'll contact you later.

yuefanhao, 2021-01-26:
I'm using exactly the same code as you, in continuous mode, and I get this result, which is really weird. @desengph 
[image]

graugans, 2021-01-26:
I agree this is kind of strange, one thing you could try is to use the **ifm3d hz** command to check what is going on. This is using a custom ByteBuffer which does not contain a conversion to OpenCV or PCL which might could be an issue.

```
IFM3D_MASK=3 ifm3d hz --nframes=100 --nruns=3
```

yuefanhao, 2021-01-26:
Ok, I will check it.
@graugans

yuefanhao, 2021-01-26:
When I use the ifm3d hz command to check it. It always show as follows:
[image]
@graugans

graugans, 2021-01-26:
This is most probably due to the fact your camera is configured to software trigger please retry with:

```
IFM3D_MASK=3 ifm3d hz --nframes=100 --nruns=3 --sw
```

yuefanhao, 2021-01-26:
The same result, "Timeout waiting for camera!" this is the current configuration of the camera:
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "636708761",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "true",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "false",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "7556",
          "ExposureTimeList": "1511.2;7556",
          "ExposureTimeRatio": "5",
          "FlyingPixelEraser": "0",
          "FrameRate": "5",
          "MaxAllowedLEDFrameRate": "6.7",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_moderate",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Sample Application",
        "PcicEipResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"diagnostic_data\" }, { \"type\": \"st [... cut ...]
        "Calibration_Device": "00:02:01:41:e0:4f",
        "Calibration_Version": "1.0.2",
        "Diagnostic_Controller": "v1.0.44-2dcaabbc73-dirty",
        "IFM_Recovery": "unversioned",
        "IFM_Software": "1.30.5309",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Mar 27 18:25:10 CET 2019",
        "Main_Application": "1.0.44"
      },
      "ifm3d_version": 1800
    }
  }
}
@graugans

yuefanhao, 2021-01-26:
I have this problem on NXP LS1043A with Ubuntu 18.04, have you tested on it? @graugans

graugans, 2021-01-26:
We unfortunately do not have a ``LS1043A`` available. It seems to me that there is something related to your network set-up. Are you able to perform some checks with an other system and for example ``iperf3``

You are facing the timeout with cause the hz command uses a fixed timeout of 1000 which is typically fine. The ``hz`` command does not repack the data so to me this looks like an issue with the network set-up. Maybe a dupliacted IP or some connection issues. You also can check the output of ``ifconfig/ip`` top verify there are no transmit errors.

It maybe also worth to check ``dmesg`` (maybe with -w) on the ``LS1043A`` to check if the kernel has detected any issues

yuefanhao, 2021-01-26:
Ok, I will check that. @graugans

yuefanhao, 2021-01-26:
I use iperf3 to check the network set-up, and use dmesg command to check kernel error, but I didn't find any problems. This is the iperf3 test result:
[image]
And this is the WaitForFrame cost time:
[image]
@graugans

graugans, 2021-01-26:
Please check the ``Retr`` column in your host screenshot. Those numbers typically shall be 0 most of the time. Given the fact you have this big numbers of retries with small packages when dealing with iperf3 the TCP/IP stack has to perform a big number of retries to get the image data delivered to your host. This can explain the slow transfer rates IMHO.

graugans, 2021-01-26:
From a gigabit interface I would have expected a throughput in the 900 Mbit/s range. I get those figures:

```
Connecting to host 192.168.0.69, port 5201
[  4] local 192.168.0.201 port 59138 connected to 192.168.0.69 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec   110 MBytes   922 Mbits/sec    0    819 KBytes       
[  4]   1.00-2.00   sec   104 MBytes   875 Mbits/sec    0    947 KBytes       
[  4]   2.00-3.00   sec   109 MBytes   914 Mbits/sec    0   1.07 MBytes       
[  4]   3.00-4.00   sec   108 MBytes   908 Mbits/sec    0   1.38 MBytes       
[  4]   4.00-5.00   sec   108 MBytes   906 Mbits/sec    0   1.52 MBytes       
[  4]   5.00-6.00   sec   108 MBytes   903 Mbits/sec    0   1.68 MBytes       
[  4]   6.00-7.00   sec   107 MBytes   901 Mbits/sec    0   1.76 MBytes       
[  4]   7.00-8.00   sec   109 MBytes   912 Mbits/sec    0   1.76 MBytes       
[  4]   8.00-9.00   sec   108 MBytes   905 Mbits/sec    0   1.86 MBytes       
[  4]   9.00-10.00  sec   110 MBytes   922 Mbits/sec    0   2.06 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  1.06 GBytes   907 Mbits/sec    0             sender
[  4]   0.00-10.00  sec  1.05 GBytes   904 Mbits/sec                  receiver

iperf Done.
```

And this is the Server part:
```
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.0.201, port 59136
[  5] local 192.168.0.69 port 5201 connected to 192.168.0.201 port 59138
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   103 MBytes   860 Mbits/sec                  
[  5]   1.00-2.00   sec   105 MBytes   882 Mbits/sec                  
[  5]   2.00-3.00   sec   109 MBytes   914 Mbits/sec                  
[  5]   3.00-4.00   sec   107 MBytes   895 Mbits/sec                  
[  5]   4.00-5.00   sec   108 MBytes   906 Mbits/sec                  
[  5]   5.00-6.00   sec   108 MBytes   903 Mbits/sec                  
[  5]   6.00-7.00   sec   107 MBytes   901 Mbits/sec                  
[  5]   7.00-8.00   sec   109 MBytes   918 Mbits/sec                  
[  5]   8.00-9.00   sec   109 MBytes   911 Mbits/sec                  
[  5]   9.00-10.00  sec   109 MBytes   915 Mbits/sec                  
[  5]  10.00-10.04  sec  4.79 MBytes   928 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
```

yuefanhao, 2021-01-26:
Okay, thank you. I'll check the abnormality of the Retry, which may be the key.

yuefanhao, 2021-01-27:
I have a question about getting data from the camera, is the data transmitted over TCP/IP protocol? If so, how is the port number defined? @graugans

yuefanhao, 2021-01-27:
In addition, in the process of data transmission, how long is a frame of data? I am trying to solve the problem of data transmission delay, so I need this parameter. thank you. @graugans

graugans, 2021-01-27:
Due to our background in Industrial automation we do use TCP/IP as a protocol. The port for the data communication is 50010 the port for the parameters is 80. The length of the data transmission depends on the data set. The protocol itself comes with the information about a transfer size:

```python
import socket
import re
import struct

class Client(object):
    def __init__(self, address, port):
        # open raw socket
        self.pcicSocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.pcicSocket.connect((address, port))
        self.recvCounter = 0
        self.outFile = None
        self.debug = False
        self.debugFull = False

    def __del__(self):
        self.close()

    def recv(self, numberBytes):
        data = bytearray()
        while len(data) < numberBytes:
            dataPart = self.pcicSocket.recv(numberBytes - len(data))
            if len(dataPart) == 0:
                raise RuntimeError('connection to server closed')
            data = data + dataPart
        self.recvCounter += numberBytes
        if self.outFile != None:
            self.outFile.write(data)
        return data

    def close(self):
        self.pcicSocket.close()

# Execute the application
if __name__ == '__main__':

    client = Client("192.168.0.69", 50010)
    while True:
        answer = client.recv(16)
        ticket = answer[0:4]
        answerLength = int(re.findall(r'\d+', str(answer))[1])
        answer = client.recv(answerLength)
        answer = answer[4:-2]
        dataLen = len(answer)
        answerIndex = 0
        chunkCounter = 1
        print("Data to parse: {}\n".format(dataLen))
        print("start: {}\n".format(answer[0:4]))
        print("TICKET: {}\n".format(ticket))
        if b'star' == answer[0:4]:
            answer = answer[4:]
        while True:
            if ticket == b'0020':
                print("Skipping Algo Debug")
                break
            data = answer[answerIndex:answerIndex+16]
            if b'stop' == data[0:4]:
                break

            chunkType, chunkSize, headerSize, headerVersion = struct.unpack('IIII', bytes(data))
            print("chunkType: {}\nchunkSize: {}\nheaderSize: {}\nheaderVersion: {}\n".format(
                chunkType,
                chunkSize,
                headerSize,
                headerVersion,
            ))
            answerIndex += 16
            # read rest of chunk header
            data += answer[answerIndex:answerIndex+headerSize-16]
            answerIndex += headerSize-16
            chunkType, chunkSize, headerSize, headerVersion, imageWidth, imageHeight, pixelFormat, timeStamp, frameCount, statusCode, timeStampSec, timeStampNsec = struct.unpack('IIIIIIIIIIII', bytes(data[0:48]))
            print('''Data chunk %d:
            Chunk type: %d
            Chunk size: %d
            Header size: %d
            Header version: %d
            Image width: %d
            Image height: %d
            Pixel format: %d
            Time stamp: %d
            Frame counter: %d''' % (chunkCounter, chunkType, chunkSize, headerSize, headerVersion, imageWidth, imageHeight, pixelFormat, timeStamp, frameCount))
            if headerVersion == 3:
                json_meta = data[48:]
                print("JSON: {}\n".format(json_meta))

            # read chunk data
            data = answer[answerIndex:answerIndex+chunkSize-headerSize]
            answerIndex += chunkSize-headerSize
            chunkCounter = chunkCounter + 1
            #print("Data 0-10 {}".format(data[0:100]))
            print("answerIndex {} dataLen {} len(data): {}\n".format(answerIndex,dataLen,len(data)))
            if answerIndex >= dataLen:
                print("Read next frame....\n")
                break
```
In the code above the line
```python
answerLength = int(re.findall(r'\d+', str(answer))[1])
```
should contain the answer length. You mighht have to adjust the IP Address to match your usecase. I have tested this code with Python3 on a Ubuntu 18.04 Machine.

yuefanhao, 2021-01-28:
@graugans If the network port doesn't support jumbo frame, will it cause a big transmit delay? Our network port doesn't support jumbo frame, the max MTU value only can be set as 2048.

yuefanhao, 2021-01-29:
No solved.

closed 2021-01-29 by yuefanhao

graugans, 2021-01-29:
We do not support Jumbo Frames on the O3D3xx family of devices.