ifm/ifm3d #295: As the detection depth changes, can I know what the theoretical horizontal distance in reality is for every two pixels, assuming both pixels are in the same plane and parallel to the camera plane?

yuefanhao, 2021-02-05:


graugans, 2021-02-05:
I guess what you are looking for is the [Euclidean Distance](https://en.wikipedia.org/wiki/Euclidean_distance#Higher_dimensions) To calculate this you can use the cartesian coordinates XYZ for the two pixels you want to calculate the distance for.

yuefanhao, 2021-02-05:
Shouldn't this have something to do with pixel density? If I want to know the distance between A and B, do I need to know the density of the light emitted by the camera?I'm sorry I didn't quite understand this.
[image]

graugans, 2021-02-05:
By providing the Cartesian coordinates we have taken the optical model of the camera into account. This already contains the pixel density in my opinion. The only thing to consider is that a pixel is a square on the "Wall" which grows with the distance between the Wall and the camera. But @GreNait can provide more details and I guess even some graphics on this topic

closed 2021-02-23 by yuefanhao