ifm/ifm3d #296: How do I configure the camera with code?

yuefanhao, 2021-02-07:
When I configure the camera with the following code：
[image]
There was an error：
[image]

lola-masson, 2021-02-08:
Hi @yuefanhao !
Which camera are you using? O3X, O3D?
Could you provide the full error message? Is there a specific line where the configuration fails or does it fail for every parameter?

yuefanhao, 2021-02-09:
I'm using an O3D camera. The screenshot above is all the error messages. There is no specified line parameter configuration error. @lola-masson

closed 2021-02-23 by yuefanhao