ifm/ifm3d #298: Image quality acquired is not same as it appears in vision assistance

ShettyHarapanahalli, 2021-02-13:
Image acquired from the API are not as they appear in the vision assistance, how can I get the same quality images.

ShettyHarapanahalli, 2021-02-15:
[image]
Sharing an example image captured.

lola-masson, 2021-02-17:
How do you acquire the image outside of the vision assistant? Do you use the right application on the camera?
Within the VA, you can edit the application within the application tab. You can verify that you are using the same application configuration by doing `ifm3d dump` and checking the active application.
You can also try a different viewer (for instance [ifm3d-pcl-viewer](https://github.com/ifm/ifm3d))

ShettyHarapanahalli, 2021-02-17:
Missed updating it, I tried on a different pc and this worked. So I uninstalled each package of IFM and reinstalled its working now.

Few observation -

1. Was not able to uninstall the build version with the command "sudo make uninstall" support of this will be helpful.
2. Conflit or multiple installations from the ubuntu store and from cmake was present, might be good if a warning or handled better.

closed 2021-02-24 by ShettyHarapanahalli