ifm/ifm3d #299: What exactly do Intrinsics() and Extrinsics() represent? Can I store the parameters myself to Extrinsics()?

yuefanhao, 2021-02-23:
For Intrinsics(), the elements are: fx, fy, mx, my, alpha, k1, k2, k5, k3, k4, transX, transY, transZ, rotX, rotY, rotZ
For Extrinsics(),the elements are: tx, ty, tz, rot_x, rot_y, rot_z
what is the difference between (transX, transY, transZ, rotX, rotY, rotZ) in Intrinsics and (tx, ty, tz, rot_x, rot_y, rot_z) in Extrinsics?
Has the camera been internally calibrated by Intrinsics?

yuefanhao, 2021-02-23:
If I need to know the coordinate transformation relationship between the camera coordinate system and the world coordinate system, can I obtain the coordinate points in the camera coordinate system according to the XYZImage obtained by the camera, and then use ICP algorithm to obtain the rotation matrix and translation matrix?

lola-masson, 2021-03-02:
The intrinsic calibration in the "internal" calibration of the camera. It defines the position of the imager within the camera. Every camera is calibrated in our lab and we do not recommend changing the intrinsic calibration.
The extrinsic calibration defines the position of the camera in an external coordinate system of your choice. This is what you would use to define the relation between the camera coordinate system and the world coordinate system. If you define the extrinsic calibration, the XYZ image you receive will be in the coordinate system defines by your extrinsic calibration.

closed 2021-04-22 by yuefanhao