ifm/ifm3d #300: Problems installing Python binding

MattVoge, 2021-02-24:
Hi there,
I am using Ubuntu 20.04 and want to use your python interfaces to acquire images from my 3d tof cam.
Installing ifm3d from source worked fine, now I can communicate with the cam via terminal. However, installing python bindings did not work.
1. Installing them using pip (setup.py) raises an error: `ERROR: Command errored out with exit status 1:
   command: /home/vinroc/anaconda3/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-s6nd3177/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-s6nd3177/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-hil1n2yq
       cwd: /tmp/pip-req-build-s6nd3177/`

2. so I tried to install them using Cmake. This seems to work, at least there's no error and there is a .so file in the site-packages folder, however, I can not import ifm3dpy (no module error).

Could you please help me how to get ifm3dpy to work?

inbangsa, 2021-03-03:
@MattVoge, I tried to reproduce the issue with CMake on Ubuntu20.04 and all the steps in [documentation](https://github.com/ifm/ifm3d/blob/master/doc/python.md#build-dependencies-same-for-cmake-and-setuptools) are good to install ifm3dpy. No issues found.

So to  debug issue at your system please provide me with following information
*  which version of ifm3d you are using 
*  what is the install location of the ifm3d:  ```/usr``` or ``` /usr/local```

for me installing ifm3d in ```/usr``` and my ifm3dpy is installed at ```/usr/lib/python3/dist-packages/ifm3dpy.cpython-38-x86_64-linux-gnu.so``` works well.

MattVoge, 2021-03-08:
@inbangsa thanks for your reply. I managed to get it to work using the released v0.18 version instead of master branch.

ahmadLyneSystems, 2021-05-11:
I'm getting a similar error code building the python module on windows 10. Following the steps in the documentation, I arrive at this error: 
`ERROR: Command errored out with exit status 1: 'C:\Program Files\Python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\ahmad\\AppData\\Local\\Temp\\pip-req-build-v1f2zlsz\\setup.py'"'"'; __file__='"'"'C:\\Users\\ahmad\\AppData\\Local\\Temp\\pip-req-build-v1f2zlsz\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\ahmad\AppData\Local\Temp\pip-record-yy0yjd7e\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\ahmad\AppData\Roaming\Python\Python39\Include\ifm3dpy' Check the logs for full command output.`

and the final line from the traceback call: 
`File "C:\Program Files\Python39\lib\subprocess.py", line 373, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '/m']' returned non-zero exit status 1.`

As MattVoge suggested, I tried building the setup.py with ifm v.018 and it surprisingly works, but I still faced issues where python seemingly could not load the module or import the DLL files, which is the following error: 
> ImportError: DLL load failed while importing ifm3dpy: The specified module could not be found.

I also could not find the ifm3d module in site-packages in the python directory. 

Any help with this would be appreciated.

graugans, 2021-05-12:
Hey @ahmadLyneSystems, thanks for the feedback. Currently the Python support on Windows is in beta state. We have seen issues on different versions of the Visual Studio version. Could you please be so kind and provide the exact version of Visual Studio you have used.

Have you checked [running-the-ifm3d-command-line-tool](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#running-the-ifm3d-command-line-tool) and followed the instructions to add the ifm3d installation to the `PATH` variable.

ahmadLyneSystems, 2021-05-12:
Hey @graugans, 

Thanks for the reply. I'm using Visual Studio 16.9.4. And yeah I did add the ifm3d to Path and it works well both on cmd and on Visual Studio using the C++ modules. 

So some of the things I've tried include building from both 0.20 and 0.18, building with and without PCL, and using different versions of pybind11 (2.3.0 and the latest version).

I should also mention that there appears to be a linker error for several modules when building with setup.py: 

>  frame_grabber.obj : error LNK2019: unresolved external symbol "public: bool __cdecl ifm3d::Camera::CheckMinimumFirmwareVersion(unsigned int,unsigned int,unsigned int)" (?CheckMinimumFirmwareVersion@Camera@ifm3d@@QEAA_NIII@Z) referenced in function "protected: void __cdecl ifm3d::FrameGrabber::Impl::SetSchemaBuffer(unsigned short)" (?SetSchemaBuffer@Impl@FrameGrabber@ifm3d@@IEAAXG@Z) [C:\Users\ahmad\AppData\Local\Temp\pip-req-build-yk06f4wb\build\temp.win-amd64-3.9\Release\modules\framegrabber\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]

Which leads me to believe there is an incompatibility with the visual studio compiler as you might have suspected. I will try to build it again with a different version of visual studio and see if that works.

inbangsa, 2021-06-03:
@MattVoge  @ahmadLyneSystems for this issue, I am able to reproduce the issue  for the ifm3d version 0.20.0 [unreleased] which is the current master. Issue is with linking the header only asio module with Framegrabber module.  This will be fixed in next release.

closed 2022-11-08 by github-actions[bot]