ifm/ifm3d #302: XMLRPC Timeout - libcurl failed to execute the HTTP POST transaction

Silent-Souls, 2021-03-19:
Hi 
   It seems libcurl have some problems,but i don't know how to solve this problem.
   Hope someone could help me,thanks!

`$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
=;eth0;IPv4;O3D3XX-41-B3-A9\032\09100\05802\05801\05841\058b3\058a9\093;Workstation;local;O3D3XX-41-B3-A9.local;192.168.100.69;9;
`
`$ ping 192.168.100.69
PING 192.168.100.69 (192.168.100.69) 56(84) bytes of data.
64 bytes from 192.168.100.69: icmp_seq=1 ttl=64 time=0.377 ms
64 bytes from 192.168.100.69: icmp_seq=2 ttl=64 time=1.00 ms
64 bytes from 192.168.100.69: icmp_seq=3 ttl=64 time=1.00 ms
64 bytes from 192.168.100.69: icmp_seq=4 ttl=64 time=1.20 ms
64 bytes from 192.168.100.69: icmp_seq=5 ttl=64 time=0.990 ms
^C
--- 192.168.100.69 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4009ms
rtt min/avg/max/mdev = 0.377/0.916/1.202/0.281 ms
`
`$ IFM3D_VLOG=20 GTEST_FILTER=CameraTest.FactoryDefaults make check
[ 18%] Built target ifm3d_camera
[ 31%] Built target ifm3d_framegrabber
[ 37%] Built target ifm3d_image
[ 43%] Built target gtest
[ 50%] Built target gtest_main
[ 59%] Built target ifm3d-image-tests
Note: Google Test filter = CameraTest.FactoryDefaults
[==========] Running 0 tests from 0 test cases.
[==========] 0 tests from 0 test cases ran. (1 ms total)
[  PASSED  ] 0 tests.
[ 59%] Built target check_image
[ 75%] Built target ifm3d-camera-tests
Note: Google Test filter = CameraTest.FactoryDefaults
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from CameraTest
[ RUN      ] CameraTest.FactoryDefaults
/home/ubuntu/Documents/PCLEnv/install-folder/ifm3d-0.18.0/modules/camera/test/ifm3d-camera-camera-tests.cpp:28: Failure
Expected: this->cam_->FactoryReset() doesn't throw an exception.
  Actual: it throws.
/home/ubuntu/Documents/PCLEnv/install-folder/ifm3d-0.18.0/modules/camera/test/ifm3d-camera-camera-tests.cpp:30: Failure
Expected: this->cam_->DeviceType() doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] CameraTest.FactoryDefaults (6030 ms)
[----------] 1 test from CameraTest (6031 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (6033 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] CameraTest.FactoryDefaults

 1 FAILED TEST
make[3]: *** [modules/camera/test/CMakeFiles/check_camera] Error 1
make[2]: *** [modules/camera/test/CMakeFiles/check_camera.dir/all] Error 2
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
make: *** [check] Error 2
`
`Log file created at: 2021/03/19 21:10:56
Running on machine: tegra-ubuntu
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0319 21:10:56.891172  1900 camera_impl.hpp:406] Increasing XML-RPC response size limit...
I0319 21:10:56.892032  1900 camera_impl.hpp:410] Initializing Camera: ip=192.168.100.69, xmlrpc_port=80, password=
I0319 21:10:56.892068  1900 camera_impl.hpp:414] XMLRPC URL Prefix=http://192.168.100.69:80
E0319 21:10:56.892974  1900 camera_impl.hpp:259] ->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  <url> malformed
W0319 21:10:56.893585  1900 camera.cpp:220] Could not probe device type: Lib: XMLRPC Timeout - can you `ping' the sensor?
W0319 21:10:56.893628  1900 camera.cpp:240] Returning instance of base camera class!
E0319 21:10:56.893831  1900 camera_impl.hpp:259] ->requestSession:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  <url> malformed
E0319 21:10:56.893906  1900 camera_impl.hpp:193] Lib: XMLRPC Timeout - can you `ping' the sensor?
E0319 21:11:02.902560  1900 camera_impl.hpp:259] ->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  <url> malformed
I0319 21:11:02.918155  1900 camera_impl.hpp:419] Dtor...
~
~
~
~
~
~
~
~
"ifm3d.INFO" 13L, 1428C`

graugans, 2021-03-19:
Hey @Silent-Souls, am I right you are running ifm3d on a Nvidia TK1? Is this a L4T or some custome OS?

Silent-Souls, 2021-03-19:
> Hey @Silent-Souls, am I right you are running ifm3d on a Nvidia TK1? Is this a L4T or some custome OS?

Oh ,It’s  Nvidia  Tegra K1 ,running Linux
 for tegra .

graugans, 2021-03-19:
Given this [table](https://developer.nvidia.com/embedded/linux-tegra-archive) it is [L4T R21.8](https://developer.nvidia.com/linux-tegra-r218)

Silent-Souls, 2021-03-19:
> Given this [table](https://developer.nvidia.com/embedded/linux-tegra-archive) it is [L4T R21.8](https://developer.nvidia.com/linux-tegra-r218)?

Sorry ，it‘s apalis tk1，R21.7，I think It should be an custom os.

Silent-Souls, 2021-03-22:
> Given this [table](https://developer.nvidia.com/embedded/linux-tegra-archive) it is [L4T R21.8](https://developer.nvidia.com/linux-tegra-r218)?

Hi graugans，
    Do you have any suggestions to deal with this problem?

Silent-Souls, 2021-03-22:
Hi  @graugans 
Do you have any suggestions to deal with this problem?Is this might be an libcurl verison problem？I installed libcurl4-openssl-dev 7.35.

graugans, 2021-03-22:
@Silent-Souls do you have any idea what underlying Ubuntu OS is the R21.7 using. I do not have a TK1 running so it is hard to test it. I would consider this more a routing than a SW issue. But I am not 100% sure.

Silent-Souls, 2021-03-22:
Thank  you @graugans ，I'll do more tests anyway.

> $ uname -a
Linux tegra-ubuntu 3.10.40-V2.6.1b1+g0e41a93 #1 SMP PREEMPT Thu Sep 29 22:57:04 CEST 2016 armv7l armv7l armv7l GNU/Linux

> cat /proc/version
Linux version 3.10.40-V2.6.1b1+g0e41a93 (linuxdev@linuxdev.toradex.int) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) #1 SMP PREEMPT Thu Sep 29 22:57:04 CEST 2016

> cat /etc/os-release
NAME="Ubuntu"
VERSION="14.04.1 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.1 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

graugans, 2021-03-22:
@Silent-Souls are you sure that you do not use any proxy set-up? Please check the environment variables:

- http_proxy
- https_proxy
- no_proxy

Within our company we do use a proxy, I have to disable the proxy for my camera on IP:  192.42.125.26
```
no_proxy=192.42.125.26 wget http://192.42.125.26/licenses.html
```

Silent-Souls, 2021-03-23:
> @Silent-Souls are you sure that you do not use any proxy set-up? Please check the environment variables:
>
> * http_proxy
> * https_proxy
> * no_proxy
>
> Within our company we do use a proxy, I have to [...]

 I am sure there is no use of proxy.
> $ no_proxy=192.168.100.69 wget http://192.168.100.69/licenses.html
 --2021-03-23 10:11:06--  http://192.168.100.69/licenses.html
Connecting to 192.168.100.69:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-03-23 10:11:06 ERROR 404: Not Found.

graugans, 2021-03-23:
What is the firmware on the O3D3xx installed? Can you please confirm that there is no other device installed on the IP 192.168.0.69 by unplugging the network cable and ping.

inbangsa, 2021-03-30:
@Silent-Souls After looking at your logs, its seems the factory default test is failing. Test must be run on the device having an default IP ```192.168.0.69```  in your case IP is ```912.168.100.69``` . 

currently it is not possible to test the camera at any other IP then default one. This is due to the fact that tests will do a factory reset of device and hence the IP will be changed to default.

Silent-Souls, 2021-03-30:
> @Silent-Souls After looking at your logs, its seems the factory default test is failing. Test must be run on the device having an default IP `192.168.0.69` in your case device IP is `912.168.100.69` .
> 
> currently it is not possible to test the camera at any other IP then default one. This is due to the fact that tests will do a factory reset of device and hence the IP will be changed to default.

> Thank you @inbangsa ,I will test it as soon as possible.

Silent-Souls, 2021-04-01:
Finally，I figure out the problem.
First,my apalis tk1's gcc version is 4.8.4, which is not support c++14,so i modified the CMakefile and  found a way to  implement make_unique function in C++11,so ,compile the project doesn't have any error,but the "std::regex" or "std::regex_replace" is not supported completely in gcc4.8.4,so the problems comes out. Currently i just commented the "url = std::regex_replace(url, std::regex("\\$XXX"), this->SessionID());" in camera_impl.hpp,to my suprise it worked.

closed 2021-04-01 by Silent-Souls