ifm/ifm3d #303: WaitForFrame much longer than IfmVisionAssistant

FritSEA, 2021-04-01:
Hi,
I am using an O3D301 camera on Windows with Visual Studio 2017. The thing is, whatever settings I use, IfmVisionAssistant is about twice faster than a custom source code we made and I can't find the reason why. 
For example, when I have 5 fps and about 130ms per frame on IfmVisionAssistant, I get 280ms on my source code only for the WaitForFrame function. The dump has been done and I did load the correct .json file.
I tried using Wireshark and there is more data coming from the camera with my custom source code. I just don't know why.
Does anyone have an idea ? Thanks.

[Wireshark VisionAssistant.txt](https://github.com/ifm/ifm3d/files/6244749/Wireshark.VisionAssistant.txt)
[config.json.txt](https://github.com/ifm/ifm3d/files/6244750/config.json.txt)
[image]
[image]
[image]
[image]
[image]
[Wireshark CustomSourceCode.txt](https://github.com/ifm/ifm3d/files/6244752/Wireshark.CustomSourceCode.txt)

inbangsa, 2021-04-01:
@FritSEA from config file, it seems your trigger mode is ```2```, which is software trigger mode. Can you try changing it to  ```1``` which is continous mode. Also Remove framgrabber->SwTrigger() from your code when using continous mode. 

if you need more framerate rate please change it. you can change value in config file or can also do it as explained it in this [document](https://github.com/ifm/ifm3d/blob/master/doc/configuring.md#config).

FritSEA, 2021-04-01:
Hi,
the example does indeed use software trigger. I already did try with continuous mode and an external trigger.
I have the same problem.

Arag24, 2021-04-01:
@inbangsa : 
I got the customer on phone this afternoon. 
This topic looks weird for me. Customer's application is SW trigger. 

We did the following tests : 

- Vision Asssistant : check Frame delay: Let says it's around 140ms with his config using simulated 'sw triggered'
- ifm3d jitter tool: At that point we go back to Freerun to get some latency info. Within ifm3d : we keep 140ms with stable delays (no jitter)
- Customer application (C# Wrapper from our cpp examples) : Delay time is doubled : 280ms. Only thing in the app is Loadconfig file + SWTrigger and waitforframe()

What is weird? It looks like in Wireshark they are almost twice data information that are sent with customer's app. No idea why at this stage. 
Hope it's a bit clearer @inbangsa ?

inbangsa, 2021-04-05:
@FritSEA @Arag24 iVA frame duration value is calculated on camera and it doesn't consider the network delay. 

@Arag24 when you said more data, is it that wireshark showing same frame data twice OR data is more in byte size, I think the device is configured for 100k image resolution ( Resolution : "1") and hence the data on network (wireshark) will be much more than the default 23k.  

to compare the delay in C# application, I suggest to use ```ifm3d hz --sw``` it will show the frequency with network delay considered given the software trigger( process interface). if this value is not near to 1/280ms then we can debug more in C# application for additional delays. 

Let me know if you need any more information.

FritSEA, 2021-04-06:
The device is configured for 100k image resolution for everything I try (VisionAssistant and custom code).
I am not much familiar for decoding data from wireshark so I only know for now that I receive less data with VisionAssistant.

I tried using ifm3d hz --sw and it appears that I have 4hz vs 8hz in VisionAssistant.
Doesn't that seem a lot for network delay only ?

[image]
[image]

inbangsa, 2021-04-06:
4hz is what I am able reproduce with your config file on my device.

@FritSEA  The software trigger is also sent over the same connection on which the data is received. After trigger camera captures the image and send over connection. So the time ```ifm3d hz --sw``` compose of ( sending the trigger + capturing  the data on device + grabbing the data from device to PC + very small computation time). Whereas the iVA show 129ms which is just the 'capturing the data on device'. 

I will check why iVA is showing less data.

FritSEA, 2021-04-06:
@inbangsa At least you have the same results which is good news.
I did compare between software and external trigger and both results are about the same, which means that sending the software trigger isn't that long.

graugans, 2021-04-07:
Hey @FritSEA,
is there any chance that you can release the C# wrapper as OpenSource Code. I guess there are some people interested in such a wrapper.

Related to the frame rate issue. Like @inbangsa already have described the evaluation time is the time which is needed for the Pointcloud calculated on the camera it is not the Framerate. There is additional time needed for receiving data from the 3D imager hardware and the network transmission. In case you need high framerate settings it is recommended to use the continuous trigger mode, which provides the camera with the ability to optimally schedule the exposure and calculation of the Point cloud.
In 100k mode the calculation is 4x times more than in the 23k mode which of course lowers the throughput.

The Max framerate displayed in your [message above](https://github.com/ifm/ifm3d/issues/303#issuecomment-813891541) is only a recommendation from the Vision Assistant not the framerate of the device.

Related to the different amount of data transferred by the vision assistant and your application. This can be fine tuned with the schema mask. In the example in your initial message above you request both the radial distance image and the cartesian XYZ images, this leads to extra data. I guess the vision assistant may also disables the amplitude data for the 3D view cause it is not needed there. The ifm3d enables the confidence image by default which may be disabled by the vision assistant for displaying data. In general less data to transfer means less work on the camera and less data for your application to deal with. This will not double the framerate.

We have talked a lot about the difference between the vision assistant and the c++ interface, but what is the framerate you really need in your application? Maybe it is more intersting to fine tune your application settings than comparing with the vision assistant.

inbangsa, 2021-04-07:
@graugans  After reading you comment, I have tried to get the framerate with different schema with the config file attached in this issue.

following are the results: 

with schema ```ifm3d::IMG_RDIS```
```
sw_trigger 5.21 HZ 
free_run 7.71 Hz
```
with schema ```ifm3d::IMG_RDIS|ifm3d::IMG_AMP|ifm3d::IMG_CART```
```
sw_trigger 3.74 HZ
free_run 7.07 HZ
```

FritSEA, 2021-04-08:
@graugans It's unfortunate that I can't release my current C# wrapper because it uses a lot of abstract classes for cameras and other internal generic purposes.

About the free run mode, although I know it would be faster, I can't use it in my application because I need to synchronize the snapshot with another material which is why I use the external trigger. The software trigger is only convenient while searching for my delay problem and it enables me to easily compare with VisionAssistant.

About my needed framerate, it is about 5hz which is what seems to be possible on VisionAssistant.

FritSEA, 2021-04-12:
Good news, the problem I got disappeared ! The less good news is that I don't know why...
Last week, I wanted to check my C++ code and C# wrapper so I began by starting the camera with the same configuration. While VisionAssistant did behave the same (about 4-5hz), I only got about 1hz with both hz --sw command and my C# wrapper. Since I couldn't get a better framerate, I did a reset command and I have now the same results using ifm3d command and my C# wrapper, which is what I was looking for !

What is weird is that I already tried the reset command several times and always got the delay problem before.
I also did a reset with @Arag24 on the phone with the same result.

I noticed that when making a dump file, the IP configuration can be wrong. 
Is it possible that the dump file isn't the same as the camera configuration ?

[image]
[image]

FritSEA, 2021-04-12:
@inbangsa with schema ifm3d::IMG_RDIS only, I also have 5hz, but with a single exposure, without any filter and without rectifing the data.
Did you use that same configuration ?

inbangsa, 2021-05-28:
@FritSEA  Hope you are doing well. 

As i mentioned in my comment frame rate calculation was made with the config file attached. Yes above framerate calculation is without any filters as per the config file.  I can recalculate the frame rate, please provide the config file with the filter settings needed.

closed 2022-11-08 by github-actions[bot]