ifm/ifm3d #304: About Camera Filter Configuration?

yuefanhao, 2021-04-22:
Hello, I am having a problem with the camera configuration while detecting the width and depth of an object. When I configured the bilateral filter of O3D303 camera separately, for multiple recognition of the same scene, the depth value is more stable than that without a filter, and the width value is more stable than that without a filter. When the temporary filter of the O3D303 camera is configured alone (temporary filter type is mean with 2 images), the depth value and width value are more stable than those without filter for multiple recognition of the same scene. However, when I configured the bilateral filter and temporary filter ((temporary filter type is mean with 2 images)) of O3D303 camera at the same time, for the same scene recognition for many times, although the depth value is more stable than the filter configured alone, the stability of the width value becomes poor and fluctuates greatly, which may be caused by some reasons. Do you have any suggestions for the filter configuration in the O3D303 camera? @graugans

yuefanhao, 2021-04-22:
@inbangsa

graugans, 2021-04-22:
Hey @yuefanhao, I have forwarded this issue Image processing team. I guess the behavior depends on ordering of both filters. Could you please provide us with a dump of your current configuration and some details about the application. In case you do not want to share this in public please use this e-mail: support.sy @ ifm.com

lola-masson, 2021-04-22:
Hi @yuefanhao,
I do not get these results with a basic setup I have here. Here is what I get with the different configurations (standard deviation values):
- No filter: x: 0.004, y:0.002, z.0.015
- Bilateral alone: x: 0.002, y: 0.001, z: 0.004
- Temporal alone: x: 0.002, y: 0.001, z: 0.008
- Both filters: x: 0.001, y: 0.001, z: 0.003
I expect that this would differ with your setup (different dynamic range, configuration, etc). As @graugans pointed out, it is hard to assess without more details on your application. Please tell us as much as you can about what you are trying to achieve and we will get back to you.

yuefanhao, 2021-04-23:
I have sent an email to support.sy@ifm.com. I hope you can reply as soon as possible. Thank you. @graugans @lola-masson

closed 2021-06-02 by haoyuefan