ifm/ifm3d #305: Anybody ported the ifm3d camera libs to Yocto Linux 4.14.98?

pjbesl44, 2021-04-22:


graugans, 2021-04-23:
Hey @pjbesl44 related to this [overview](https://wiki.yoctoproject.org/wiki/Releases) there is no YOCTO version 4.14.98 I guess this is the Linux kernel version you refer to. Could you please be so kind and provide the name of your release for example `dunfell` we do provide a simple [meta-ifm3d](https://github.com/ifm/meta-ifm3d) which may need some update to match your YOCTO version.

pjbesl44, 2021-04-23:
Looks like we are still on Sumo
2.5 | Sumo | 04/2018
I've setup the fsl-imx-xwayland cross compile environment successfully. 
I just need ifm3d_camera and ifm3d_framegrabber but having trouble with 
glog, gflags, xmlrpc cross-compiles.

graugans, 2021-04-26:
I am not sure if I got your question 100% right do you want to build the ifm3d with a YOCTO based SDK or included in your image which will need a meta layer which contains ifm3d?

Most of the libraries are already in the meta-openembedded, you can easily add them to your SDK:
- [glog] (https://github.com/openembedded/meta-openembedded/tree/sumo/meta-oe/recipes-support/glog)
- [xmlrpc-c](https://github.com/openembedded/meta-openembedded/blob/sumo/meta-oe/recipes-devtools/xmlrpc-c/xmlrpc-c_1.31.0.bb)
The gflags is needed for gtest only or am I wrong. You easily can disable testing for your cross comopile builds. By adding ``-DBUILD_TESTS=OFF``

pjbesl44, 2021-04-26:
> CAUTION: This email originated from outside of the Symbotic organization. Do not click links or open attachments unless you recognize the sender and know the content. Please report any suspicious emails to Symbotic IT Helpdesk.

Hi Christian,
No success yet.
Problem is I know next to nothing about Yocto bb’s and bbconf’s etc.
I believe I have a plan now to get to my goal.
We are on Sumo and I found out I need Ubuntu 16.04 to build the OS that I want to do building and testing with.
Thinking my chances of success may be better building a full OS and doing work on it rather than cross-compiling XMRPC.

I seem to have successfully cross-compiled everything except XMLRPC.
That’s some very old code in the XMLRPC-C directory. Perhaps it’s not used.

Why does LDD show 52 separate Shared Object libraries dragged in to do even basic Camera to  JSON and JSON to Camera operations?
Seems like a lot.   Thanks for any info on that.
- Paul

PAUL BESL | senior principal engineer
e: ***@***.******@***.***>
p: 978-284-8525 | www.symbotic.com<http://www.symbotic.com>
***@***.***D73A7F.8A7D15F0]<http://www.symbotic.com/>
SYMBOTIC
200 Research Drive
Wilmington, MA 01887
Reinvent the warehouse | Reimagine the supply chain™
**Symbotic Confidential**
This e-mail and any attachments contain Symbotic confidential information and are for the sole use of the intended recipient(s). Unauthorized review, use, or disclosure is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

closed 2022-11-08 by github-actions[bot]