ifm/ifm3d #307: Problem displaying images on visual studio (Too dark or only shows Invalid pixels)

ahmadLyneSystems, 2021-05-11:
Hello, 

I've been trying to setup ifm3d with our O3D311 camera on a windows environment with visual studio 2019. I built ifm3d using the PCL library and followed your windows installation guide on multiple versions of ifm3d but I'm still facing the same issue. I'm mainly concerned with obtaining the distance image but when I run the code below, all I get is either a black image or a white one with the black pixels representing the invalid pixels from the VA software. 

Our code on C++: 
`int main(int argc, const char** argv)
{
    auto cam = ifm3d::Camera::MakeShared("10.10.0.85");
    auto fg = std::make_shared<ifm3d::FrameGrabber>(cam, ifm3d::IMG_RDIS | ifm3d::IMG_AMP | ifm3d::IMG_CART | ifm3d::IMG_GRAY);
    auto im = std::make_shared<ifm3d::ImageBuffer>();
  
    auto imTypes = cam->ImagerTypes();
    auto st = cam->ToJSONStr();
       for (std::string type : imTypes)
        std::cout << st << '\n';

    cv::Mat amp, dist, gray;
    cv::Mat xyz;

    while (true)
    {
        if (!fg->WaitForFrame(im.get(), 1000))
        {
            std::cerr << "Timeout waiting for camera!" << std::endl;
            return -1;
        }
     
        amp = im->AmplitudeImage();
        xyz = im->XYZImage();
        dist = im->DistanceImage();
        gray = im->GrayImage();
        
        //dist.convertTo(dist, CV_8U);
        cv::imshow("w", dist);
        cv::waitKey(0);
    }
    return 0;
}`

outputs: 
Distance image:
[image]

Amplitude image:
[image]

XYZ image:
[image]
I could not get the gray image to appear either as it gives an error: 

`OpenCV(3.4.9) Error: Assertion failed (size.width>0 && size.height>0) in cv::imshow, file C:\build\3_4_winpack-build-win64-vc15\opencv\modules\highgui\src\window.cpp, line 382
`
And our app dump: 

`{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "1146356873",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "1",
          "Channel": "0",
          "ClippingBottom": "263",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
[... 102 lines cut ...]
      "HWInfo": {
        "Connector": "#!02_A300_C41_04880799_008026112",
        "Diagnose": "#!03_D322_C34_04873267_008026824",
        "Frontend": "#!02_F342_C34_20_00027_008023607",
        "Illumination": "#!02_I300_001_04880153_008001175",
        "MACAddress": "00:02:01:41:F9:63",
        "Mainboard": "#!03_M381_C41_04901813_008023690",
        "MiraSerial": "8430-9701-0eeb-0794"
      },
      "SWVersion": {
        "Algorithm_Version": "3.0.5",
        "Calibration_Device": "00:02:01:41:f9:63",
        "Calibration_Version": "1.0.2",
        "Diagnostic_Controller": "O2I_1.20.7428-0-g91c60a74be",
        "IFM_Recovery": "O3D3XX 0.6.73",
        "IFM_Software": "1.71.7444",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-d",
        "Main_Application": "O2I_1.20.7428-0-g91c60a74be"
      },
      "ifm3d_version": 2000
    }
  }
}`

I've also tried to use the openCV image buffer instead but I faced some uninitialized pointer errors on that side of things which I could share if required. Please let me know if need any additional info.

lola-masson, 2021-05-11:
As for the invalid pixels: these, at first glance, look over exposed. Did you try reducing the long exposure time or using multiple exposures?

ahmadLyneSystems, 2021-05-11:
Hey lola, 

Thanks for the reply. You're right about the amplitude image being normalized which I could fix using a simple image filter. I simply thought the dark contrast was a symptom of anothe problem. However, as I mentioned I'm mainly looking to reproduce the distance image output from the VisionAssistant software. 

As for the distance image, I should have mentioned that I only get the white image when I change the openCV data format to CV_8U, otherwise it's just a black image.

The invalid pixels themselves aren't the problem because as you said, I could modify the application settings and exposure to eliminate them. When I run the above code for a distance image, I just get a black image. I tried applying different filters and transforms to it with no success. 

[image]

graugans, 2021-05-12:
Hey @ahmadLyneSystems there is some sample code for scaling the [image](https://github.com/ifm/ifm3d-examples/blob/master/simpleimage/example/ex-simpleImage_ppm_io.cpp#L68) The amplitude data typically does have more dynamics than a 255Bit grayscale image.

ahmadLyneSystems, 2021-05-12:
Hey @graugans, thanks for linking to the example code. I was actually trying to build a similar RGB scaling function on my own yesterday so this will be a great reference. Initially, I thought I had a problem with my ifm3d build which is why I asked the question. But since it's the expected output, I generally know how to proceed now. 

Most of the confusion is due to it being my first time working with a 3D camera instead of regular images so I have a much better idea of what to do now.

Thanks for the assistance @graugans & @lola-masson!

closed 2021-05-12 by ahmadLyneSystems