ifm/ifm3d #311: ifmO3X

mrcjw, 2021-05-27:
In vs2015, I have connected to O3D and can run it. Now I want to connect to o3x and use the following linked sample program:

https://github.com/ifm/ifm3d-examples/blob/master/getmac/ex-getmac.cpp , can be compiled and run, but can't get the data. If I set IP in makeshared(), the runtime program will report an error

inbangsa, 2021-05-27:
@mrcjw can you please share the error that you are getting ? can you ping the device ? 

Also to reproduce the issue, provide following details 

* dump of the device , use following command on cmd
 ```
ifm3d --ip=<ip address of device> dump 
```
* share the code snippet which can reproduce the issue

mrcjw, 2021-05-27:
> @mrcjw can you please share the error that you are getting ? can you ping the device ?
>
> Also to reproduce the issue, provide following details
>
> * dump of the device , use following command on cmd [...]

I can ping the camera. My code is:

std::string CAMERA_ IP = "192.168.0.69";

int main()

{

ifm3d::Camera::Ptr cam = ifm3d::Camera::MakeShared(CAMERA_ IP);

auto jsonConfig = cam->ToJSON();

// print out the MAC address

std::cout << "The MAC address of the camera: "

<< jsonConfig["ifm3d"]["Net"]["MACAddress"]

<< std::endl;

}

Sorry, I don't know how to do it

ifm3d --ip=<ip address of device> dump
The system prompts that the file cannot be found

inbangsa, 2021-05-27:
@mrcjw Your code works fine for me. Can you check the build type. ifm3d instruction always build the release version.  Please use Release mode if you are building from IDE. What exception you see in output windows pf VS

There is also a small typo in code provided. A space between "CAMERA_ IP" I changed it to  "CAMERA_IP" and everything works fine,

mrcjw, 2021-05-27:
> @mrcjw Your code works fine for me. Can you check the build type. ifm3d instruction always build the release version. Please use Release mode if you are building from IDE. What exception you see in output window of VS
> 
> There is also a small typo in code provided. A space between "CAMERA_ IP" I changed it to "CAMERA_IP" and everything works fine,

My program is release. If I set IP in makeshared, my program will jump to

static _ Elem *__ CLRCALL_ OR_ CDECL copy(_ Elem *_ First1, const _ Elem *_ First2,

size_ t _ Count)

{ // copy [_ First2, _ First2 + _ Count) to [_ First1, ...)

return (_ Count == 0 ? _ First1

: (_ Elem *)_ CSTD memcpy(_ First1, _ First2, _ Count));

}。

This is wrong

mrcjw, 2021-05-27:
> @mrcjw Your code works fine for me. Can you check the build type. ifm3d instruction always build the release version. Please use Release mode if you are building from IDE. What exception you see in [...]

My vs will report these problems at runtime. Does this mean that there is something wrong with my computer environment.

"Console application 2. Exe" (Win32): loaded "C::'windows','system32 ','ntdll. DLL'". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (Kernel32. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (kernelbase. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (ucrtbase. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: \ \ users \ \% CC \ \% desktop \ \% code \ \% console application 2 \ \% x64 \ \% release \ \% ifm3d \"_ camera.dll”。 Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (vcruntime140. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:  windows  system32  msvcp140. DLL". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C::'windows','system32 ','vcruntime140'"_ 1.dll”。 Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: \ \ users \ \ \\\\\\\\\\\\\\\\\\\_ client++.dll”。 Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C: / / users / cc / desktop / code / console application 2 / x64 / release / GLOG. DLL". Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C::'users','cc ','desktop','code ','console application 2','x64 ','release','xmlrpc + +. DLL '. Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C:: \ \ users \ \% CC \ \% desktop \ \% code \ \% console application 2 \ \% x64 \ \% release \ \% XMLRPC. DLL \". Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (msvcp140d. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (vcruntime140d. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: \ \ users \ \ \\\\\\\\\\\\\\\\\\\_ client.dll”。 Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C:: \ \ users \ \ \\\\\\\\\\\\\\\\\\\_ packetsocket.dll”。 Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C::" windows ⁃ system32 ⁃ ucrtbased. DLL ". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C::'windows','system32 ','ws2'"_ 32.dll”。 Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (rpcrt4. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: \ \ users \ \ \\\\\\\\\\\\\\\\\\\_ util.dll”。 Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C::'windows','system32 ','vcruntime140'"_ 1d.dll”。 Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C: / / users / cc / desktop / code / console application 2 / x64 / release / libcurl. DLL". Symbol loaded.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (advapi32. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (MSVCRT. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C::" windows ⁃ system32 ⁃ sechost. DLL ". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (secur32. DLL)". Unable to find or open pdb file.

"Console application 2. Exe" (Win32): loaded "C::" windows ⁃ system32 ⁃ sspicli. DLL ". Unable to find or open pdb file.

inbangsa, 2021-05-27:
@mrcjw 

From your output log it seems that a debug dll are getting loaded . which hints a either your application or a dependent library is built in debug mode.

```
"Console application 2. Exe" (Win32): loaded "C:: (Windows) = (system32) = (msvcp140d. DLL)". Unable to find or open pdb file.
```

please try following commands on cmd 

```
> ifm3d version
> ifm3d ls
> ifm3d dump 
```
this [doc](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#running-the-ifm3d-command-line-tool) provide instruction to build and install ifm3d from sources and also running ifm3d command line tools.

mrcjw, 2021-05-27:
> @mrcjw
>
> From your output log it seems that a debug dll are getting loaded . which hints a either your application or a dependent library is built in debug mode.
>
> ```
> "Console application 2. Exe" [...]

c:\ifm3d\ifm3d\build> ifm3d version
ifm3d: version=0.18.0

c:\ifm3d\ifm3d\build> ifm3d ls

c:\ifm3d\ifm3d\build> ifm3d dump

c:\ifm3d\ifm3d\build>

Ifm3d ls and ifm3d dump are not displayed

> @mrcjw
>
> From your output log it seems that a debug dll are getting loaded . which hints a either your application or a dependent library is built in debug mode.
>
> ```
> "Console application 2. Exe" [...]

c:\ifm3d\ifm3d\build> ifm3d version
ifm3d: version=0.18.0

c:\ifm3d\ifm3d\build> ifm3d ls

c:\ifm3d\ifm3d\build> ifm3d dump

c:\ifm3d\ifm3d\build>

Ifm3d ls and ifm3d dump are not displayed

inbangsa, 2021-05-27:
whats the error message on the console for ```ifm3d ls```  and ``` ifm3d dump```

mrcjw, 2021-05-27:
> @mrcjw
>
> From your output log it seems that a debug dll are getting loaded . which hints a either your application or a dependent library is built in debug mode.
>
> ```
> "Console application 2. Exe" [...]

C:\ifm3d-master\ifm3d\build>ifm3d ls
[
  {
    "Active": true,
    "Description": "",
    "Id": 984411720,
    "Index": 1,
    "Name": "1"
  }
]

C:\ifm3d-master\ifm3d\build>ifm3d dump
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "984411720",
        "Imager": {
          "ExposureTime": "1350",
          "ExposureTimeRatio": "50",
          "FrameRate": "25",
          "MaxAllowedFrameRate": "4.46428571428571",
          "MinimumAmplitude": "42",
          "SpatialFilter": {
            "MaskSize": "0"
          },
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {
            "": ""
          },
          "TemporalFilterType": "0",
          "Type": "upTo30m_moderate"
        },
        "Index": "1",
        "Name": "1",
        "OutputAmplitudeImage": "true",
        "OutputConfidenceImage": "true",
        "OutputDistanceImage": "false",
        "OutputGrayscaleImage": "false",
        "OutputXYZImage": "true",
        "TriggerMode": "1",
        "Type": "Camera"
      }
    ],
    "Device": {
      "ArticleNumber": "O3X100",
      "ArticleStatus": "AA",
      "Description": "",
      "DeviceType": "1:512",
      "IPAddressConfig": "0",
      "ImageTimestampReference": "1499980225",
      "Name": "New sensor",
      "OperatingMode": "0",
      "PasswordActivated": "false",
      "SessionTimeout": "30",
      "TemperatureIllu": "31",
      "UpTime": "0.0258333333333333"
    },
    "Net": {
      "MACAddress": "00:02:01:40:8D:DA",
      "NetworkSpeed": "0",
      "StaticIPv4Address": "192.168.0.69",
      "StaticIPv4Gateway": "192.168.0.1",
      "StaticIPv4SubNetMask": "255.255.255.0",
      "UseDHCP": "false"
    },
    "Time": {
      "CurrentTime": "1499980224",
      "NTPServers": "",
      "StartingSynchronization": "false",
      "Stats": "",
      "SynchronizationActivated": "false",
      "Syncing": "false",
      "WaitSyncTries": "2"
    },
    "_": {
      "Date": "Thu May 27 16:56:35 2021",
      "HWInfo": {
        "MACAddress": "00:02:01:40:8D:DA",
        "Mainboard": "#!02_M162_B30_12345678_008026357",
        "MiraSerial": "Not implemented"
      },
      "SWVersion": {
        "Algorithm_Version": "0.1.7",
        "Calibration_Device": "00:02:01:40:8d:da",
        "Calibration_Version": "1.0.0",
        "ELDK": "GOLDENEYE_YOCTO_HARDFP-312-06d9c894636352a6c93711c7284d02b0c794a527",
        "IFM_Software": "1.0.62",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00014-gb439e15-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Thu Jul 13 10:49:08 CEST 2017",
        "Main_Application": "0.4.1382"
      },
      "ifm3d_version": 2000
    }
  }
}

mrcjw, 2021-05-27:
@inbangsa I installed the environment again, and the program failed to compile. The error is: fatal error C1189:

#error: ERROR macro is defined. Define GLOG_ NO_ ABBREVIATED_ SEVERITIES before including logging.h.

inbangsa, 2021-05-27:
@mrcjw Happy to see tools working on your systems. 

are you using <windows.h> in your code ? 

Please follow following issue on GLOG 
https://github.com/google/glog/issues/267

mrcjw, 2021-05-27:
@inbangsa thank you. No error will be reported when compiling the program, but the previous error will still be reported when running,

Exception thrown at 0x00007ffc8fb91460 (vcruntime140d. DLL) in console application 2.exe: 0xc0000005: access conflict occurred while reading location 0x0000000039362e30.
The program stops here.

static _ Elem *__ CLRCALL_ OR_ CDECL copy(_ Elem *_ First1, const _ Elem *_ First2,

size_ t _ Count)

{ // copy [_ First2, _ First2 + _ Count) to [_ First1, ...)

return (_ Count == 0 ? _ First1

: (_ Elem *)_ CSTD memcpy(_ First1, _ First2, _ Count));

}

mrcjw, 2021-05-28:
@inbangsa Thank you for your patience. Finally, I found that my system environment was not right. At first, I put some dependent programs into my program running directory. Causes the program to run incorrectly.

inbangsa, 2021-05-28:
@mrcjw I am glad to hear this. Loading of vcruntime140d.dll hints the system env  is not correct. As issue is resolved hence closing this . feel free to create new issues. 

Happy to help !!

closed 2021-05-28 by inbangsa