ifm/ifm3d #313: Problems using software trigger mode

Tom127, 2021-06-29:
Hi, I am using O3D303 camera, when using continuous mode, the camera heats up very badly, and the image delay is relatively large after applying my algorithm. So I want to use the software trigger mode, how can it be implemented in software and hardware?
In addition, I tried this demo [https://github.com/ifm/ifm3d/blob/master/doc/basic_usage.md](url) and received some data. I am curious what does the data belonging to the variable` xyz `mean? And how to convert it into PointCloud2 format data?

[image]

graugans, 2021-06-29:
Hey @Tom127 , can you please provide more details about ``So I want to use the software trigger mode, how can it be implemented in software and hardware?`` The camera currently supports the following Trigger modes:

- Hardware, through the PINs at the the connector cable
- Software, through the ifm3d library
- Free-Running, with a configured frame-rate

To minimize the heating you may can lower the frame-rate or the overall exposure time. It might also help to mount the camera in a way that there is air-flow around the camera possible. If the camera is lying on your desk this may lead to high temperatures as-well.

To become a better understanding of your application It would help us to get your current configuration. The easiest way to get the current configuration is to use the `dump` command

```
$ ifm3d dump > config.json
```

To the last part of your question related to the XYZ data. Those are the Cartesian  coordinates for each pixel. In case you want to get the Pointcloud data in `PointCloud2` format you can change your code in the following way:

```cpp
#include <iostream>
#include <memory>
#include <opencv2/core/core.hpp>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>
#include <pcl/point_cloud.h>
#include <pcl/point_types.h>

int main(int argc, const char **argv)
{
  auto cam = ifm3d::Camera::MakeShared();
  auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
  auto im = std::make_shared<ifm3d::ImageBuffer>();

  cv::Mat amp;
  cv::Mat xyz;

  while (true)
  {
    if (! fg->WaitForFrame(im.get(), 1000))
    {
      std::cerr << "Timeout waiting for camera!" << std::endl;
      return -1;
    }

    amp = im->AmplitudeImage();
    xyz = im->XYZImage();
    pcl::PointCloud<pcl::PointXYZI> point_cloud = im->Cloud();  

    // now do something with `amp` and `xyz` and `point_cloud`
  }

  return 0;
}
```
The example above may need some tweaks and managing linkage against the PCL library but should give an idea how to get the data you are interested in.

Tom127, 2021-06-30:
> Hey @Tom127 , can you please provide more details about `So I want to use the software trigger mode, how can it be implemented in software and hardware?` The camera currently supports the following Trigger modes:
> 
> * Hardware, through the PINs at the the connector cable
> * Software, through the ifm3d library
> * Free-Running, with a configured frame-rate
> 
> To minimize the heating you may can lower the frame-rate or the overall exposure time. It might also help to mount the camera in a way that there is air-flow around the camera possible. If the camera is lying on your desk this may lead to high temperatures as-well.
> 
> To become a better understanding of your application It would help us to get your current configuration. The easiest way to get the current configuration is to use the `dump` command
> 
> ```
> $ ifm3d dump > config.json
> ```
> 
> To the last part of your question related to the XYZ data. Those are the Cartesian coordinates for each pixel. In case you want to get the Pointcloud data in `PointCloud2` format you can change your code in the following way:
> 
> ```c++
> #include <iostream>
> #include <memory>
> #include <opencv2/core/core.hpp>
> #include <ifm3d/camera.h>
> #include <ifm3d/fg.h>
> #include <ifm3d/image.h>
> #include <pcl/point_cloud.h>
> #include <pcl/point_types.h>
> 
> int main(int argc, const char **argv)
> {
>   auto cam = ifm3d::Camera::MakeShared();
>   auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
>   auto im = std::make_shared<ifm3d::ImageBuffer>();
> 
>   cv::Mat amp;
>   cv::Mat xyz;
> 
>   while (true)
>   {
>     if (! fg->WaitForFrame(im.get(), 1000))
>     {
>       std::cerr << "Timeout waiting for camera!" << std::endl;
>       return -1;
>     }
> 
>     amp = im->AmplitudeImage();
>     xyz = im->XYZImage();
>     pcl::PointCloud<pcl::PointXYZI> point_cloud = im->Cloud();  
> 
>     // now do something with `amp` and `xyz` and `point_cloud`
>   }
> 
>   return 0;
> }
> ```
> 
> The example above may need some tweaks and managing linkage against the PCL library but should give an idea how to get the data you are interested in.

Tom127, 2021-06-30:
Thanks for your reply! Maybe my description confuses you. In the continuous grabber mode, when the process speed of my algorithm cannot keep up with the grabber speed, a large delay will accumulate. Equivalently, my algorithm was still processing the image grabbered a few seconds ago. So, I want to control when the camera starts grabber images in my code, instead of continuous grabber images.  could you share some relevant sample codes?

regards,
Tom

graugans, 2021-06-30:
Hey @Tom127, no worries. From my point of view the easiest approach for you would be the usage of the Software Trigger. I have no out of the box example, but in [the test-cases](https://github.com/ifm/ifm3d/blob/master/modules/framegrabber/test/ifm3d-fg-tests.cpp#L278) you'll find something pretty close.

With the SW Trigger you can have a tight loop with your algorithm as the slowest part which dictates the framerate.

Tom127, 2021-07-01:
@graugans Thanks for your reply! I wrote a test code for reference [here](https://github.com/ifm/ifm3d/blob/399ea893141a9bcac1d962c6dec5ff709659e856/modules/framegrabber/test/ifm3d-fg-tests.cpp#L278). However, the image still has a large delay, and the grabber speed is only 1 frame per second. Here's my code:
[test_code.txt](https://github.com/ifm/ifm3d/files/6746234/test_code.txt)

regards,
Tom

GreNait, 2021-07-07:
@Tom127 - Could you share your configuration of the camera too? Different modes lead to fairly high evaluation times. Perhaps you also limited (by accident?) the framerate etc.

closed 2022-10-27 by github-actions[bot]