ifm/ifm3d #315: Clarification about Extrinsic Calibration: Do I need to calibrate the Camera before using it?

ahmadLyneSystems, 2021-07-06:
Hello again, 

I'd just like to confirm whether performing calibration is required to obtain accurate Z-distance data from the camera. I don't think the wording in the operations manual (under CHUNK types) is clear enough in this regard. It mentions under image data chunk types that:

"The Z matrix denotes the Z component of the Cartesian coordinate of a PMD 3D measurement. The origin of the camera's coordinate system is in the middle of the lens' front glass, if the extrinsic parameters are all set to 0." (page 28)

and

"The device extrinsic calibration can be set by the user, but it may be changed by an automatic calibration feature of the device." (page 29)

Does this mean the camera calibrates the extrinsic parameters on its own, somehow? I'm currently measuring about 2cm difference between the camera's front (from the glass) and an object it's aimed straight at. Is this intended behavior? 

Apologies for the confusing and probably beginner question. I'm familiar with the usual camera calibration using checkerboard images to eliminate distortions but I'm not sure if the same concepts apply to ToF cameras. Any assistance with this would be appreciated :)

graugans, 2021-07-07:
Hey @ahmadLyneSystems , thanks for the feedback on the documentation. The Extrinsic calibration can for example be used to transform the camera coordinate system into a robot coordinate system. Please be aware that the ifm3d library does perform [another transformation](https://github.com/ifm/ifm3d/blob/master/modules/image/src/libifm3d_image/image_buffer_impl.hpp#L234) .

The questions on accuracy can be best answered by @GreNait  and @lola-masson. To give them an better idea about your application please provide the current config of your camera and a description how far the object is from the camera and what kind of material it consist of.

ahmadLyneSystems, 2021-07-13:
Hey @graugans, sorry about the late reply!

Thanks for your response. If I understand correctly, it means that there is no required calibration for taking measurements from a stationary position. Also, if I'm understanding the transformation you linked to, it's taking the existing ExtrinsicCalibTrans parameters and offsetting to perhaps center the point cloud in the camera coordinate frame? 

As for our current setup, I pasted out camera dump into this pastebin: https://pastebin.com/mihzN4WT

We have setup the camera at 1500 mm from a small conveyor belt sitting on a white steel workbench. The belt holds tills (Recycled PET material) filled with mushrooms which are the main objects to be scanned. I should say that this is not an ideal testing area since the table itself can be quite reflective (which I suspect might be the problem) but the belt and tills are not. 

Unfortunately I don't have an image of the setup right now but I could arrange one tomorrow if it helps. 

Thanks!

lola-masson, 2021-07-13:
Hi @ahmadLyneSystems,
You are right, there is no calibration required for taking measurement "out of the box", each camera being intrinsically  calibrated in production (internal calibration setting the position of internal components in relation to the center of the front window).

Reflective surfaces could indeed be the cause of a wrong distance measurement. Do you observe the same 2cm shift for a "normal" surface (pointed straight at the wall for example)? Is the point cloud distorted (flat surfaces bent, ...)? An image of your setup would be super helpful, and a screenshot of the point cloud.

ahmadLyneSystems, 2021-07-15:
Hi @lola-masson 
Thanks for the reply. I spent the last couple of days conducting more tests and I believe the problem lies with the amount of noise being generated when taking distance measurements. 

I've replaced our setup with as many non-reflective materials as possible, using mostly cardboard and a conveyor belt and while the measurements are accurate some points across a cardboard box, it's the amount of noise which is causing the different measurements across the surface which sometimes have a range of 3-6 cm difference when taking a snapshot (swTrigger).

When running the setup in continuous mode, I found it easy to eliminate the measurement differences through the median or mean temporal filtering. 

The different spatial filters don't really help and while we are using continuous mode, we have a tight time span in our application of about 0.5 seconds from capture to placement so having to average the pixel values over several images might be too costly in terms of performance. If you have any suggestions on how I could reduce the noise (even through openCV), it would be a great help! 

[image]

lola-masson, 2021-07-29:
Hi @ahmadLyneSystems, sorry for the delay in my answer.
Have you tried adjusting the exposure times?  Your measurement should not be impacted this much by noise. From the configuration that you sent, the max exposure time (for the auto exposure of application 1) seems quite low. Try increasing it and see if it helps.

ahmadLyneSystems, 2021-10-19:
Hey @lola-masson 
Apologies for forgetting to close the issue as this is not my main account. Your suggestion did work and we are now capturing stable measurements with little fluctuations.

Thanks again!