ifm/ifm3d #318: IFM O3X1XX autoexposure

VitoTringolo, 2021-07-15:
I am using the IFM O3X1XX with firmware version 1.0.126 I would like to know if there is a way to get back the exposure value when the camera is set in autoexposure mode. I noticed that if I try to read the exposure value when the camera is in autoexposure mode I get back the last value that I set before enable the autoexposure. Also with the IFMVisionAssistant the exposure value is not updated when the autoexposure is enabled.

Also, I noticed that in some circumstances the camera is not able to acquire with the autoexposure enabled. For example if, before to enable the autoexposure, I set the "Image setting type" to "upTo15m_moderate" the "Exposure Time" to "50" and the "Exposure time ratio" to "10", once the autoexposure is enabled the camera is not able to acquire frames anymore. This happens both using the IFMVisionAssistant or the example code. There are some known limitation about the autoexposure?

lola-masson, 2021-08-03:
Hi @VitoTringolo. Sorry for the delay in the answer.
You can read the exposure values by telling the FrameGrabber to stream this info back. Check out this example: https://github.com/ifm/ifm3d-examples/blob/master/exposure_time/ex-exposure_times.cpp
I am not aware of limitations with the auto exposure that would prevent the acquisition of frames. Could you give more information about your setup? Do you get an error?

VitoTringolo, 2021-08-04:
No worries!

Perfect I missed that!

About the auto exposure issue, you should be able to replicate it if you setup your application as I described before. 
I have no errors using the IFMVisionAssistant just the camera is unable to acquire new frames. Using the C++ code, instead, I get a timeout calling the FrameGrabber WaitForFrame function.

lola-masson, 2021-08-23:
@VitoTringolo indeed I am having the same issue with the autoexposure, I am not too sure what is up, I will ask around for more details. Which FW version are you using? 
Is there any specific reason you are using autoexposure instead of manually setting the exposure times?

VitoTringolo, 2021-09-07:
@lola-masson the camera firmware version is 1.0.126. 
Do you have any news?

VitoTringolo, 2021-09-23:
@lola-masson or @graugans any news?

graugans, 2021-09-23:
@VitoTringolo , sorry for the delay. There is no way to get the current exposure time when the camera was configured to auto exposure. I highly recommend updating to the lastest Firmware. We had some issues with the auto exposure in FW 1.0.126.

VitoTringolo, 2021-09-24:
Thank you @graugans 
I will update the camera to the latest firmware. Do you know if is it possible to execute also the downgrade eventually?

graugans, 2021-09-24:
@VitoTringolo a downgrade is always possible

closed 2022-10-24 by github-actions[bot]